dresden-elektronik/deconz-rest-plugin #336: Xiaomi buttons not visible in Phoscon

jo-me, 2017-12-20:
I have a Xiaomi Smart button and an Aqara Smart button - both added to deconz. Their events are visible in the websocket api and in the REST API.

In the Phoscon App, however, the buttons are neither listed under switches nor under sensors.

ebaauw, 2017-12-20:
Same here for the Xiaomi Aqara Smart button.  And for the Cube.  The Xiaomi Aqara door sensor and weather sensors do show.

The IKEA Trådfri dimmer shows, but no icon.

ldplusse, 2018-01-14:
_The IKEA Trådfri dimmer shows, but no icon._
Is that the rund dimmer that you can only rotate?

ebaauw, 2018-01-14:
Yep.

Try2Fly, 2018-01-17:
I'm also unable to connect the Xiaomi Smart Home Wireless Switch (round) with Deconz Rest plugin. With the Phoscon App it does connect but not show up. When trying through the Wireless Light Web App I get not connection at all. Is there a fix for this?

ErnstEeldert, 2018-02-03:
Same here for the Xiaomi cube. It shows up in the REST API like this:

`    "11": {
        "config": {
            "on": true,
            "reachable": true
        },
        "ep": 2,
        "etag": "daade30ae2d367f3848e487902830fd7",
        "manufacturername": "LUMI",
        "mode": 1,
        "modelid": "lumi.sensor_cube",
        "name": "lumi.sensor_cube 11",
        "state": {
            "buttonevent": 4000,
            "lastupdated": "2018-02-03T11:57:38"
        },
        "type": "ZHASwitch",
        "uniqueid": "00:15:8d:00:01:14:82:de-02-0012"
    },
    "12": {
        "config": {
            "on": true,
            "reachable": true
        },
        "ep": 3,
        "etag": "fcb0cdd7c52306b0a788f5abbc0aa599",
        "manufacturername": "LUMI",
        "mode": 1,
        "modelid": "lumi.sensor_cube",
        "name": "lumi.sensor_cube 12",
        "state": {
            "buttonevent": 3065,
            "lastupdated": "2018-02-03T11:45:05"
        },
        "type": "ZHASwitch",
        "uniqueid": "00:15:8d:00:01:14:82:de-03-000c"
    }
}`

But nothing in the classic nor the new web app. Also, it doesn't show up in Home Assistant, but that maybe a different issue?

manup, 2018-02-03:
The Xiaomi button and cube are currently not visible in the Phoscon App, they can only be configured via REST API and other projects which use the REST API. Work that matter has started already in form of a generic switch configuration editor where various actions can be assigned to physical buttons or gestures (as the cube has). I have no ETA yet but the feature has a high priority for us.

etique57, 2018-03-03:
hi @manup 

any news on the generic button configurator?
Would existing bindings appear? or would they have to be created anew?

closed 2018-11-23 by manup

Oliviakrkk, 2018-12-08:
Hi

Any news on that? 

I added Xiaomi button, it is visible in Windows GUI but not in the phoscon. I connected Home Assistant, and was under impression that it uses the REST api, so the button should be visible there.. But it is missing :/.

willsy555, 2018-12-08:
> Hi
> 
> Any news on that?
> 
> I added Xiaomi button, it is visible in Windows GUI but not in the phoscon. I connected Home Assistant, and was under impression that it uses the REST api, so the button should be visible there.. But it is missing :/.

Just got mine. Same problem. Shows up in Deconz on the server, but not in the Phoscon app! Nothing in hassio... dammit! :-(
Cheers
Will.

manup, 2018-12-09:
https://github.com/dresden-elektronik/deconz-rest-plugin/releases/tag/V2_05_50

The square buttons are now supported, round one still on the todo list.