dresden-elektronik/deconz-rest-plugin #337: support for Living Color Remote

thundergreen, 2017-12-20:
Hi all. FIrst thanks for this awesome work!

I successfully added the Philips LivingColor Lamp to Deconz. But the Remote wont work to pair or I can't figure out how.

Just to be precise. This is the lamp and the remote:

https://www.download.p4c.philips.com/files/7/7099760pu/7099760pu_dfu_deu.pdf

ThorstenDtk, 2018-01-02:
I'd also like to have this feature. With the Hue gateway it's possible.

ebaauw, 2018-01-02:
I understand you need to touchlink the LC remote to the Hue bridge to get it to pair?  Not sure if procedure could be performed at all with deCONZ.  I think it only supports resetting devices through touchlink.

And then, I understand, it would only control the lights directly - not through the Hue bridge.  So we'd need to sniff the ZigBee commands sent by the LC to support it on the gateway.  If I misunderstood, could you please post what the resource on the Hue bridge looks like?   And do you know what button presses/releases/holds correspond to what `buttonevent` values?

ThorstenDtk, 2018-01-03:
First of all - I'm new to Zigbee/Deconz/Hue/Tradfri, so please excuse if I'm asking stupid questions :-)

>  I understand you need to touchlink the LC remote to the Hue bridge to get it to pair? 

Not quite. Without using a bridge I can pair the remote control directly with the LC. But in order to control the LC by the gateway (no matter if I use hue or RaspBee) I need to reset LC before it can be discovered by the gateway.  So I end up in a situation where I can control LC with RaspBee but no longer with the remote control. It would be nice to be able to continue using the RC.

There are a couple of descriptions on how to achieve this, see for example [https://www.housecontrollers.de/lichtsteuerung/anleitung-philips-livingcolors-mit-hue-verbinden/](url)
Don't know if you speak German... scroll down to 'Anleitung: Anbidnung der LivingColors an Philips Hue'

The steps they do are:
- reset hue bridge
- reset the remote control
- pair the remote control with the LC
- pair the remote control with the hue bridge

I was just wondering whether this is possible with RaspBee too. Btw - RaspBee works great for me, so I won't use the hue gw any longer. What confused me is that there is PhosCon and the Light Manager. Are there any plans to merge both into a single product?

ebaauw, 2018-01-03:
> It would be nice to be able to continue using the RC.

For this, the LC lamp, the LC remote, and the RaspBee need to be on the same ZigBee network.

> There are a couple of descriptions on how to achieve this, see for example [...] Don't know if you speak German...

I do speak (and read) some German.  In this example (and the others I've seen), they touchlink the LC lamp to the LC remote (Schritt 3) and then they touchlink the Hue bridge to the LC remote (Schritt 4).  This means that the Hue bridge joins the network created by the remote (hence the need to reset the Hue bridge (Schritt 1)), rather than the LC remote joining the network created by the Hue bridge.

When you link the LC lamp to deCONZ, the lamp joins the network created by deCONZ, and loses its connection to the LC remote network.  As I said before, afaik, deCONZ doesn't support joining another network trough touchlink.  If you have a ZigBee sniffer, you might and capture the network PANID and encryption key from the LC remote and try manually configuring those in deCONZ, but I'm not sure if that'll work.

The procedure they describe in the Update is basically the same as linking the LC lamp to deCONZ: they use touchlink to reset the lamp to factory new, so it can join the Hue bridge network.

> What confused me is that there is PhosCon and the Light Manager. Are there any plans to merge both into a single product?

Phoscon is the new version of the web app, currently in beta.  Light Manger is the old (current) web app.  I think it will be deprecated once Phoscon is fully functional and out of beta.

manup, 2018-01-04:
I'm afraid the remote can't be supported since it won't join ZLL or ZHA networks, it uses a proprietary touchlink key only known by Philips. Would be cool though to get it working, it's not impossible to retrieve the key via hardware hacking since the remote uses an older controller with some flaws ... better solution would be Philips to certify the remote.

thundergreen, 2018-01-04:
So non chance for the moment to get the remote work... Pity....

thundergreen, 2018-01-05:
https://github.com/inguardians/zigbee_tools

This might be useful?

cosmo84, 2018-05-14:
I am quite confused...I have a LC and as I read above you were able to add it to deconz, just the remote is not working.
I don't get the LC added at all. Using touchlink I can make it blink and also reset, but it does not appear when I open the network. What am I doing wrong?I read something about encryption that only Phillips knows, but how could I blink/reset it then?

closed 2019-03-30 by stale[bot]

Thorhammerk, 2019-06-13:
Hi, any news on this. I am not able to link the LC to deconz. The LC is blinking and also reset but it does not appear when i search for new lights.

regards
Thor

manup, 2019-06-17:
Nope still the same issue, we don't have the security key which the remote uses and therefore can't join it.

klodeckl, 2021-04-15:
Still no news here?

klodeckl, 2021-04-15:
I am searching for a replacement to order. I would like to have one where I can change the color (not just presets).

How about one of these or is it the same problem?

https://www.iluminize.com/de/shop/neuheiten-iluminize/product/488-511-344-hand-fernbedienung-zigbee-rgbw-led-4z.html

https://www.everen.de/led-steuerungen-controller/zigbee-smart-light-controller/2017/zb-re5-1-rgbcct-5-in-1-zigbee-3.0-fernbedienung

https://www.amazon.de/Paulmann-500-67-Smarthome-Fernbedienung-Wandhalterung/dp/B071WZZ12Y/

https://led-homeshop.de/immax-neo-remote-control-rgb-cct-fernbedienung-zigbee-3.0?gclid=EAIaIQobChMIz-K9r46B8AIVB-7tCh0PhAC-EAQYAiABEgKKDfD_BwE

https://www.bueromarkt-ag.de/funkfernbedienung_mueller-licht_tint_404011_weiss,p-404011,l-google-prd,pd-b2c.html?gclid=EAIaIQobChMIz-K9r46B8AIVB-7tCh0PhAC-EAQYBSABEgLa6fD_BwE

https://www.longlife-led.de/ATOS-ZigBee-3.0-RGB-CCT-Fernbedienung/3989