dresden-elektronik/deconz-rest-plugin #339: deCONZ uses 100% CPU

onalzengin, 2017-12-21:
Hey,

Some time after starting of the app, deCONZ hits 100%. I couldn't find a way to reproduce this, it just happens after some time. I copied this line from `top`: 
`24388 pi        20   0  242448  55452  43248 R 100.0  5.9 218:23.08 deCONZ`

I removed everything and re-installed but it didn't help. Currently it has 2 IKEA Trådfri bulbs and 1 IKEA Trådfri remote. Any idea on how to debug?

onalzengin, 2017-12-22:
Update: I removed and reset everything, reinstalled deCONZ and I just let it run without adding anything and it hit 100% again.

NathanSweet, 2017-12-25:
Maybe try running with logging. Kill deCONZ and run it again with:
```
deCONZ --dbg-info=2 --dbg-zdp=1 --dbg-zcl=1 --db-aps=1 --dbg-http=1
```

onalzengin, 2017-12-25:
I did but couldn't come up with anything useful. I'll try again and try to post the logs when it hits 100%

onalzengin, 2017-12-29:
Looks like if I disable UPnP, problem is solved. Some SSDP message probably makes deCONZ hit 100%.

closed 2018-09-17 by manup