dresden-elektronik/deconz-rest-plugin #340: Motion sensor notification received even with no battery

steveStavros, 2017-12-22:
Hi, 

I have used the deconz in order to connect several different devices. I recently added the Centralite's Motion Sensor and I noticed the following strange behaviour:

When I enter a room a "presence=true" is received. If I leave the room, after about 30 secs a "presence=false" notification is received. So far, so good.
But if I stay in the room and keep moving, after exactly 1 minute a "presence=false" notification is received. I thought that motion sensors were supposed to send this kind of notifications after an amount of time with no motion activity.

At first I thought that this is was a device issue, but then I noticed this:
I turned on the motion sensor and it sent a "presence=true" alarm. I immediately removed the battery of the device and after exactly one minute the same "presence=false" alarm re-appeared, although the device is not supposed to send anything, so I suppose this is a deconz related issue.

Is anyone aware of this kind of functionality?? Is there a way to override it? 
Also, is there a way to config the time needed for the device to go to "presence=false" status?? Right now, it needs about 30sec

Thanks

ebaauw, 2017-12-22:
As far as I know, only the Philips Hue motion sensor sends a notification when motion is no longer detected.  For the other sensors, deCONZ sets `state.presence` to false autonomously, `config.duration` seconds after the motion sensor sent a notification that motion was detected.  See #317.

steveStavros, 2017-12-22:
It seems that Centralite's motion sensor works like Philips Hue motion sensor. By using the rest-api I set the config.duration=0 and that disabled the autonomously change of presence by deconz.

Thanks a lot!!!

ebaauw, 2017-12-22:
What deCONZ version are you on?  Can you confirm that `config.duration` is still 0 after restarting deCONZ?

steveStavros, 2017-12-22:
I use the 2.04.86 version. After restarting deCONZ the config.duration value went back to 60!! Does this mean that I have to change the config.duration value of all my sensors every time the app restarts? Is there a way to keep permanently the value I set?

ebaauw, 2017-12-22:
> Is there a way to keep permanently the value I set?

No, they broke it, see #317.  I'll see if I can create a PR to fix this (again...), but there's a lot of confusion about motion sensors, and I only know the Philips Hue and IKEA Trådfri.  Could you post a screenshot from the deCONZ GUI, what the Centralite looks like?  Does it have a 0x04040 _Occupancy sensing_ cluster?  If so, does it implement attribute 0x0010 _PIR Occupied to Unoccupied Delay_?

> Does this mean that I have to change the config.duration value of all my sensors every time the app restarts? 

Unfortunately, for now, yes.  
If you're comfortable with the REST API, you could automate this using gateway rules.  Below is how I will do this.

I keep a CLIPGenericFlag, _Night_ as a virtual switch to enable my house's "night mode":
```
$ ph_get /sensors/103
{
  "config": {
    "on": true,
    "reachable": true
  },
  "manufacturername": "homebridge-hue",
  "modelid": "CLIPGenericFlag",
  "name": "Night",
  "state": {
    "flag": false,
    "lastupdated": "2017-12-22T10:30:25"
  },
  "swversion": "1",
  "type": "CLIPGenericFlag",
  "uniqueid": "/sensors/100-103"
}
```
I have a rule to set this flag at 23:30:
```
{
  "actions": [
    {
      "address": "/sensors/103/state",
      "body": {
        "flag": true
      },
      "method": "PUT"
    }
  ],
  "conditions": [
    {
      "address": "/config/localtime",
      "operator": "in",
      "value": "T23:30:00/T07:00:00"
    }
  ],
  "created": "2017-11-25T10:06:12",
  "lasttriggered": "none",
  "name": "Night On",
  "periodic": 0,
  "status": "enabled",
  "timestriggered": 19
}
```
and to clear it at 07:00:
```
{
  "actions": [
    {
      "address": "/sensors/103/state",
      "body": {
        "flag": false
      },
      "method": "PUT"
    }
  ],
  "conditions": [
    {
      "address": "/config/localtime",
      "operator": "in",
      "value": "T07:00:00/T23:30:00"
    }
  ],
  "created": "2017-11-25T10:06:12",
  "lasttriggered": "2017-12-22T11:30:25",
  "name": "Night Off",
  "periodic": 0,
  "status": "enabled",
  "timestriggered": 68
}
```
I find that one of these rules always triggers when deCONZ starts.  Note that the rule's `lasttriggered` shows local time (when did they change that?!), whereas the sensor's `state.lastupdated` shows UTC.

I use this to set another CLIPGenericFlag, _Boot Time_:
```
{
  "config": {
    "on": true,
    "reachable": true
  },
  "manufacturername": "homebridge-hue",
  "modelid": "CLIPGenericFlag",
  "name": "Boot Time",
  "state": {
    "flag": true,
    "lastupdated": "2017-12-22T10:30:25"
  },
  "swversion": "0",
  "type": "CLIPGenericFlag",
  "uniqueid": "/sensors/100-107"
}
```
through this rule:
```
{
  "actions": [
    {
      "address": "/sensors/107/state",
      "body": {
        "flag": true
      },
      "method": "PUT"
    },
    {
      "address": "/groups/0/action",
      "body": {
        "on": false
      },
      "method": "PUT"
    }
  ],
  "conditions": [
    {
      "address": "/sensors/103/state/lastupdated",
      "operator": "dx"
    },
    {
      "address": "/sensors/107/state/flag",
      "operator": "eq",
      "value": "false"
    }
  ],
  "created": "2017-11-25T10:06:12",
  "lasttriggered": "2017-12-22T11:30:25",
  "name": "Boot Time",
  "periodic": 0,
  "status": "enabled",
  "timestriggered": 68
}
```
The rule fires only once, the first time _Night_ is changed, due to the condition on _Boot Time_ being false (deCONZ initialises each sensor's `state.flag` to false on startup).  Consequenly, _Boot Time_'s`state.lastupdated` records the time deCONZ started (what's in a name...).

Currently, I use this _Boot Time_ rule to turn off all lights on startup (assuming the restart is due to a power outage), but instead you could set the motion sensors' `config.duration` to 0 here.

This is just my setup.  I think you could do without _Night_ and combine the three rules into two (with conditions on `localtime` and _Boot Time_'s `state.flag`.

In the Hue bridge, I used the update of the Daylight sensor after startup instead of `localtime`.  I hope to be able to that for deCONZ as well, once Daylight is implemented (see #73).

steveStavros, 2017-12-22:
Here's a screenshot of the sensor's clusters:
[image].

The Occupancy sensing cluster contains the following attributes:
[image]

Thanks also for the rules. I'll give them a try, although I will probably try to override this issue differently (feeling more comfortable when using Java). I have created a Java app that listens to all the events sent to deCONZ and uses the REST-API to retrieve devices, configure them, etc. Actually all the logic related to listened events is written in Java.

I have created a function that checks if the websocket connection is lost (generally this happens if deCONZ restarts or closes) and tries to re-establish it. I will probably try to re-config all motion sensors through the rest-api whenever a new connection is created. I am already on it. Hopefully it will work!! 

Thanks again.

ebaauw, 2017-12-22:
> Here's a screenshot of the sensor's clusters:

Thanks!  Indeed, that looks a lot like the Hue motion sensor.

> If I leave the room, after about 30 secs a "presence=false" notification is received.

That's probably the value for _PIR Occupied to Unoccupied Delay_.  Did you try and change that in the deCONZ GUI (double-click on the line to get a popup).  Does the sensor support attribute reporting for this attribute (what happens if you read the report config)?

> I have created a function that checks if the websocket connection is lost (generally this happens if deCONZ restarts or closes) and tries to re-establish it. I will probably try to re-config all [...]

It should.  I'm still trying to run my home automation without any external programmes, but as long as these run on the Raspberry as deCONZ (and homebridge), an additional programme wouldn't hurt that bad.

steveStavros, 2017-12-22:
> Did you try and change that in the deCONZ GUI (double-click on the line to get a popup). Does the sensor support attribute reporting for this attribute (what happens if you read the report config)?

Probably you are right. The default value of _PIR Occupied to Unoccupied Delay_ is 30. I changed it but with no results. 
I am a little bit confused about the way sensors work. All the events received are related to cluster IAS ZONE. Specifically, the device id of the event belongs to a device with uniqueid=....-0500.

I haven't realized yet how I can enable the sensor to send events related to cluster 0406 (Occupacy sensing cluster), although occasionally I receive some of these events.  I guess this is why altering the value of  _PIR Occupied to Unoccupied Delay_  has no results

ebaauw, 2017-12-22:
Value 65536 (0xffff) means attribute reporting is supported, but not configured.  Unsurprising, since deCONZ doesn't configure reporting for this attribute.  It should have configured attribute reporting for  _Occupancy_, though:
[image]

Next to this configuration, there must be a binding from the 0x0406 cluster to the ZHA endpoint on the RaspBee.  If this is setup correctly, should should see the value of _Occupancy_ change when the sensor (no longer) detects motion.  If not, you might try and create the binding manually: open the Bind Dropbox, drag the cluster to the _Source_ and the ZHA enpoint of the RaspBee to the _Destination_ and click _Bind_.
[image]

closed 2019-03-30 by stale[bot]