dresden-elektronik/deconz-rest-plugin #342: Group of non-colour lights appear to have colour attributes

marthoc, 2017-12-23:
I’ve noticed this while testing the forthcoming Home Assistant integration; version .97 or .98 fixed the issue with individual non-colour lights having colour attributes set, but it seems like groups of non colour lights (both Hue white ambiance and also Hue white) have colour attributes applied to them as they appear to have colour x,y 0,0 and 0,63,255 (which results in a blue bulb icon in Home Assistant). 

Is this expected behaviour or a bug?

ebaauw, 2017-12-23:
The Hue bridge only shows attributes of its members.  So a group of only plugs only shows `action.on`.

marthoc, 2017-12-23:
So @ebauuw if I understand correctly this is a bug?

ebaauw, 2017-12-23:
I think so.

marthoc, 2018-11-23:
@manup Only commenting because I got the notification from stale_bot - this is still an issue on the latest release of deCONZ. If there are no colour lights in a group, then a group should not have colour attributes.

manup, 2018-11-24:
Yes problem isn't fixed yet, closing this issue so that we have just one to track process of the problem in https://github.com/dresden-elektronik/deconz-rest-plugin/issues/415

ebaauw, 2021-03-20:
This is still an issue

manup, 2025-05-02:
Closing as DDFs now explicitlly control which attributes ere exposed for a device.

closed 2025-05-02 by manup

ebaauw, 2025-05-02:
This issue is **not** about the attributes exposed for a device, but about the attributes exposed on a `/groups` resource.

Currently, a `/groups` resource gets a default set of `action` attributes, roughly whatever could be exposed on the `state` of a `/lights` resource back in the day.  This includes attributes that none of the member lights support, so e.g. a group of only_Dimmable Light_ devices still gets `action/xy`.  Also this excludes attributes that (some) member lights support, e.g. a group of _Window Covering_ devices won't get `state/lift`.