dresden-elektronik/deconz-rest-plugin #343: Can't open GUI

GatoPharaoh, 2017-12-23:
deCONZ and its GUI alaways worked for me - I didn't use it as much in the last weeks as my setup was working perfectly. But now I need to get back into the GUI and it just won't appear on my desktop. I killed the PID via terminal multiple times and restarted it via desktop and also tried restarting it via its systemd service.
It just won't appear. The only thing of deCONZ that appears in the syslog is the following:
`Dec 23 18:04:59 raspberrypi systemd[1]: Started deCONZ: ZigBee gateway -- GUI/REST API.
Dec 23 18:05:00 raspberrypi deCONZ[5568]: libEGL warning: DRI2: failed to authenticate
Dec 23 18:05:00 raspberrypi deCONZ[5568]: libpng warning: iCCP: known incorrect sRGB profile
`

Anynone with a solution?

DOliana, 2017-12-30:
I have the same problem with different log entries in syslog:
```
Dec 30 09:32:44 raspberrypi deCONZ[9060]: Cant find EGLConfig, returning null config
Dec 30 09:32:44 raspberrypi deCONZ[9060]: Unable to find an X11 visual which matches EGL config 0
Dec 30 09:32:44 raspberrypi deCONZ[9060]: Could not initialize OpenGL for RasterGLSurface, reverting to RasterSurface.
```

Interestingly I see a service deconz-gui that is running.
BTW: I use RDP to connect to my Raspberry.

Somberland, 2018-01-01:
Same problem here. Even with the new image from the homepage (pre-installed deconz)

GatoPharaoh, 2018-01-03:
so as a temporary solution I went back to the last version that does show the GUI for me - which is `v. 2.04.92` - this also seems to be the last version before the systemd service was introduced. Any version with it won't show the GUI for me.

manup, 2018-01-04:
You need to stop the deCONZ headless service before starting the GUI.

```bash
$ sudo systemctl disable deconz
$ sudo systemctl stop deconz
```

There is also a gui systemd service available (which is used in the dresden elektronik sd-card image).

```bash
$ sudo systemctl enable deconz-gui
$ sudo systemctl start deconz-gui
```

Note in order to see the GUI remotely login via VNC.

GatoPharaoh, 2018-01-04:
@manup tried that but I won't show up - I'm using Raspbian btw

manup, 2018-01-04:
How are you logged into the RPi – VNC or otherwise?

Note: The latest dresden elektronik sd-card image already has the deconz-gui service activated, so you should see the GUI when logged in via RealVNC viewer.

What is the output of:

```
$ systemctl status deconz

 and

$ systemctl status deconz-gui
```

mossmd, 2018-01-05:
I use tightvnc on RPi3b with stretch and deconz 99.  Glad to hear some have brought up this problem. I finally figured out I could get a gui with
  sudo systemctl stop deconz
and then run deCONZ from a terminal window on RPi and the gui will come up.  Also, the --dbg-xyx messages will go to sysout instead of /var/log/daemon.log, so you can watch them or redirect to a file from the deCONZ command.

mossmd, 2018-01-05:
I mean stdout (perl-talk, not 370-talk)

ebaauw, 2018-01-05:
To watch the debug messages in real-time when running deconz-gui as a service, issue:
`journalctl -afn 1000 -u deconz-gui`.  To enable the debug messages, see https://github.com/dresden-elektronik/deconz-rest-plugin/issues/295#issuecomment-352182378.

mossmd, 2018-01-06:
Thanks, ebaauw, was not familiar with journalctl.  I usually just do tail -f /var/log/xxx |grep.  I mentioned dbg output to stdout since it then doesn't go to daemon.log and flood /var/log.  Don't like having to clean up /var/log.
FWIW, your, manup's, and all the other posters' comments within the issues area are fantastically helpful. I decided to bite the bullet with Zigbee after deciding to go with homekit and got the RaspBee (to open up more than BLE devices or multiple commercial hubs to AppleTV).  Looks promising and worth the effort.

GatoPharaoh, 2018-01-08:
@manup after stopping deconz, `systemctl status deconz` returns:
`Jan 08 13:24:28 raspberrypi systemd[1]: Started deCONZ: ZigBee gateway -- REST API.
Jan 08 13:24:28 raspberrypi deCONZ[5566]: libpng warning: iCCP: known incorrect sRGB profile
Jan 08 13:24:29 raspberrypi deCONZ[5566]: This plugin does not support propagateSizeHints()
Jan 08 13:24:29 raspberrypi deCONZ[5566]: This plugin does not support propagateSizeHints()
Jan 08 13:24:30 raspberrypi deCONZ[5566]: This plugin does not support propagateSizeHints()
Jan 08 13:24:58 raspberrypi systemd[1]: Stopping deCONZ: ZigBee gateway -- REST API...
Jan 08 13:24:58 raspberrypi systemd[1]: deconz.service: Main process exited, code=exited, status=143/n/a
Jan 08 13:24:58 raspberrypi systemd[1]: Stopped deCONZ: ZigBee gateway -- REST API.
Jan 08 13:24:58 raspberrypi systemd[1]: deconz.service: Unit entered failed state.
Jan 08 13:24:58 raspberrypi systemd[1]: deconz.service: Failed with result 'exit-code'.
`

after starting deconz-gui, `systemctl status deconz-gui` returns 
`Jan 08 13:31:53 raspberrypi systemd[1]: Started deCONZ: ZigBee gateway -- GUI/REST API.
Jan 08 13:31:53 raspberrypi deCONZ[11675]: libEGL warning: DRI2: failed to authenticate
Jan 08 13:31:53 raspberrypi deCONZ[11675]: libpng warning: iCCP: known incorrect sRGB profile
`
I'm accessing my pi via VNC - I'm using raspbian jessie

GatoPharaoh, 2018-03-07:
tightvncserver dosen't support the new deconz-gui version - useing x11vnc instead works!

closed 2018-03-07 by GatoPharaoh