dresden-elektronik/deconz-rest-plugin #344: Tradfri motion sensor - first trigger return dark=false

mariusmotea, 2017-12-25:
Hi,

I have a  problem with one of my tradfri motion sensors, when i enter in a room where the sensor is present i need to wait about 5 seconds to have the light turned on. I start to analyse the issue and i find that first event on websocket is sent with dark attribute set to false, and only the next one is received with dark set to true to have the light tuned on. I'm not sure if my sensor is faulty and these are exactly the realtime events sent by him, or deconz apply some extra customization and for some reasons here it fail. On another sensor i saw almost the same behavior, but no problem when is completely dark, just in very low light, and this made me believe that the issue is related to the sensors.
All run firmware version 1.2.214.

Marius

ebaauw, 2017-12-25:
There's a dial on the back of the sensor to switch between Day mode (sun) and Night mode (moon).  In Night mode, when it's light (enough), the sensors sends a command to turn the lights on, but only if they're on already.  In Day mode, and in Night mode when it's dark, it sends a command to turn the lights on unconditionally.  deCONZ reverse engineers these commands to set `state.presence` and `state.dark`.

mariusmotea, 2017-12-26:
I believe the issue is related to sensor, i made now a better inspection, from three sensors two have this issue, both looks to be a new revision and have an extra screw that secure the battery compartment. I will try to ask for a replacement on one of my sensors to see if new one have same behavior.

ebaauw, 2017-12-26:
Does the newer version still have the two dials: Day/Night mode and Duration?  I only have the old (?) version (without screw on the battery compartment).  It shows _SW Build ID_ 1.2.214 and _Date Code_ 20170302.  I cannot get it to query for OTAU, but I'd love to see if the older and newer sensor have the same firmware _Image_ (i.e. if the change is purely the enclosure or also the ZigBee logic).

mariusmotea, 2017-12-26:
I open both new and old to check for differences. It seams just the case is different (extra screw, different fonts...), inside is the same pcb marked with version 2.0, same components with same placement, the only difference is that on new one they use laser to print the firmware number of zigbee module, on old one is write by hand with a marker. I will ask ikea to see if they had a bad lot of sensors or maybe i was unlucky.
Thanks for help!

closed 2017-12-26 by mariusmotea