dresden-elektronik/deconz-rest-plugin #346: Supported smoke detectors

ftonat, 2018-01-01:
Hi,
Are there any smoke and/or CO detectors supported by deconz?
Preferable also working with homebridge. 

Thx gor any hints 
Frank

ebaauw, 2018-01-01:
Not at the moment, see #152,  #261.  Do you have another one?

I'm happy to support them in homebridge-hue, once they're supported in deCONZ.

ftonat, 2018-01-01:
Hmmm... Xiaomi wand honeywell devices seem to be not supported in short time. 
Has anybody tried the ABUS Smartvest devices yet?
https://www.abus.com/ger/Sicherheit-Zuhause/Alarmanlagen/Smartvest-Funk-Alarmanlage/Sensoren-und-Aktoren/Smartvest-Funk-Rauch-Hitzewarnmelder

I have allready Paulmann lights in use. 
Seems to me that devices from „Smart friends“ concept are very close to the zigbee 3.0 standard. 

If deconz / homebridge-Hue development team is interested to support smoke detectors, I would give it a try and order one for testing purpose. 

Which data or log is needed from deconz to evaluate if supporting this device is possible once its connected?

Frank

ebaauw, 2018-01-01:
> Seems to me that devices from „Smart friends“ concept are very close to the zigbee 3.0 standard.

Do your Paulmann lights connect to deCONZ?  I see no mention of ZigBee on the ABUS website nor on the smart-friends website, but Paulmann mentions:
> Use our Smart Friends Box and app to control all your Paulmann lamps and luminaires via the ZigBee radio standard, or use our optional remote control.

> homebridge-Hue development team

That would be me...

> If deconz development team is interested to support smoke detectors

I cannot speak for @manup, but the delay in supporting the Xiaomi and Honeywell detectors is not due to lack of interest.

> Which data or log is needed from deconz to evaluate if supporting this device is possible once its connected?

If the device is fairly standard, some screenshots from the deCONZ GUI of the endpoints and clusters the device exposes.  If the device has a more exotic interpretation of the ZigBee standards, we might need to resort to sniffing the ZigBee messages the device sends and receives.

ftonat, 2018-01-02:
Paulmann lights are working fine. 
I am currentky using:
5 Dresden Scene Switches
1 Dresden Lightnibg Switch
7 tuneable White Spot Sets 3x3.5
2 tuneable White flood 7W
1 tuneable White flood 12W
1 light Ball rgbw 
3 GU10 bulbs (tuneable white)
All from paulmann Smart Home / Smart Friends zigbee series. 

Since some deconz versions ago, they were recognized as „Paulmann CCT light“ unstead of „unknown“ (.84?)

„Smart Friends“ seems to use zigbee 3 as base technology to interact between participating brands. Please refer to https://de.paulmann.com/beratung/licht-tipps/paulmann-home/smart-home/ for more details. 

I have ordered an ABUS Smarvest Smoke detector now (fair price of 45€, compared to Nest, Eve and other „smart devices“)
I will check as soon as it will arrive if it connects to deconz (probably on thursday)

Frank

ftonat, 2018-01-04:
So sorry to tell you, that ABUS smoke detector is not connecting to deCONZ. 
I guess the SmartVest base is mandatory to be used for making the sensors “Smart Friends” ready. 

But other good news:
There is an other company offering zigbee smoke detectors and other sensors and devices: develco products. This is an OEM manufacturer for zigbee devices. 
I have some contacts to the management. Maybe they are interested in supporting @manup and you to integrate their products in deCONZ and homebridge-hue.

ebaauw, 2018-02-18:
As of deCONZ v2.05.04 and homebridge-hue@0.6.5, the Heiman sensors are supported as ZHAFire (Smoke, Combustable Gas), ZHAWater, and ZHACarbonMonoxide.

Please post deCONZ GUI screenshots for other sensors.

Ronoon, 2018-07-11:
Well fyi
 this is a snap from the Heiman / Schwaiger Smoke detector with the get command from the rest Api.
[image]

closed 2019-03-18 by stale[bot]