dresden-elektronik/deconz-rest-plugin #348: modify scene cannot update colormode

vanviegen, 2018-01-05:
It seems that once a colormode is chosen for a light within a scene, it cannot be changed using the /groups/*/scenes/*/lights/*/state endpoint. Is this a bug, or is there another way this should be done?

```
▶ huescript get /groups/63/scenes/1
lights:
- bri: 137
  colormode: xy
  id: "4"
  on: true
  transitiontime: 10
  x: 0.299989
  y: 0.199988
name: test
state: 0

▶ huescript put /groups/63/scenes/1/lights/4/state '{"colormode":"ct","ct":366}'
- success:
    id: "1"

▶ huescript get /groups/63/scenes/1
lights:
- bri: 137
  colormode: xy
  id: "4"
  on: true
  transitiontime: 10
  x: 0.299989
  y: 0.199988
name: test
state: 0
```

Thanks!
Frank

ademobile, 2018-06-28:
I am trying to adjust the colour temperature in a scene for a particular lamp and having a similar issue, that you cannot seem to adjust it. My scene has the following:

`{"lights":[{"bri":254,"colormode":"ct","ct":164,"id":"7","on":true,"transitiontime":10},{"bri":254,"colormode":"ct","ct":164,"id":"6","on":true,"transitiontime":10},{"bri":254,"colormode":"ct","ct":173,"id":"14","on":true,"transitiontime":10},{"bri":254,"colormode":"ct","ct":164,"id":"15","on":true,"transitiontime":10}],"name":"front-white","state":0}`

As you can see lamp ID 14 has a different ct to the others and even if I issue a 'put' command to change it, nothing happens. Its as if ct is not supported, when in reality it needs to be!

ebaauw, 2018-06-28:
You cannot PUT colormode, probably a bug (bad error handling) in the API.  It’s set automatically, by PUTting xy, hue/sat, or ct.  What lights do you have?  The IKEA lights don’t support ct in scenes, so they’re stored as xy instead.   See #53.

ademobile, 2018-06-29:
They are all OSRAM in that group. If a scene has been 'set' you should be able to modify it, i.e. change colour temperature if the lamp supports it (you can via the /lights/ api) and also if the lamp supports colour mode, then you should be able to change it in the scene.

I got round this by turning the lights on setting each on to the required ct and then saving the scene. 

From a programming point, there should be consistency in API calls.

closed 2019-05-26 by stale[bot]