dresden-elektronik/deconz-rest-plugin #350: API key not persistent?

jaal2001, 2018-01-06:
I'm using the Raspbee Image with the current 2.04.99 Version of deconz.
On the Raspbee Image I've been running homebridge, and the https://github.com/gutweiler/homebridge-platform-deconz plugin to connect deconz to homebridge.
After every restart of the Raspbee and deconz I do have to renew the API key. Are the API keys not stored persistent somewhere?
(I've noticed that after every reboot there is a "A new software update is available" message in the GUI. Seems like the update is also not persistent. Maybe both of these are leading to a common problem).

ebaauw, 2018-01-06:
Hadn’t seen that plugin before.  You might try homebridge-hue, it offers way more functionality.

Not sure how homebridge-platform-deconz works, but it would need to store the api key, or you would need to edit config.json manually (as with homebridge-hue) to store it.  The deCONZ REST plugin stores the API keys in its database, and exposes these through `config.whitelist`, but you can’t get to that without a valid API key.

jaal2001, 2018-01-07:
The API key ist stored in the config.json, but after a reboot the deCONZ didn't accept that API key anymore. But I think is based that I was not able to generate an API key with the correct application name and so deCONZ could not match that after the reboot. At least this is my theory.
But I will follow your suggestion and switch to homebridge-hue. Looks like your plugin is more maintained.

closed 2018-01-07 by jaal2001

ebaauw, 2018-01-07:
deCONZ only checks for the key, not for the application name.