dresden-elektronik/deconz-rest-plugin #351: Global on/off state with rules - not working

balu-, 2018-01-06:
I think the mistake is on my side somewhere but i can't find it so maybe someone else spots it.

I'm tring to model a global "on/off" State that is changed by a Switch (for future use in combination with motion detectors & to learn about the possibilities etc.).
The Switch is a IKEA Tradfri remote. 
It has the sensor id 4.

**What I did**
Therefor I created a sensor:

<details>

  <summary>Click to expand</summary>

```
POST /api/<auth>/sensors
{
    "config": {
        "on": true
        "reachable": true
    }
    "manufacturername": "Me"
    "modelid": "T1000"
    "name": "HomeSwitch"
    "swversion": "1.0"
    "type": "CLIPGenericFlag"
    "uniqueid": "0x001fee00000008bb"
}
```
Response
```
[{
  "success": {
    "id": "5"
  }
}]
```

</details>

Afterwards i created two rules to change the Virtual-Sensor values if the Button is pressed.
First rule, set the state to off if it was on:

<details>

  <summary>Click to expand</summary>

```
POST /api/<apikey>/rules
{
    "actions": [{
      "address": "/sensors/5/state",
      "body": {
        "flag": false
      },
      "method": "PUT"
    }],
    "conditions": [{
      "address": "/sensors/4/state/buttonevent",
      "operator": "eq",
      "value": 1002
    },
    {
      "address": "/sensors/5/state/flag",
      "operator": "eq",
      "value": true
    }],
    "name": "Switch Global Home to OFF",
 }
```

</details>

Second set it to one if it was off

<details>

  <summary>Click to expand</summary>

```
POST /api/<apikey>/rules
{
    "actions": [{
      "address": "/sensors/5/state",
      "body": {
        "flag": true
      },
      "method": "PUT"
    }],
    "conditions": [{
      "address": "/sensors/4/state/buttonevent",
      "operator": "eq",
      "value": 1002
    },
    {
      "address": "/sensors/5/state/flag",
      "operator": "eq",
      "value": false
    }],
    "name": "Switch Global Home to ON",
 }
```

</details>

Unfortunately it doesn't work, the Value of Sensor 5 never changes.
Anyone any ideas what i'm doing wrong?

ebaauw, 2018-01-06:
You need to specify string values for the `value` in conditions (so `"value": "true"` and `"value": "1002"`).  In the `body` of the action you need to set the actual JSON value (as you currently do).

balu-, 2018-01-07:
@ebaauw thanks for the tipp,
unfortunately it's not working

I recreated the rules with the changes you suggested,
stat stays unchanged. Via Websocket the button events are broadcasted so the button seems to work.

<details>

  <summary>Rules</summary>

  "3": {
    "actions": [{
      "address": "/sensors/4/state",
      "body": {
        "flag": false
      },
      "method": "PUT"
    }],
    "conditions": [{
      "address": "/sensors/5/state/buttonevent",
      "operator": "eq",
      "value": "1002"
    }, {
      "address": "/sensors/4/state/flag",
      "operator": "eq",
      "value": "true"
    }],
    "created": "2018-01-03T20:33:51",
    "etag": "someEtag",
    "lasttriggered": "none",
    "name": "Switch Global Home to OFF",
    "owner": "someId",
    "periodic": 0,
    "status": "enabled",
    "timestriggered": 0
  },
  "4": {
    "actions": [{
      "address": "/sensors/4/state",
      "body": {
        "flag": true
      },
      "method": "PUT"
    }],
    "conditions": [{
      "address": "/sensors/5/state/buttonevent",
      "operator": "eq",
      "value": "1002"
    }, {
      "address": "/sensors/4/state/flag",
      "operator": "eq",
      "value": "false"
    }],
    "created": "2018-01-03T20:34:22",
    "etag": "someEtag",
    "lasttriggered": "none",
    "name": "Switch Global Home to ON",
    "owner": "someOwner",
    "periodic": 0,
    "status": "enabled",
    "timestriggered": 0
  }
</details>

balu-, 2018-01-07:
Mea Culpa, 
after waiting some time, and pressing the button again it worked to well 😉 
(i had to add change condition to not trigger a endless loop) 

<details> 
    <summary>for futur reference or if someone is interested</summary>
{
    "actions": [{
      "address": "/sensors/4/state",
      "body": {
        "flag": false
      },
      "method": "PUT"
    }],
    "conditions": [{
      "address": "/sensors/5/state/buttonevent",
      "operator": "eq",
      "value": "1002"
    },
    {
      "address": "/sensors/4/state/flag",
      "operator": "eq",
      "value": "true"
    }, 
    {
      "address": "/sensors/5/state/lastupdated",
      "operator": "dx"
    }
],
    "name": "Switch Global Home to OFF",
 }
</details>

closed 2018-01-07 by balu-