dresden-elektronik/deconz-rest-plugin #352: Smoke detector Schwaiger ZHS20 resp. HEIMAN HS1SA

zbzhs20, 2018-01-07:
Hi,

I connected a smoke detector to deCONZ, it is labeled "ZHS20" from Schwaiger GmbH (www.schwaiger.de). It seems to be a HEIMAN HS1SA, see Basic Cluster Info.
The HEIMAN smoke detector is certified, id is ZIG16075ZHA25799-24.
There is a Z-WAVE version of the HEIMAN smoke detector, so be careful when ordering.

I can get the cluster infos (sometimes I had to press the button at the device to get a response,
probably because it's a battery powered device).

Is there a way to get the battery status and alarms through the web socket?
The cherry on top would be to turn on the siren through the REST API :-).

If you need more information or if I should check anything for you please feel free to comment!

Here are the cluster infos:

[image]

[image]

[image]

[image]

[image]

[image]

[image]

[image]

ebaauw, 2018-02-18:
It looks a lot like the Heiman Fire sensor that's now supported with v2.05.04, but with a different _Model Identifier_.  It even reports the same firmware version.

I'll relax the checks for _Model Identifier_ a bit, so this one will be recognised as well.

manup, 2018-02-18:
I totally need this siren for the office :D

ebaauw, 2018-02-18:
The Heiman siren would be excellent for the office: it even continues to scream when unplugged.  You can activate the strobe and siren independently or combined through the _IAS WD_ cluster, see https://github.com/dresden-elektronik/deconz-rest-plugin/issues/404#issuecomment-364436095.  Now all we need is REST API support...

Still haven't figured out what's easier: a `/lights` resource, or a `/sensors` resource.  It provides an _IAS Zone_ cluster as well, but it doesn't seem to report anything through it.  Maybe I should just create a ZHAAlarm `/sensors` resource for this cluster, and see if deCONZ will commission it.

I suppose it should be possible to bind a sensor directly to the siren so the alarm will sound even when deCONZ isn't running?

manup, 2018-02-18:
> The Heiman siren would be excellent for the office: it even continues to scream when unplugged. You can activate the strobe and siren independently or combined through the IAS WD cluster, see #404 (comment). Now all we need is REST API support...

Haha can't wait to combine this with a motion detector 

> Still haven't figured out what's easier: a /lights resource, or a /sensors resource. It provides an IAS Zone cluster as well, but it doesn't seem to report anything through it. Maybe I should just create a ZHAAlarm /sensors resource for this cluster, and see if deCONZ will commission it.

Some IAS devices go through a *enroll process* for setup and need to be configured with the writeable IAS_CIE_ADDRESS ZCL attribute (0x0010) before they start to operate.

Might be worth to check the ZigBee specification for this matter, it's likely that there are some bits'n'pieces missing in the deCONZ  configuration of IAS devices.

> I suppose it should be possible to bind a sensor directly to the siren so the alarm will sound even when deCONZ isn't running?

I'm not aware of a device which supports it, ClimaxTechnology does have various IAS products maybe there is one which can trigger the alarm directly.

closed 2019-03-30 by stale[bot]

opechunka, 2019-10-18:
I'm using Domoticz. Smoke sensor works fine, but no battery reports. Can it be fixed? The same problem with Heiman water sensor. In contrary Heyman Temperature and humidity sensor, which reports battery well.

popy2k14, 2020-02-04:
@ebaauw Want to get/integrate some of the Schwaiger ZHS20 (Heiman  HS1SA-M) sensors in my deCONZ+fhem system.
Searched here and fhem forum but could not find an answer to my question:

Will they be exposed as an special device or motion sensor?
Means -> Motion -> alarm?

thx

popy2k14, 2020-02-05:
Got it, the following are supported:

`state.fire
config.battery
state.lowbattery
state.tampered`

In fhem forum a user confirmed they are working: https://forum.fhem.de/index.php?topic=108130.0