dresden-elektronik/deconz-rest-plugin #354: Latency when controlling lights with homemade switches

esseki, 2018-01-08:
Hi,

This is not exactly an issue that I would like to report, but more a question that I would appreciate to receive clarifications about.

In the webapp (Phoscon), I have added a Philips dimmer remote and I have to admit that I’m impressed by the reactivity : when I press a button on my remote it feels like the light switch on or off instantly. To be honest it is not faster than the native Philips Hue ecosystem, but still it is impressively fast and I love it.

Then I bought some Xiaomi button to create some cheap fake switches that fit perfectly with my existing wired switches (see the picture below).

[image]

Xiaomi button cannot be added as switches in the Phoscon app. I added them in the legacy app and they do show as sensors in the REST API. Which is good for me. I can live without a UI and use the API to create rules. 

So I created some rules to switch one light on and off when the button is clicked. Before that I created a “CLIPGenericFlag” sensor to store the state of the light (On or Off) as it seems that lights (in the API) cannot be used in the condition part of a rule. Only sensors can be (correct me if i'm wrong).

Here is an example of rule that I have created to switch on a light when the corresponding  “CLIPGenericFlag” sensor is marked as off :

```
"6":{  
      "actions":[  
         {  
            "address":"/lights/16/state",
            "body":{  
               "on":true
            },
            "method":"PUT"
         },
         {  
            "address":"/sensors/16/state",
            "body":{  
               "flag":true
            },
            "method":"PUT"
         }
      ],
      "conditions":[  
         {  
            "address":"/sensors/18/state/buttonevent",
            "operator":"eq",
            "value":"2002"
         },
         {  
            "address":"/sensors/16/state/flag",
            "operator":"eq",
            "value":"false"
         },
         {  
            "address":"/sensors/18/state/lastupdated",
            "operator":"dx"
         }
      ],
      "created":"2018-01-07T21:14:41",
      "etag":"b5cca3df7e16d60a84fa7cad4dfe987c",
      "lasttriggered":"2018-01-08T07:34:09",
      "name":"Switch button Plafonnier Salle a manger On",
      "owner":"6644606069",
      "periodic":0,
      "status":"enabled",
      "timestriggered":5
   },
```

As I mentionned it at the beginning of my post, it works. I can switch on and off my light with one tap on the button. So I’m not reporting any bug.

The only thing is that doing so is adding some latency and my Xaomi buttons are a bit less responsive than the Philips dimmer switch. It is enough to be noticeable and I don’t understand how you are dealing (programmatically) with the Philips switches to avoid any latency ? Is this due to the rule engine ? Is there any hope I can do the same to remove any latency with my buttons ?

ebaauw, 2018-01-08:
The Hue dimmer switch, when configured through Phoscon or the deCONZ web app, controls the lights directly, not going through the gateway (shutdown deCONZ and the switch still works!).  I.e. the switch sends a ZigBee group command, which is picked up by the lights in that group.  Same for the IKEA Trådfri remote and many other switches that can be “configured” by Phoscon (it actually configures the lights to become a member of the group, not the switch).

The Xioami switches don’t send any group commands.  Instead, they send a Report Attributes command to the gateway, which causes the rule to fire, which, in turn, sends a new command to the light.  So basically they incur the ZigBee network latency twice.  Next to that, the gateway throttles commands it sends, not to flood the ZigBee network, so that could cause some more latency.

I don’t think executing the rule causes any noticeable delay.  I use a series of cascading rules (dimmer switch sets CLIPGenericStatus, CLIPGenericStatus sets CLIPGenericFlag, CLIPGenericFLag sends group action), and the third rule typically fires within 10 ms after the first.

jnmarczi, 2018-10-05:
**huge thx @esseki**, 

following your example and your description i created a low-cost push switch using a xiaomi window sensor and a regular 55mm wall push switch :-)

[image]

[image]

esseki, 2018-12-02:
Answering to myself as after reading few issues reported about Xiaomi button behavior here (https://github.com/dresden-elektronik/deconz-rest-plugin/issues/349) and here (https://github.com/dresden-elektronik/deconz-rest-plugin/issues/917), I ended up realizing that I was using a wrong event : 

```
 {  
            "address":"/sensors/18/state/buttonevent",
            "operator":"eq",
            "value":"1002"
 }
```

Indeed corresponding button events for the Xiaomi round switch (WXKG01LM) are : 
- press : 1000
- simple press release : 1002
- double press : 1004
- treble press : 1005
- quadruple press : 1006

It means that part of the latency I was reporting in this issue was not due to the rule being executed in the rule engine but due to the rule being fired on the "simple press release" (1002) event rather than on the  "press" event (1000).

I still feel a bit of latency, less honestly but as explained by @ebaauw it is due to the fact that 

> The Xioami switches don’t send any group commands. Instead, they send a Report Attributes command to the gateway, which causes the rule to fire, which, in turn, sends a new command to the light. So basically they incur the ZigBee network latency twice.

In case people might be interested, I have improved the solution described above by getting rid f the unnecessary “CLIPGenericFlag” sensor that was storing the state of the light (On or Off). As I finally found a way to use lights (in the API) as part of a condition in a rule. 

simplified version of rule to switch on the light with one click : 
```
{
    "actions": [
      {
        "address": "/groups/25/action",
        "body": {
          "on": true,
          "transitiontime": 4
        },
        "method": "PUT"
      }
    ],
    "conditions": [
      {
        "address": "/sensors/9/state/buttonevent",
        "operator": "eq",
        "value": "1002"
      },
      {
        "address": "/groups/25/state/any_on",
        "operator": "eq",
        "value": "false"
      },
      {
        "address": "/sensors/9/state/lastupdated",
        "operator": "dx"
      }
    ],
    "name": "Switch button Douche lights On"
  }
```

simplified version of rule to switch off the light with one click
```
{
    "actions": [
      {
        "address": "/groups/25/action",
        "body": {
          "on": false
        },
        "method": "PUT"
      }
    ],
    "conditions": [
      {
        "address": "/sensors/1/state/buttonevent",
        "operator": "eq",
        "value": "1000"
      },
      {
        "address": "/groups/25/state/any_on",
        "operator": "eq",
        "value": "true"
      },
      {
        "address": "/sensors/1/state/lastupdated",
        "operator": "dx"
      }
    ],
    "name": "Switch button Douche lights Off"
  }

```
I'm closing this issue as it is much more acceptable now and it is not a bug anyway.

closed 2018-12-02 by esseki