dresden-elektronik/deconz-rest-plugin #355: Hue motion sensor

frahlg, 2018-01-08:
I have two Hue Motion sensors which does not work as intended. One of them is working fine as a motion, and illuminance sensor but the temperature reading is always 0 degrees. The other one is only reporting illuminance and nothing else. Worth noting is that both of them worked flawlessly with my Hue hub.

I have deleted them from the network, re-paired them, reset them (via the deConz GUI). I am on 2.04.99 and FW 0x26190500. Any ideas?

ebaauw, 2018-01-08:
Sounds like deCONZ hasn’t yet setup attribute reporting for the corresponding clusters.  You could wait a couple of hours after pairing, or setup attribute reporting manually in the deCONZ GUI.  As an easy check, just read the corresponding clusters in the GUI, and the REST values should refresh.

I seem to have less issues with firmware 0x261A0500, see #316.

frahlg, 2018-01-08:
Thank you. I have now flashed FW 0x261A0500, nothing changed since before. I will give it some time during the night and see what happens.

frahlg, 2018-01-09:
It didn't happen anything by it self during the night. But now I have once again deleted the sensor, reset the sensor and paired it.  It seems to report Motion and Lux, but no temperature reading (not even 0 as the other one). So it all seems a bit buggy with the Philips motion sensor.

frahlg, 2018-01-09:
It took a couple of hours, but it seems that one of my sensors is working now even with temperature readings.

ebaauw, 2018-01-09:
If the other motion sensor at least updates one cluster/resource (presence, lightlevel, or temperature), deCONZ should setup the attribute reporting for the other clusters.  It only tries to do so, when it's receiving traffic from the motion sensor, so it's sure the sensor is awake.  If nothing gets updated, deCONZ might be stuck: it need a report to setup reporting, but it needs to setup reporting before it gets a report.  Rather than re-pairing the sensor, you might try and get out of this catch 22 by just reading the attributes in the _Cluster Info_ panel in the deCONZ GUI.  The "light" on the node in the deCONZ GUI should blink blue, when deCONZ receives messages from the sensor.

frahlg, 2018-01-17:
My Hue motion sensor has still not started to function properly, even after a reset and re-pair, I have honestly not put that much time into it since last week. I'll hope that some upcoming firmware will fix it, it is not a dealbreaker for me at the moment.

NathanSweet, 2018-01-17:
I have suffered with this for more than 3 weeks. It severely hampers my enjoyment of my Zigbee devices and I also hope it is being worked on.

manup, 2018-01-18:
yes, it has a high priority hope to fix it as well as the dimmer in early February.

NathanSweet, 2018-01-18:
Thanks @manup. Until then, enjoy your vacation time! :)

sevos, 2018-02-23:
any chances of getting this within few weeks? ;-)

manup, 2018-02-23:
Have you tried with latest version?
2.05.06 as well as firmware version 0x261e0500 should help to prevent the outtakes.

ebaauw, 2018-11-23:
I think this has been solved?  Especially with the recent versions, the setting up of the binding and attribute reporting has improved significantly.

manup, 2018-11-23:
Yes the setup process should be much more stable now, only keeping the sensor stable has room for improvements but that's another issue.

closed 2019-03-30 by stale[bot]

Raniz85, 2019-09-18:
I've got the same issue.

I got the motion sensor yesterday. It reports motion like it should, but neither temperature nor lux is reported.

Manually triggering a read from the UI gets a value (that shows up in the Phoscon web-app) but doesn't start automatic reporting - a configuration read says 65535 on min/max/change.

I've tried to manually enable reporting on lux by setting min/max/change to 1/10/1 and I can read back those values again if I trigger a configuration read, but they don't seem to have any effect because it's still not reporting anything (still gives me values if I manually trigger a read though).

I've let it sit overnight but there's no change, I get back 1/10/1 if I read the configuration of lux and temperature still says 65535 on all three.

The motion sensor is on 6.1.0.18912 and Deconz on 2.05.69 running on a Rpi 4 with a Conbee II.

Raniz85, 2019-09-18:
Never mind, I forgot about the binding. After configuring reporting _and_ binding it's now reporting both lux and temperature.

Pirol62, 2019-09-23:
> Never mind, I forgot about the binding. After configuring reporting _and_ binding it's now reporting both lux and temperature.

How did you configure the reporting?

Raniz85, 2019-09-26:
Follow the instructions in the post ebaauw linked above