dresden-elektronik/deconz-rest-plugin #357: Setting colors sometimes flashes the old color for a split second

Hypfer, 2018-01-10:
Sometimes, by sending a state similar to the following one, the lamp lights up in its old color for a split second:
```
{
      xy: [x,y],
      transitiontime : 0,
      bri: 254,
      on: true
} 
```
Before sending said state the lamp is turned off completely.

Tested with:
- IKEA TRADFRI bulb E27 CWS opal 600lm
- dresden elektronik FLS-PP3

DeConz version 2.04.99, raspbee, rpi3

ebaauw, 2018-01-10:
deCONZ, or any other gateway/bridge sends three different ZigBee messages to the light, one for `on`, one for `bri`, and one for `xy`.  Maybe `on` and `bri` could be or are combined in one message, but `xy` cannot.  In case of a hiccup in the ZigBee network (or in deCONZ) the light has just enough time to turn on, before receiving the command to set colour.  The only way to prevent this from happening is to create a scene with this state.  Recalli g a scene is a single ZigBee message.

Hypfer, 2018-01-10:
Would it be possible to send the `xy` zigbee message before the `on` message?

ebaauw, 2018-01-10:
Theoretically, yes, but it won't do you any good, as the light would ignore it when it's off.

closed 2019-03-23 by Hypfer