dresden-elektronik/deconz-rest-plugin #358: Raspbee no longer discovering new devices

squirrel289, 2018-01-11:
Running 2.06.99 on my raspbian rpi3 and I accidentally deleted the pi user while it was still expected by deconz.  Of course all of my devices and settings disappeared but, before I figured out what had caused the problem, I reset the osram lights and tried to reconnect them.  Unfortunately, I cannot seem to get it to discover them anymore.  I've tried restarting the service, rebooting the pi, completely uninstalling deconz, deconz and all dependencies and deleting every file and folder with the word deconz in it rebooting and reinstalling (and updating the daemon user), and flashing a few different versions of firmware.

I've had no luck and raspbee is a key part of my lighting.

manup, 2018-02-08:
If it still isn't working, can you provide the deCONZ network settings?
The lights - network association is stored in the lights and RaspBee itself so normally there is no need to reset them.

ebaauw, 2018-02-08:
Did you power-cycle the lights after opening the network?  They OSRAM lights are only discoverable for a limited time.  Same for Hue with the latest firmware.

squirrel289, 2018-02-08:
Thank you both for getting back to me.
@manup, I'll send the settings tonight when I get home.  Are you just looking for the channel, etc?  Is there a command I can run to pull all of the details you're looking for or just pull it from the web ui?  I should mention that my pi is headless and I have no window manager installed so I'm not able to use the desktop app.

@ebaauw, I tried every combination of power cycling I could think of: immediately after opening the network, immediately before# long before, 30s after...  I even disconnected all lights and repeated the above one at a time and with different bulbs.

squirrel289, 2018-02-11:
@manup,
I'm not sure this is what you're looking for, but my network subs are as follows:
Channel: 0
Network open duration: 1 min

Place key me knew if you were looking for something else and I will get it over to you.

squirrel289, 2018-03-02:
Does anyone have any suggestions?  I’m now 2+ months without being able to use any of my 22 zigbee lights.

What I’ve tried:
Rebooting, 
Resetting my lights (OSRAM)
Resetting the Raspbee:  `sudo GCFFlasher_internal -r`,
Resetting through the Web UI: Settings->Advanced->Reset Gateway,
Flashing several different versions of the firmware,
Fully uninstalling and reinstalling deconz: `sudo dpkg -r deconz; sudo apt autoremove;’

I’ve done all of the above multiple times and in multiple different combinations and still had no luck joining even a single light.  I’m not sure if it is relevant, but all of these lights were previously linked to this same gateway/raspberry pi.

Since I’m running headless, the only network settings I can find anywhere are in the Web UI under Advanced (Channel: 11, Network ID: 0xe817).  Is there a headless way to check or reset the TC Link Key, Security Mode, MAC Address, etc as mentioned in issue [#280]([https://github.com/dresden-elektronik/deconz-rest-plugin/issues/280)?

I’ll take any suggestions that will move me towards resolution.  At this point I have a $90 brick for a gateway, >$400 in useless lights and a lot of dark rooms.

manup, 2018-03-02:
To solve the problem I would suggest to temporary use the gui version so that the settings can be checked and corrected in the deCONZ ui.

For that please make a backup of your sd-card and load the latest image from here:

https://www.dresden-elektronik.de/funktechnik/solutions/wireless-light-control/raspbee-gw-sd-card-image/?L=1

Since the important network settings are stored on the RaspBee, after fixing stuff the prior sd-card image can be restored from backup.

squirrel289, 2018-03-23:
I think I've gotten to the source of this.  For security, I run deCONZ under it's own user (deconz).  After some poking around I found that the working directory was still chown'd by pi:pi so deCONZ couldn't write to the files.  I've managed to pair the bulbs but the web UI constantly tells me that an update is available, no matter how many times I click "update" or manually update the firmware.  Any chance this could be related?  Is there a list of all files that need to be owned by the running process's user I can use to make sure I've gotten them all?

SwoopX, 2018-07-11:
Hi there,

just have pretty much the same "issue". During raspi setup, I always remove the default user pi. Then I installed deconz with a different user (headless on my zero) and I wasn't able to discover any devices... Took me some hours to recognize that.

After some investigation, I noticed that some deconz bash scripts statically require user ID 1000, which is (or in my case was) user pi. Also, one of the services tries to run as pi. I'll try to check that out within the next days, maybe I can get it to work. However, I didn't catch the dir ownership yet and I'm curious if that does the trick. Nice one.

Cheers!

manup, 2018-07-12:
There are a few changes in the latest version which relate to this issue.

https://github.com/dresden-elektronik/deconz-rest-plugin/releases/tag/V2_05_32

To user other user id than 1000 you can use know a overwrite service in /etc/systemd/system to specify that.

closed 2018-12-10 by stale[bot]