dresden-elektronik/deconz-rest-plugin #361: Possible to trigger OTA without GUI?

fiskfan, 2018-01-13:
I've been trying to find information or guides related to performing OTA updates for my trådfri devices.

I run HASSBIAN on my raspberry and GUI is not installed by default on hassbian.

Is it possible to perform OTA thru commands instead of GUI? I have downloaded the firmware with the  ikea-ota-download.py script and rebooted the raspberry. I have also enabled "Otau server" thru the deConz  settings page

ebaauw, 2018-01-13:
The lights should update automatically, the updates for the motion sensor, remote, and dimmer need to be started manually.  Not sure if the OTAU plugin works without the GUI, but the REST API plugin does (obviously).

fiskfan, 2018-01-13:
Yeah, lights was updated automatically. No issues with those. But I can't find anything in the REST documentation on how to trigger it manually thru the REST interface for the devices which needs it... 

Any suggestions?

ebaauw, 2018-01-13:
None, sorry.  The REST API plugin and the OTAU plugin are two independent things.  OTAU isn't exposed over the REST API.

You might get an extra µSD card, put the default deCONZ image on it, configure the same ZigBee network settings (if they aren't already saved on the RaspBee's persistent memory), do the update, and put back the HASSBIAN card.

fiskfan, 2018-01-14:
Running a parallell on another SD card is what i will do. Thanks for the tip.

However, the IKEA Remote does not seem to popup in the OTAU interface, only the IKEA Wireless Dimmer remote popups. I can see the Ikea remote in the deconz GUI, but it is not but it is not visible in the OTAU plugin. - Any ideas on why?

ebaauw, 2018-01-14:
Has the remote been fully paired?  Are all clusters visible in the deCONZ GUI, in particular the OTAU client cluster (0x0019) ?  You might try and read the attributes for this cluster, while pressing the remote's button to keep it awake.

The devices might need to send a firmware check request (or something like that) to the gateway before they can be updated.  This might take a day or so.  You might try and speed this up by taking out and re-inserting the battery - typically such request is sent when the device starts up.

fiskfan, 2018-01-15:
You were on to something when you asked if it was funny paired. I deleted it thru the deconz fui and then added it again. After that more properties appeared and OTA worked well.

I also had some issues with the battery in the remote. The OTA ran to 1 or 10 percent and then stopped while the led was blinking. After lot of retries I swapped the battery even though I only had the remote for a week and after the battery swap OTA completed on first try.

Thanks for your help

closed 2018-01-15 by fiskfan