dresden-elektronik/deconz-rest-plugin #362: Xiaomi Cube Rotation only gives values of 0

ldplusse, 2018-01-14:
I have 3 Xiaomi Cubes and they all update the Rotation status with a value of 0.
The flipping/sides is working as it should.

Is on DeCONZ .99 using newest HomeSeer with the Jowihue plugin.
Other HomeSeer/Jowihue users report that the rotation is working.

Jowihue plugin author reports that it only receives the '0' value on my installation in the logs and that is also what i see in HomeSeer status. 

How to troubleshoot this?

ebaauw, 2018-01-14:
Let's first check that the cube notifies deCONZ when it rotates.  In the deCONZ GUI, open the _Cluster Info_ panel for the _Analog Input (Basic)_ cluster (0x000C) of the cube's third endpoint (0x03).  Check that the _Present value_ attribute (0x0055) changes value when you rotate the cube.  Right turns should result in positive values, left turns in negative values.

If not, setup a binding between this cluster and the RaspBee's HA endpoint.  Open the _Bind Dropbox_ panel, drag the 0x000C cluster to the _Source_, drag the RaspBee's HA endpoint to the _Destination_ (make sure _Type_ is IEEE), and press _Bind_.

ldplusse, 2018-01-14:
Thanks for the answer.
I use Conbee. I tryed your procedure, but cannot get it to bind. i just get a timeout. I have tryed to move, shake, rotate the device while binding and have it close to the ConBee, it just says 'Timeout.
Cluster Info:
[image]
Cube to Source:
[image]
ConBee to Destination:
[image]
Binding:
[image]

Not sure if i am doing this correct. Its pretty confusing stuff in DeCONZ.

ldplusse, 2018-01-14:
I got it Bind.
I pushed the button inside right after pushing bind.
Put it still only gives 0 in output.

ebaauw, 2018-01-14:
That looks right to me, but you never know with Xiaomi.  I take it, the _Multistate input_ cluster does update when you shake, flip, push the cube?  

> In HS3 i can see it updates with the value of 0.

I don't know HS3, nor where it gets the updates from.  Would be interesting to know whether the cube actually sends, and deCONZ actually receives updates, but always with value 0, or whether the updates don't reach deCONZ.  Does the deCONZ REST API send websocket notifications (with `buttonevent` value 0)?

ldplusse, 2018-01-14:
Yes the Multistate input is working.

When rotation it does send buttonevent value 0 trough the Rest API.
Other users of HomeSeer report that the Cube is working with DeCONZ and rotation.

ebaauw, 2018-01-16:
I fear they might have changed the hardware and/or firmware.  My cube has a fourth attribute, 0xFF05, in the _Analog Input_ cluster.  Find the other differences (you might need to read each attribute of the _Basic_ cluster individually, while pressing the "inside" button - this might take a couple of attempts):
[image]
[image]
[image]

superwerder, 2018-01-16:
OK, so after a lot of trying I found a way to get the values and they are identical to yours. Only thing is that 0x0012 and 13 are not "grey" in my setup.
And yeah, I also don't have the 0xFF05 attribute in the analog input cluster. Why is it called unknown?

ebaauw, 2018-01-16:
Because it's manufacturer specific, not defined by the ZigBee Cluster Library.  I haven't been able to figure out what it's for.  Sometimes it changes value when I rotate the cube, sometimes it doesn't.  Haven't been able to detect a pattern in the values either.

superwerder, 2018-01-17:
Ok. But as it changes it seems to be important. But why don't we have it. Could it be something with using conbee and not raspbee?

ldplusse, 2018-01-17:
As superwerder commented.
We are both on the ConBee and i believe Erik is on the RaspBee. Is that right Erik?

We have another Cube user here in denmark confirmed working on the RaspBee.
I am hoping i can get my Cube tested, on his setup.

ldplusse, 2018-01-17:
I checked my setup and i have the same values but also the two that is not gray and also missing the 0xFF05.
[image]
[image]
[image]

ebaauw, 2018-01-17:
I am on RaspBee, indeed, but I don't see how RaspBee vs ConBee could make a difference.  You might check the ConBee/RaspBee firmware, I'm on 0x261A0500, but, really that's a long shot.

> We have another Cube user here in denmark confirmed working on the RaspBee.
I am hoping i can get my Cube tested, on his setup.

If that would work, it would be interesting to know whether his cube works on your setup.

I have a ConBee lying around, used for ZigBee sniffing.  I could hook it up to my test Raspberry and pair my cube (I only have one) to it.  That will have to wait for the weekend, though - it's rather elaborate.

ldplusse, 2018-01-17:
I have just arranged the test next tueday.
There is hardware change between RaspBee and ConBee and also the change in the operation system.

Not sure he is willing to lend me his Cube :-)
But is my Cube is working on his RaspBee, i dont see that a test with his Cube on my setup is needed.

ldplusse, 2018-01-23:
i have an update. Today i tried one of my Cube's at anothers place.
When my Cube was connected to his RaspBee, it worked immediately with rotation.
I also had a ConBee, so we moved the Cube to his Conbee. There the rotation did not work. Only the multistate input was updating with values. It worked in the exact same way as on my ConBee.

So think we can deffently say that there is some different between the ConBee and the RaspBee.
How can i proceed finding the problem.

manup, 2018-01-25:
There is no difference in hardware or software, I figure tge ConBee might just not have the latest firnware flashed?

ldplusse, 2018-01-25:
DeCONZ does not say i have a firmware update available.
[image]

superwerder, 2018-01-25:
I have the same versions as Idplusse.

ldplusse, 2018-01-28:
I have checked my FTDI version and it is the newest version 2.12.28
I cannot see any 'manuel' firmware update. So assuming that my ConBee firmware is also the newest available.
How can we proceed with further troubleshooting.

ebaauw, 2018-01-28:
The latest firmware is 0x261A0500, as I mentioned above.  deCONZ doesn’t prompt to install this version, you need to install it manually.

ldplusse, 2018-01-28:
Could not find any firmware besides the link.
The reference in the #230 link is not for Windows.

I need info on how to update the ConBee firmware from a windows PC.
All this github is not easy to understand for a windows guy.

And as i have seen DeCONZ auto ask me to update firmware, i dont understand why some firmware versions does not.

ebaauw, 2018-01-29:
You’re on Windows!  That would have been worth mentioning.  Can some-one please double-check that the Windows distribution contains the latest `general.xml` file, which includes the _Analog Input_ cluster?  See https://github.com/dresden-elektronik/deconz-rest-plugin/issues/138#issuecomment-324911430.

manup, 2018-01-29:
>You’re on Windows! That would have been worth mentioning. Can some-one please double-check that the Windows distribution contains the latest general.xml file, which includes the Analog Input cluster?

You're right it's missing, will be updated in the next release.

ldplusse, 2018-01-29:
Great findings guys.
Sorry for not providing the 'Windows' information.

Looking forward to the next release.
Any ETA on that.

manup, 2018-01-30:
New version for all platforms should arrive this week.

manup, 2018-02-01:
Hi, version 2.05.00 for Windows and Ubuntu is online:
 
https://www.dresden-elektronik.de/deconz/

ldplusse, 2018-02-01:
Perfekt.
And it works :-)

Thanks for the fix.

closed 2018-02-01 by ldplusse