dresden-elektronik/deconz-rest-plugin #364: Tradfri Wireless Dimmer not working

doodieb15, 2018-01-15:
Hi,

I am running deCONZ 2.04.99 on an Raspberry Pi3 Raspbian Stretch with 2 Tradfri bulbs (TRADFRI bulb E27 opal 1000lm and TRADFRI bulb E27 WS opal 980lm).

The Tradfri wireless dimmer can be connected to deCONZ but it will not dim any of the lamps. The button Events come through but the bulbs will not dim. If I pair the dimmer directly to the lamp the dimmer will work.

If have upgraded all firmwares (of the bulb and the dimmer)

I also have a Tradfri remote Control. It will dim the lights but can not Change the light Color of the TRADFRI bulb E27 WS opal 980lm. Also the button events come through (cooler, warmer) but no effect.

Is there anything else I can try or are these functions not yet implemented in deCONZ?

xibriz, 2018-01-21:
Same issue here.

Martin-Gardner, 2018-01-25:
My TRADFRI Remote works fine, dimming and colour temp works fine. My three Wireless Dimmers, do nothing. I can seem them joined in deCONZ and have assigned lights using the Phoscon HTML 5 app. I can see updates getting sent from the devices in deCONZ. But nothing happens to the lights.

Are these Dimmers beta supported?

Many thanks,

Martin

Martin-Gardner, 2018-01-26:
I managed to get a repeatable method of getting the IKEA Tradfri wireless dimmer to work. I removed the object from deCONZ by selecting and choosing remove node from the menu at the top of the GUI. I then opened the standard webapp and went to add new switches, I choose the Philips hue dimmer. I then presses the button on the back on the dimmer five times to reset it. This then showed up in deCONZ. I then opened Phoscon HTML 5 app and added the lights to the switch, it shows up as a square in Phoscon - this should be changed to a circle but I digress - I then waited for the red light to stop. After this I pressed the button once. After this process I was able to control a group of 7 lights. Its a bit iffy, all the lights are sometimes not in sync, but I can live with that for a while. I did get issues when binding two wireless dimmers to the same set of lights as well. 

Hope this helps,

Martin

doodieb15, 2018-01-26:
How do you pair the Tradfri remote control to get the color temp to work? Thanks for the info on the wireless dimmers, will try this Weekend.

Martin-Gardner, 2018-01-26:
Same steps to remove, I added via the Phoscon web app following the instructions provided. Open PWA app, go to add switch select IKEA, click on the picture, reset, wait for bind, it tells you when its bound. Then its says to push and hold the off button for 10 seconds. After this I was able to change the colour temp of all the bulbs in the house at once (I use mine as a master dimmer for the house, comes in very handy when putting the baby to sleep).

Hope that helps,

Martin-Gardner, 2018-01-26:
Let me know if the above works for you as well.

doodieb15, 2018-01-27:
Works for me too :), the dimming experience of the Wireless Dimmer is not too good however (seems to dim in steps only).

However much better than bevore (no function).

Thanks Martin

closed 2018-01-27 by doodieb15