dresden-elektronik/deconz-rest-plugin #366: Powerplug shows as light

andreasjoner, 2018-01-16:
Hi,
I have connected a Zigbee power-plug from HomeControl (https://www.elektroimportoren.no/home-control-stikkontakt-dobbel-m-jord-hcoutlete1/4512603/Product.html?Event=searchlist) and it shows up in "lights" not "Sensors". Is it possible to somehow "move" it.....

looks like this:
    "4": {
        "etag": "081218aae367d8786d9b8bbb75bde031",
        "hascolor": false,
        "manufacturername": "Unknown",
        "modelid": "HCOHCD",
        "name": "power-plug1",
        "state": {
            "alert": "none",
            "on": true,
            "reachable": true
        },
        "swversion": "",
        "type": "On/Off output",
        "uniqueid": "00:0d:6f:00:04:16:89:7c-01"

ebaauw, 2018-01-16:
No, see https://github.com/dresden-elektronik/deconz-rest-plugin/issues/153#issuecomment-326418124.

manup, 2018-02-08:
Technically it's a actor not a sensor so light might be a bit misleading but sensor would be the wrong place to put the a plug.

Closing the issue for now.

closed 2018-02-08 by manup