dresden-elektronik/deconz-rest-plugin #368: Problems configuring Busch & Jäger switches

Daniel0984, 2018-01-17:
Hi all,

i recently (tried) to move from a Hue hub to deconz/conbee as network controller.
My network contains 24 Busch&Jäger Zigbee switches (all mains powered: some relais, some control only)

The behavior is very diffrent from what it was used to be with the Hue Hub. In the Hue network, i needed to program each switch by setting up the light state(s) and hold the button on the switch to assign the lamps to it.

With deconz this does no longer work, programming the switch itself does nothing (totally fine, if i could setup everything in software).

But the software (i used Phoscon) has some flaws:

* Every switch automatically get a group created while joining. If i assing lights to this group, all buttons on the switch toggle these lights - even the primary button on a relais swich which are supposed to only toggle the physical connected light.

* Phoscon does not allow asinging the primary button for control only switches (for relais switches this is fine)

* As lights need to be in the group for configuring scenes - the primary button still just turns off/on all lights

* the API lists every physical switch as multiple ZHASwitches:

```
  "1": {
    "config": {
      "group": "1",
      "on": true,
      "reachable": true
    },
    "ep": 10,
    "etag": "ced8f8aff1d18b6dde24ab483cf9c2dc",
    "manufacturername": "Busch-Jaeger",
    "mode": 4,
    "modelid": "RM01",
    "name": "Zimmer A",
    "state": {},
    "swversion": "1.2.0",
    "type": "ZHASwitch",
    "uniqueid": "d8:5d:ef:11:a1:00:19:6c-0a-1000"
  },
  "2": {
    "config": {
      "group": "1",
      "on": true,
      "reachable": true
    },
    "ep": 11,
    "etag": "5ba243f4c8c7d4b2824db5dbb0be6df5",
    "manufacturername": "Busch-Jaeger",
    "mode": 4,
    "modelid": "RM01",
    "name": "Zimmer A",
    "state": {
      "buttonevent": 3002,
      "lastupdated": "2018-01-16T20:49:30"
    },
    "swversion": "1.2.0",
    "type": "ZHASwitch",
    "uniqueid": "d8:5d:ef:11:a1:00:19:6c-0b-1000"
  }
```
My assumption is, the group mentioned above gets switched. So i maybe could remove the group config from the primary button.

* mode parameter is explained in the [docs ](https://dresden-elektronik.github.io/deconz-rest-doc/sensors/) as "Only available for dresden elektroink Lighting Switch", "Number (1|2|3)". Some of mine are mode 4, [code ](https://github.com/dresden-elektronik/deconz-rest-plugin/blob/master/sensor.h) says thats valid, the docs need an update

* Update of the mode setting via API is only valid if  (pi.key() == "mode" && sensor->modelId() == "Lighting Switch") 

Is there anything i can do to sort out my light config?

Daniel0984, 2018-01-19:
Little update: Removing the group from the sensor config did not work.

Cleaning up the B&J button config in sensors.cpp (which is a mess) did also not work (i just set every button to activate secene)

So its maybe the inital configuration of the device.
For now i am unfortunatly out of ideas and i'll re-setup the Hue network.

manup, 2018-02-08:
Hi, documentation is scheduled to be updated with device specific descriptions.

Currently we only support a fixed configuration: When a BJG switch is added the top buttons will be bound to control on/off/pushdimm and the remaining lower buttons control scenes. All endpoints do control the same group.

It's different from the manual BJG way of configuring the switch.

To edit the scenes you need to use the scene editor in Phoscon App (currently tablet or PC only).
No need to edit the cpp files.

There is a new switch editor in the works which does allow assigning of custom actions to buttons, no ETA yet but it should be available in the next months.

Daniel0984, 2018-02-08:
Hi,

Thank you for your anwser.
The first or top 2 buttons are always bound to the physical light **if the unit is a relay**, otherwise they should be programmable.
There is a check in bindings.cpp line 1244+, i can confirm this works as intented.
However, the Phoscon App  does not handle this seperatly and you can't programm the first/top buttons.

For the scenes, maybe its a user / configuration error i did. As an example the following setup:

Relais: http://bit.ly/2C33pFp
Switch: http://bit.ly/2nTs2iO
4 Lights added to the Group with Phoscon.

first/top controlls the wired light **and** every light in the group

Then i created scenes and assigned them to different buttons.
But every button except the first/top will switch all the lights in the group.

I hacked in support for a different group per button, a solution i can now work with.
Anyway if there are upcoming changes that will improve the situation i'll be happy to test them.

closed 2018-10-08 by Daniel0984