dresden-elektronik/deconz-rest-plugin #369: linking xiaomi wireless switch

Try2Fly, 2018-01-17:
I'm having trouble linking the Xiaomi wireless switch to deconz (2.04.99).

I've installed deconz headless, and accessing it through the Phoscon App.

First I linked a Ikea lamp and xiaomi motion detector with no problem.

Then I wanted to link a xiaomi wireless switch (which is already linked to the xiaomi gateway). I understood I have to add it as a sensor. So when selecting add new sensor --> other I get the Add sensor view. Then I reset the xiaomi wireless switch and I get a message that the sensor is ready. But after clicking Ready the new sensor is not listed as connected sensor.

Try2Fly, 2018-01-17:
I think it is the same issue as #336

closed 2018-01-17 by Try2Fly

ebaauw, 2018-01-17:
In that case you should see the wireless switch through the REST API and get web socket notifications when pressing the button on the switch.