dresden-elektronik/deconz-rest-plugin #370: deconz update process calling google every ten seconds

GatoPharaoh, 2018-01-17:
I have a Pi-Hole running on my rapsberry - the same raspberry that runs deconz.
A few days ago I noticed a spike in requests to google.de - with two requests (IPv4 and IPv6) being made every ten seconds.
After some digging I killed the deconz-update process - which stopped the requests. Nevertheless, after every reboot the process comes back on - so I have to kill it every time.

Is this expected behavior?
thanks in advance

GatoPharaoh, 2018-02-03:
btw for those having the same problem - disable the update service with `sudo systemctl disable deconz-update.service` until we get some feedback from the devs.

manup, 2018-02-05:
The service is used to query for updates and also provide the discovery service. In the API this can be configured via `/config/discovery` attribute. 

http://dresden-elektronik.github.io/deconz-rest-doc/configuration/#modifyconfiguration

The above service from the beta doesn't respect this attribute currently, it will be added in the next releases as well as a configuration option in the Phoscon App.

closed 2018-03-07 by GatoPharaoh