dresden-elektronik/deconz-rest-plugin #371: deCONZ backup/restore issues

ebaauw, 2018-01-19:
After several notes to self, I finally came around doing a backup/restore test of deCONZ, see https://github.com/dresden-elektronik/deconz-rest-plugin/issues/247#issuecomment-358939815:

> I did and export on my production Raspberry and stopped deCONZ.  Then, I did an import on my test Raspberry.  After the import, deCONZ restarted (or crashed - I'm not sure) and came back up with the test RaspBee now showing the production RaspBee's MAC address.  The GUI is scaringly empty - it only shows the RaspBee.  This is probably to be expected, as the RapsBee's neighbour table would be empty.  I power-cycled some of my lights, and, indeed, they appear and respond.  Of course power-cycling the light would also clear it's neighbour table as well, so only the power-cycled lights appear.  After a good many minutes, some other lights begin to appear in the GUI.  However, they're not responding, and turn red.  My sensors and switches blink red (no ack from the new RaspBee), but continue to work directly on the lights that are not yet reachable.  I'll try and stretch my patience to the max and leave it for now, but I fear you might need to power-cycle all your devices after a restore.  @manup, do you have any insights here?

> I did notice that, after import, the _NWK Update ID_ is 5.  In the `deCONZ.conf` of the backup, `nwkUpdateId` is 0.

After over half an hour no change.  I restored the backup from the test RaspBee.  This seemed to go without issues, the lights previously connected to test were found without a problem and are reachable.  The test RaspBee again has its original MAC address.  deCONZ did crash (deliberately?):
```
Jan 19 12:48:47 pi systemd[1]: deconz-gui.service: Main process exited, code=exited, status=41/n/a
```

There seems something broken in the restore logic.  After untar-ing the backup file, before the crash, deCONZ saves it's database and layout when resetting or leaving/joining the network.  This causes the restored database to contain records from the running database, before restore.

ebaauw, 2018-01-19:
Restarted deCONZ on my production Raspberry.  As expected, the GUI is populated straight away, with the exception of the lights that I had power-cycled to connect to the restored backup on the test Raspberry.  The _NWK Update ID_ is 0, as in the backup.  The missing lights won't come back after a power-cycle.

ebaauw, 2018-01-19:
OK take 2.  To be sure, new backup (this time from the deCONZ GUI), and new restore (also from the deCONZ GUI).  The GUI must do something more than POST `/config/import`, because deCONZ now seems to restart more smoothly.  Again, restored configuration is on channel 11 and _NWK Update ID_ is now 7 (so increased from the 6 in the backup file).  Changed channel to 25 and _NWK Update ID_ to 8, and left/joined the network.  All my lights are found.  End devices appear as well, but remain unreachable.

marthoc, 2018-01-24:
@ebauuw... not sure if this is related but do you know if you can move the RaspBee between Pi devices easily? I’m guessing that some of the issues you’ve encountered wouldn’t happen since the RaspBee would be the same between devices.

ebaauw, 2018-01-24:
Yes, of course, you can move the RaspBee and the µSD card to another Raspberry and everything runs smoothly.  That won't help you if the RaspBee breaks, though.

marthoc, 2018-01-24:
@ebaauw, yes agreed. Just wanted to make sure I understood what you were saying about backup/restore.

andrejem, 2018-03-15:
Just my 2 cents. Used the Backup funktcionality today to move deconz to a pi running my fhem instance (used 2 before). Old pi had 2.04.97. New one directly installed with 2.05.12. Backuped via pwa app and restored to the new one via pwa as well. Worked great. Only thing which i lost was the cooenction between lights and groups. For me not a big problem because I only have 1 light. All sensors and switches work great after the move. 
Using the conbee USB Stick. I have no rules or something else. Only using it for pushing sensor events and switch events to handle them in fhem.

DOliana, 2018-12-22:
I tried today and failed... I wanted to move from deConz 2.0.50 on raspbian jessie to a freshly installed raspbian light stretch running deConz 2.0.52. I had the following results:

- backup/restore through phoscon web-app did not throw any errors
- all items show in Phoscon UI
- all items are visible through REST-API
- any command to an item returns success, but no physical results happen (lights do not turn on)

I looked at the logs but could not find any errors or warnings. Any ideas where I could look?

manup, 2018-12-22:
Can you please create a second backup with the current installation and send both, the original and the new backup, to mpi@dresden-elektronik.de 

I can check if something is wrong with configuration.

Meanwhile please don't change or reset anything.

manup, 2018-12-22:
Also have you configured the serial interface on the new system as described here:

https://github.com/dresden-elektronik/deconz-rest-plugin#software-requirements

Otherwise it won't work.

DOliana, 2018-12-23:
Will do as soon as I get home. Just one question: I configured the serial interface, but wondered whether it is required since I am using ConBee and not RaspBee (so USB not gpio). The same goes for disabling Bluetooth.

DOliana, 2018-12-27:
@manup Interestingly after using the latest version it works. Thanks! (If you want, I still have the two backup files that I could send over)

closed 2019-05-03 by stale[bot]