dresden-elektronik/deconz-rest-plugin #374: Osram lightify switch support

scan5415, 2018-01-20:
Hi all

I would like to add an OSRAM Lightify Switch to deConz Rest API. 
What's about to support this device. I know on the webpage is currently not supported. Are there any plans?

Or is it possible to add the devices manually by me self?

Thanks for helping.

FlippAre, 2018-01-23:
Using mine with a few OSRAM Lightify Switches without any issues

scan5415, 2018-01-23:
And how did add the switch to deConz.
The "old" Webinterface do not find my Switch and on the "new" Webinterface is no button for OSRAM.

What did you exactly to add it on your installation?

FlippAre, 2018-01-24:
Opened the network. Factory reset the plug and turned it on and off. Running the latest beta

odgregg, 2018-05-23:
Perhaps it's the OSRAM Lightify wireless dimmer that this is referring to?

https://www.amazon.com/SYLVANIA-LIGHTIFY-Wireless-SmartThings-Assistant/dp/B0196M620Y

I have two of these and can't get them to show up in the REST API or web interface. Would be great if support can be added.

danielsjf, 2018-09-28:
I have the osram plug smart +. It works with the conbee and is recognised in home assistant. For some reason deconz reports it as a light though.

fr34kyn01535, 2018-10-19:
Is there something we can do to have it in the switch category instead light?

kevinclown, 2018-11-23:
Guys,
I feel you talk about two different devices from Osram: Lightify Switch (two double buttons) and Smart+ Plug (a power socket) or similar.
I have joined this thread in the hope, Dresden Elektronik could add the Lightify Switch to the recognized devices. As of today the software does not recognize it at all. (the power sockets work fine)
Pressing the buttons, the following lines are visible in the log:

top left:
APS-DATA.indication srcAddr: 0x000d6f000ec9XXXX, srcEp: 0x01 dstAddrMode: 2, profile: 0x0104, cluster: 0x0006, lqi: 255, rssi: -54

bottom left:
APS-DATA.indication srcAddr: 0x000d6f000ec9XXXX, srcEp: 0x03 dstAddrMode: 2, profile: 0x0104, cluster: 0x0006, lqi: 255, rssi: -52

top right:
APS-DATA.indication srcAddr: 0x000d6f000ec9XXXX, srcEp: 0x02 dstAddrMode: 2, profile: 0x0104, cluster: 0x0300, lqi: 255, rssi: -58

bottom right:
APS-DATA.indication srcAddr: 0x000d6f000ec9XXXX, srcEp: 0x04 dstAddrMode: 2, profile: 0x0104, cluster: 0x0300, lqi: 223, rssi: -63

The screenshots from deCONZ are attached. The device shows up with 6 endpoints.
[image]
[image]
[image]
[image]
[image]

The endpoints 2 till 6 seem to have identical basic data:

[image]

If more data/info is needed, please let me know.

Thank you in advance and best regards
K.

manup, 2018-11-24:
I've made some tests with the switch a few weeks ago, it supports bindings to groups on each endpoint which we can use to further translate incoming commands into button events for press, hold and release.

No ETA yet but the switch is on the todo list.

giammaleoni, 2018-12-11:
Hi @ErnstEeldert,

> I have two of these (the newer variant called OSRAM Smart+ Plug), and I have been able to pair and use them successfully.
> 
> However, I wonder why they're put into the 'light' category in the REST API, and not as a sensor of type ZHASwitch? Because of this, they show up as lights in Home Assistant, and homebridge as well, which somewhat works, but unnecessarily exposes things like brightness etc.

I have same plugs but I can't figure out to connect them to deConz. I'm using RaspBee as gateway and it is correctly displayed and connected to deConz (I can see ZigBee channel and Network id in Phoscon WebApp).

It is the first time I use them (never connected to any other gateway), already tried a factory reset keeping presses the on/off button.

Thanks for your help
Gianmaria

semaf, 2018-12-28:
> Hi @ErnstEeldert,
>
>
> I have same plugs but I can't figure out to connect them to deConz. I'm using RaspBee as gateway and it is correctly displayed and connected to deConz (I can see ZigBee channel [...]

How could you connect it? As the user manual says "Press right top + left bottom same time for 3 seconds" and / or "Press left top + right bottom same time for 10 seconds" to reset the switch. 
http://www.produktinfo.conrad.com/datenblaetter/1300000-1399999/001387115-an-01-de-LIGHTIFY_SWITCH.pdf

Phoscon can not find the switch after the reset above. Anybody idea how to do it?

derBroBro, 2019-02-20:
Hi,

I tried also to get the device paired to deconz via the webui and failed. Any hints how to solve?
By the way: Without be paired the switch can turn on/off all my devices. How does this work?

Malte

giammaleoni, 2019-02-21:
> Hi @ErnstEeldert,
>
>
> I have same plugs but I can't figure out to connect them to deConz. I'm using RaspBee as gateway and it is correctly displayed and connected to deConz (I can see ZigBee channel [...]

UPDATE: problem was in the Raspbee firmware, updating it following step-by-step guide ( [https://github.com/marthoc/hassio-addons/blob/master/deconz/FIRMWARE-UPGRADE.md](https://github.com/marthoc/hassio-addons/blob/master/deconz/FIRMWARE-UPGRADE.md) )  as suggested in different threads solved the problem: I can connect all devices successfully.

errrrata, 2019-03-16:
Just a +1 for 4x button switch support

olicooper, 2019-03-26:
Hi guys, is there any update on support for this? I have the "Osram Smart+ Switch Mini" which is pretty much the same as this I believe.

In the meantime is it possible to write something myself to make them work? I'm not sure how deconz works.

salopette, 2019-04-05:
Hello, is the Osram Smart Switch mini and Smart Switch 4x now supported?
I still can not connect mine.

flobiwahn, 2019-04-14:
> Hello, is the Osram Smart Switch mini and Smart Switch 4x now supported?
> I still can not connect mine.

No, still not supported. It's recognized by deconz but I can't add it to Phoscon or HomeAssistant.

derBroBro, 2019-05-18:
Updated to 2.05.64 / 26330500 but still unable to connect.

derBroBro, 2019-05-19:
In Zigbee2Mqtt they also have trouble with the integration. Maybe this helps: https://github.com/Koenkk/zigbee2mqtt/issues/962

Torschen, 2019-06-12:
I also cannot connect my Osram Smart Switch to Phoscon. I would appreciate if the switch could be added to the supported devices.

smoki3, 2019-07-07:
+1 for Osram Switch

deven2020, 2019-07-31:
I bought the Sylvania Smart Home 73743 Lightify Smart Dimming Switch today and tried to connect via conbee 2. Phoscon unable to connect to this device. I tried the switch and light search but nothing seems to work. Any help will be much appreciated.

Chris-LZ, 2019-10-13:
Hi guys,
I just ordered ConBee II and saw now that the Osram Lightify Switch doesn't seem to be supported yet. Thats bad, I got abound 28 of them in my house, and I'm really sick of the Lightify Gateway and the Osram App to maintain them....
Are there any news when / if the switches can be connected?
Or did one of you guys found out a workaround or something? 
Thanks for you help! :-)

tadeaustria, 2019-12-07:
+1 for adding Osram 4 x Button Switch

Palkers76, 2020-01-12:
+10 for adding Osram 4 x Button Switch

grayve, 2020-02-06:
Please add this really nice 4 button switch.

Chris-LZ, 2020-02-06:
Do you guys have any updates on that?

SchulzMichael, 2020-03-20:
Please add the 4 button Switch

emmix71, 2020-03-24:
Please also add the 3 button mini switch. Thank you!

Florian-Schmidt, 2020-04-14:
Hello,

also waiting for Osram 4 Button switch Support. Osrams Smart Plugs are working greate!

Chris-LZ, 2020-04-30:
Hi guys,
any news here?
I currently suffering with those crappy IKEA dimmswitches in my house, and want to replace them with the 4-Button-Lightify Switches (I got already 35 of them, just waiting for implementation :-/ )
THANKS!!

JonasKru, 2020-05-06:
I have one Osram 4 buttons switch and one of the mini version. I can connect them but they won't show up in the UI. So I can't edit any of their functions...

Would love to have them properly implemented!

bruesche, 2020-05-14:
Hi there. Facing same problem. The Osram Switch seems to be connecting. Nothing in Phoscon App. Only in the Desktop App i see a device. The switch can be used to switch on and off all devices in phoscon. Not cool. Can‘t programm any functions. When would be a new software Version vor Phoscon available (newer than 2.05.75) so that I have hope to use the switch soon?

Smanar, 2020-05-14:
I m on the 3 button (I think) ATM https://github.com/dresden-elektronik/deconz-rest-plugin/issues/2763

pollierer, 2020-05-22:
Hi! I´m also waiting for Osram 4 Button switch Support.

Smanar, 2020-05-22:
The 3 button isn't moving ATM.
If I make code modification for the 4 button, are you able to compile the code to test it, I need some info to finalise the integration.

pollierer, 2020-05-22:
> The 3 button isn't moving ATM.
> If I make code modification for the 4 button, are you able to compile the code to test it, I need some info to finalise the integration.

sorry, not without advice. Usualy I use deconz via hassio on a pi3, but I can use the ConBee II somewhere else (other raspbian, windows, ..) to test stuff. If I can help with a 4 Button switch to test let me know.

Smanar, 2020-05-22:
Ok, so you have the procedure here https://github.com/dresden-elektronik/deconz-rest-plugin at "Install deCONZ development package (optional, Linux only)"
Just replace the step 1 by
`git clone https://github.com/Smanar/deconz-rest-plugin.git`

And yes, Hassio is cute but too hard to use.
Better to use a raspian system (need Unix).
Like that, you can permute the conbee for some minute to make test, but take care if your "production system" stay too many time without the conbee some devices (Xiaomi) can leave the network.
And you will have an empty network on the "test machine".

So with this code the remote will be visible in the API, but button will not work.
just run deconz in debug mode, press a button and you can 
1 - check yourself in log string starting with "no button map" that appear when you press a button
2 - give me the complete log.

Florian-Schmidt, 2020-05-22:
@Smanar: Already watching your comments for 3 Button switch. Doing this on the same raspian would replace actual deconz installation, won't it? I'm actually using deconz in docker (marthoc).

Smanar, 2020-05-22:
Ha ok, I haven't understant that.
And no, don't touch deconz in docker, I have already tried, no a good idea.

You haven't another raspian machine ? Can you make another container just for test (I don't think it will be something easy for USB port) ? But in all situation, don't touch at your existing docker.

Florian-Schmidt, 2020-05-22:
Unfortunately it's my first and only raspian machine ATM. 

Perhaps I can invest some time but this will take some weeks because I'm busy. So I can't promise a test actually and hope that someone else is faster than me.

pollierer, 2020-05-23:
> So with this code the remote will be visible in the API, but button will not work.
> just run deconz in debug mode, press a button

I set up a new raspbian lite image, compiled your git version (master branch) and replaced the original plugin. unfortunately I can´t pair the switch via the PWA. Atm I´m not 100% sure if this is an issue with the remote switch or not. Now I will install an x server and start deconz in debug mode logging to the console directly to have a look into the gui. Anyway, not sure if this will help me with the pairing issue..

Smanar, 2020-05-23:
The device can be invisible in Phoscon but visible in the API.
In all situation it will be visible in deconz, so you can check if the model id is "Switch 4x EU-LIGHTIFY"

pollierer, 2020-05-23:
nice, it worked out now. One issue was, I didn´t see the pairing success in the web app at all, but it did work as I could see in deconz directly.
beside that in my first attempt I did start more than one instance of deCONZ so the second one wasn´t able to speak to the stick.
I hope the button actions did all register. I tried all 4 buttons with short and long press events.

btw until now I don´t know how to use the api, somehow you get a little lost in the beginning..

If I can provide you some more stuff please let me know :)

15:47:29:761 HTTP Server listen on address 0.0.0.0, port: 80, root: /usr/share/deCONZ/webapp/
15:47:29:787 CTRL. 3.27.215:47:29:954 COM: /dev/ttyAMA0 :  (0x0000/0x0000)
15:47:29:954 dev /dev/ttyAMA0
15:47:29:954 COM: /dev/ttyACM0 : ConBee II (0x1CF1/0x0030)
15:47:29:955 ZCLDB init file /home/pi/.local/share/dresden-elektronik/deCONZ/zcldb.txt
15:47:30:222 parent process bash
15:47:30:222 gw run mode: normal
15:47:30:222 GW sd-card image version file does not exist: /home/pi/.local/share/dresden-elektronik/deCONZ/gw-version
15:47:30:222 sd-card cid: 035344534331364780ffffffff01287d
15:47:30:224 DB sqlite version 3.27.2
15:47:30:226 DB PRAGMA page_count: 30
15:47:30:226 DB PRAGMA page_size: 4096
15:47:30:226 DB PRAGMA freelist_count: 0
15:47:30:226 DB file size 122880 bytes, free pages 0
15:47:30:226 DB PRAGMA user_version: 6
15:47:30:226 DB cleanup
15:47:30:226 DB create temporary views
15:47:30:227 DB view [0] created
15:47:30:228 DB view [1] created
15:47:30:228 DB view [2] created
15:47:30:228 DB view [3] created
15:47:30:228 sql exec SELECT apikey,devicetype,createdate,lastusedate,useragent FROM auth
15:47:30:230 sql exec SELECT key FROM config2
15:47:30:230 sql exec SELECT key,value FROM config2
15:47:30:230 Load config UTC: 2020-05-23T13:19:57 from db.
15:47:30:230 Load config announceinterval: 10 from db.
15:47:30:230 Load config announceurl: http://dresden-light.appspot.com/discover from db.
15:47:30:230 Load config apiversion: 2.05.76 from db.
15:47:30:230 Load config bridgeid: 00212EFFFF05AF15 from db.
15:47:30:231 Load config datastoreversion: 60 from db.
15:47:30:231 Load config dhcp: true from db.
15:47:30:231 Load config discovery: true from db.
15:47:30:231 Load config factorynew: false from db.
15:47:30:231 Load config fwneedupdate: false from db.
15:47:30:231 Load config fwupdatestate: idle from db.
15:47:30:231 Load config fwversion: 0x264a0700 from db.
15:47:30:231 Load config gateway: 127.0.0.1 from db.
15:47:30:231 Load config group0: 65520 from db.
15:47:30:231 Load config groupdelay: 50 from db.
15:47:30:231 Load config gwpassword: [redacted] from db.
15:47:30:231 Load config gwusername: delight from db.
15:47:30:231 Load config homebridge: not-managed from db.
15:47:30:231 Load config homebridge-pin:  from db.
15:47:30:231 Load config homebridgepin:  from db.
15:47:30:231 Load config homebridgeupdate: false from db.
15:47:30:231 Load config homebridgeupdateversion:  from db.
15:47:30:231 Load config homebridgeversion:  from db.
15:47:30:231 Load config ipaddress: 192.168.178.41 from db.
15:47:30:231 Load config linkbutton: false from db.
15:47:30:231 Load config localtime: 2020-05-23T14:19:57 from db.
15:47:30:231 Load config mac: 38:60:77:7c:53:18 from db.
15:47:30:231 Load config modelid: deCONZ from db.
15:47:30:231 Load config name: Phoscon-GW from db.
15:47:30:231 Load config netmask: 255.0.0.0 from db.
15:47:30:231 Load config networkopenduration: 60 from db.
15:47:30:231 Load config otauactive: false from db.
15:47:30:231 Load config otaustate: off from db.
15:47:30:231 Load config panid: 0 from db.
15:47:30:231 Load config permitjoin: 0 from db.
15:47:30:231 Load config permitjoinfull: 0 from db.
15:47:30:231 Load config port: 80 from db.
15:47:30:231 Load config portalservices: false from db.
[... 674 lines cut ...]
14:48:15:763 void deCONZ::zmNode::setFetched(deCONZ::RequestId, bool) fetched item: 8, node: 0x0000
14:48:15:826 poll node 00:21:2e:ff:ff:05:af:15-01
14:48:15:826 Poll light node Configuration tool 1
14:48:15:849 MAC Poll 0x02 0xC7DF
14:48:16:547 CTRL db store gui node 00:21:2e:ff:ff:05:af:15
14:48:16:548 CTRL db store gui node 00:0d:6f:00:0e:c8:e7:b0
14:48:16:563 saved node state in 0 ms
14:48:16:585 sync() in 22 ms
14:48:16:586 REST API plugin shutting down
14:48:16:586 DB save zll database items 0x000000C1
14:48:16:586 DB sql exec REPLACE INTO sensors (sid, name, type, modelid, manufacturername, uniqueid, swversion, state, config, fingerprint, deletedState, mode) VALUES ('2', 'Switch 4x EU-LIGHTIFY 2', 'ZHASwitch', 'Switch 4x EU-LIGHTIFY', 'OSRAM', '00:0d:6f:00:0e:c8:e7:b0-01-1000', 'e.1.0.0M', '{"buttonevent":1001,"lastupdated":"2020-05-23T14:48:01.682"}', '{"battery":null,"group":null,"on":true,"reachable":true}', '{"d":2064,"ep":1,"in":[0,1,4096],"out":[5,6,8],"p":260}', 'normal', '1')
14:48:16:588 DB sql exec REPLACE INTO sensors (sid, name, type, modelid, manufacturername, uniqueid, swversion, state, config, fingerprint, deletedState, mode) VALUES ('4', 'Switch 4x EU-LIGHTIFY 4', 'ZHASwitch', 'Switch 4x EU-LIGHTIFY', 'OSRAM', '00:0d:6f:00:0e:c8:e7:b0-03-1000', 'e.1.0.0M', '{"buttonevent":3002,"lastupdated":"2020-05-23T14:48:02.756"}', '{"group":null,"on":true,"reachable":true}', '{"d":2064,"ep":3,"in":[0,4096],"out":[5,6,8],"p":260}', 'normal', '1')
14:48:16:602 DB saved in 16 ms
14:48:16:615 sync() in 12 ms
14:48:16:619 ~Resource() /sensors 0x22147d8
14:48:16:619 ~Resource() /sensors 0x22148b0
14:48:16:619 ~Resource() /sensors 0x2214988
14:48:16:619 ~Resource() /sensors 0x2214a60
14:48:16:619 ~Resource() /sensors 0x2214b38
14:48:16:619 ~Resource() /sensors 0x2214c10
14:48:16:619 ~Resource() /sensors 0x2214ce8
14:48:16:619 ~Resource() /lights 0x23a5fb8
14:48:16:619 ~Resource() /groups 0x21fe3e8
14:48:16:619 ~Resource() /groups 0x21fe468
14:48:16:619 ~Resource() /config 0x224ae14

Smanar, 2020-05-23:
Ha nice, it works better with this device than the 3 buttons ...
So IDK if you remember in wich one order you have pressed the button ? But I have

- button 1001 press long
- button 1001 press long
- button 1001 press long
- button 3002 press
- button 1001 press long
- no button handler for: Switch 4x EU-LIGHTIFY ep: 0x02 cl: 0x0300 cmd: 0x4C pl[0]: 0x01
- no button handler for: Switch 4x EU-LIGHTIFY ep: 0x04 cl: 0x0300 cmd: 0x4C pl[0]: 0x03
- button 1001 press long
- button 3002 press
- no button handler for: Switch 4x EU-LIGHTIFY ep: 0x02 cl: 0x0300 cmd: 0x4C pl[0]: 0x01
- no button handler for: Switch 4x EU-LIGHTIFY ep: 0x04 cl: 0x0300 cmd: 0x4C pl[0]: 0x03
- button 1001 press long
- button 3002 press
- no button handler for: Switch 4x EU-LIGHTIFY ep: 0x02 cl: 0x0300 cmd: 0x4C pl[0]: 0x01
- no button handler for: Switch 4x EU-LIGHTIFY ep: 0x04 cl: 0x0300 cmd: 0x4C pl[0]: 0x03
- no button handler for: Switch 4x EU-LIGHTIFY ep: 0x01 cl: 0x0008 cmd: 0x05 pl[0]: 0x00
- no button handler for: Switch 4x EU-LIGHTIFY ep: 0x03 cl: 0x0008 cmd: 0x01 pl[0]: 0x01
- no button handler for: Switch 4x EU-LIGHTIFY ep: 0x03 cl: 0x0008 cmd: 0x03 pl[0]: 0x00
- no button handler for: Switch 4x EU-LIGHTIFY ep: 0x02 cl: 0x0300 cmd: 0x03 pl[0]: 0xFE
- no button handler for: Switch 4x EU-LIGHTIFY ep: 0x02 cl: 0x0300 cmd: 0x01 pl[0]: 0x00
- no button handler for: Switch 4x EU-LIGHTIFY ep: 0x04 cl: 0x0300 cmd: 0x03 pl[0]: 0xFE
- no button handler for: Switch 4x EU-LIGHTIFY ep: 0x04 cl: 0x0300 cmd: 0x01 pl[0]: 0x00

So 2 workings and 9 I can add, so 11 actions possible, I think it still miss some of them, but I think I have enought data to guess them.

BTW, I haven't see but my code have created 1 entrie by endpoint, sorry, so you have ATM 4 entries for this devices, I will give you a curl command to delete them, if you can't using Phoscon.

Smanar, 2020-05-23:
Ok so I have pull a new version (same adress, same procedure)

In this one I have corrected some button, and add a better description than "press long", you will have "TL press" for Top / Left Press.
Will be easier to reconize event / button.

I have started too, to make only 1 entrie for all endpoint, but not finished yet.

To delete old entries, if you can't using Phoscon, you can use
`curl -X DELETE http://IP:PORT/api/API_KEY/sensors/ID`

With as ID : 3, 5 ,6 , 7, 2 , 4 to delte all your "Switch 4x EU-LIGHTIFY" entrie
For ip and port but not sure > 192.168.178.41:80
For api key 1660D32323

pollierer, 2020-05-23:
Hi Smanar, its realy awesome to see there is real progress here, keep it up! :)

I created some more logs before I saw your post. No worries about the device entries, the installation is on a separate sd card only for testing. I will test to delete the entries via api call tomorrow, I have to leave now for today. Tomorrow I will get back to this. I hope this helps;
I did some test by piping the log to grep button, to see what is happening. after that I also created 4 log files for each button with short and long press. Here are the results (button "name" 1-4 is just defined by me, maybe this is different from any other logic ;) )
What I saw in the console by grep button:

```
button 1 : upper left
---------------------
button 1 short press:
button 1001 press long

button 1 long press:
no button handler for: Switch 4x EU-LIGHTIFY ep: 0x01 cl: 0x0008 cmd: 0x05 pl[0]: 0x00

button 2: lower left
--------------------
button 2 short press:
button 3002 press

button 2 long press: only 2 events in the log, different from button 3 long press:
17:37:35:669 no button handler for: Switch 4x EU-LIGHTIFY ep: 0x03 cl: 0x0008 cmd: 0x01 pl[0]: 0x01
17:37:43:598 no button handler for: Switch 4x EU-LIGHTIFY ep: 0x03 cl: 0x0008 cmd: 0x03 pl[0]: 0x00
the second one seems to be button release

button 3: upper right
---------------------
button 3 short press:
no button handler for: Switch 4x EU-LIGHTIFY ep: 0x02 cl: 0x0300 cmd: 0x4C pl[0]: 0x01

button 3 long press: there are 3 events
17:35:23:481 no button handler for: Switch 4x EU-LIGHTIFY ep: 0x02 cl: 0x0300 cmd: 0x03 pl[0]: 0xFE
17:35:24:469 no button handler for: Switch 4x EU-LIGHTIFY ep: 0x02 cl: 0x0300 cmd: 0x01 pl[0]: 0x01
17:35:30:161 no button handler for: Switch 4x EU-LIGHTIFY ep: 0x02 cl: 0x0300 cmd: 0x01 pl[0]: 0x00
I think this is button down, button hold / long press detected, button up. Hard to map because the log has a certain lag

button 4: lower right
---------------------
button 4 short press:
no button handler for: Switch 4x EU-LIGHTIFY ep: 0x04 cl: 0x0300 cmd: 0x4C pl[0]: 0x03

button 4 long press:
17:40:08:267 no button handler for: Switch 4x EU-LIGHTIFY ep: 0x04 cl: 0x0300 cmd: 0x03 pl[0]: 0xFE
17:40:09:253 no button handler for: Switch 4x EU-LIGHTIFY ep: 0x04 cl: 0x0300 cmd: 0x01 pl[0]: 0x03
17:40:13:631 no button handler for: Switch 4x EU-LIGHTIFY ep: 0x04 cl: 0x0300 cmd: 0x01 pl[0]: 0x00
(button down, hold detected, release)
```

**Log file for button 1 (upper left) short and long press:**

```
18:44:09:325 HTTP Server listen on address 0.0.0.0, port: 80, root: /usr/share/deCONZ/webapp/
18:44:09:343 CTRL. 3.27.218:44:09:482 COM: /dev/ttyAMA0 :  (0x0000/0x0000)
18:44:09:482 dev /dev/ttyAMA0
18:44:09:482 COM: /dev/ttyACM0 : ConBee II (0x1CF1/0x0030)
18:44:09:482 ZCLDB init file /home/pi/.local/share/dresden-elektronik/deCONZ/zcldb.txt
18:44:09:748 parent process bash
18:44:09:749 gw run mode: normal
18:44:09:749 GW sd-card image version file does not exist: /home/pi/.local/share/dresden-elektronik/deCONZ/gw-version
18:44:09:749 sd-card cid: 035344534331364780ffffffff01287d
18:44:09:751 DB sqlite version 3.27.2
18:44:09:752 DB PRAGMA page_count: 30
18:44:09:752 DB PRAGMA page_size: 4096
18:44:09:752 DB PRAGMA freelist_count: 0
18:44:09:752 DB file size 122880 bytes, free pages 0
18:44:09:752 DB PRAGMA user_version: 6
18:44:09:752 DB cleanup
18:44:09:753 DB create temporary views
18:44:09:754 DB view [0] created
18:44:09:754 DB view [1] created
[... 1982 lines cut ...]
17:47:17:953 verify neighbor status: APP_SUCCESS (0x00) 
17:47:18:950 MAC Poll 0x02 0xC7DF
17:47:19:028 void deCONZ::zmNode::setFetched(deCONZ::RequestId, bool) fetched item: 8, node: 0x0000
17:47:19:953 MAC Poll 0x02 0xC7DF
17:47:20:229 CTRL db store gui node 00:21:2e:ff:ff:05:af:15
17:47:20:229 CTRL db store gui node 00:0d:6f:00:0e:c8:e7:b0
17:47:20:246 saved node state in 0 ms
17:47:20:284 sync() in 38 ms
17:47:20:284 REST API plugin shutting down
17:47:20:284 DB save zll database items 0x000000C1
17:47:20:284 DB saved in 0 ms
17:47:20:292 sync() in 6 ms
17:47:20:294 ~Resource() /sensors 0xf95cf0
17:47:20:294 ~Resource() /sensors 0xf95dc8
17:47:20:294 ~Resource() /sensors 0xf95ea0
17:47:20:294 ~Resource() /sensors 0xf95f78
17:47:20:294 ~Resource() /sensors 0xf96050
17:47:20:294 ~Resource() /sensors 0xf96128
17:47:20:294 ~Resource() /sensors 0xf96200
17:47:20:294 ~Resource() /lights 0x11284e0
17:47:20:295 ~Resource() /groups 0xfe7ea8
17:47:20:295 ~Resource() /groups 0xfe7f28
17:47:20:295 ~Resource() /config 0xfc7594

```

pollierer, 2020-05-23:
strange that there is only one event showing up at (real) button 1 long press, and only 2 events at button 2 long press. looks like the events are used somewhere else. I hope you can find everything in the complete logs anyway..

Smanar, 2020-05-23:
Nope was right, you miss 2 event because was the 2 that already working ^^.
I have updated the code according to theses logs, I think all was here.

For long press, I have used only HOLD and RELEASE, because I don't see how the device can make a difference beetwen a short press and a long press during the first press.

BTW, to send log, try to use website like pastebin, because the github will not like it ^^.

So now I think all button are working. I think just need to regroup all device in one (I wait to see how the new code will react).
And for the moment group gestion is not used, I don't think Phoscon will support here, and to use this feature I think using 1 entries by endpoint will be easier.

pollierer, 2020-05-24:
I compiled your recent plugin version Smanar and created a new log.

[https://pastebin.com/9eSWxNYu](https://pastebin.com/9eSWxNYu)

What I did was: Always exactly one short press and after that one long press for each button:
UL, LL, UR, LR in that order.
As far as I can see the UL Button is working.
- LL press release is not processed (no button handler..)
- UR press short says UR press long
- UR press release  is not processed
- LR short press OK
- LR long press also says LR short press
- no messages for LR press release, events processed somewhere else eg LR short press?

btw thx for your info about the API!
Don´t know if it was supposed to be like that: Before that test I deleted the sensors via api and reconnected the switch (via web app). After reconnecting I saw again 6 new sensor ids by querying the api:

curl -X GET http://127.0.0.1:80/api/[key]/sensors/

[https://pastebin.com/Aj9mVAt3](https://pastebin.com/Aj9mVAt3)

Smanar, 2020-05-24:
Ok so In new version I will ignore Endpoint 5 and 6, I don't see if they are usefull.
And try to mix the 4 others.
As all your sensor end by cluster 0x0100, I have code in this cluster to ignore all endpoint other than 0x01, so I hope  you will have only 1 device created.

For LL release I don't understand, for me the code is good.
>13:42:04:092 no button handler for: Switch 4x EU-LIGHTIFY ep: 0x03 cl: 0x0008 cmd: 0x03 pl[0]: 0x00
>    { Sensor::ModeScenes, 0x03, 0x0008, 0x03, 0x00,    S_BUTTON_3 + S_BUTTON_ACTION_LONG_RELEASED,  "LL press release" },

For other I have (tried) to correct them, respecting the order. But I have more event than possible one ...

BTW, have you tried if this device have a "double press" feature ?

olicooper, 2020-05-25:
This might confirm if the bindings are correct?... https://community.smartthings.com/t/osram-lightify-4-button-switch-dth-binder-alpha/99877/8

Smanar, 2020-05-25:
Lol, realy good link, it's realy more cleaner thant z2m, but too late.
There are using (like the last changes) for first type button.
- push : cmd = 1
- held : cmd = 5
- stop : cmd = 3
And for the other
- push : cmd = 76
- held : cmd = 3
- long : cmd = 01 / 1
- stop : cmd =01 / 0

So according to this site I m wrong on LL short press and LL press long for first type button
and UR press Release for the second type.

If someone can test it ?

If the link is good the good code is
```
//    mode                ep    cluster cmd   param button                                       name
    // Button uper / left
    { Sensor::ModeScenes, 0x01, 0x0006, 0x01, 0x00,    S_BUTTON_1 + S_BUTTON_ACTION_HOLD,           "UL press short" },
    { Sensor::ModeScenes, 0x01, 0x0008, 0x05, 0x00,    S_BUTTON_1 + S_BUTTON_ACTION_SHORT_RELEASED, "UL long press" },
    { Sensor::ModeScenes, 0x01, 0x0008, 0x03, 0x00,    S_BUTTON_1 + S_BUTTON_ACTION_LONG_RELEASED,  "UL press release" },
    // Button uper / right
    { Sensor::ModeScenes, 0x02, 0x0300, 0x4C, 0x01,    S_BUTTON_2 + S_BUTTON_ACTION_LONG_RELEASED,  "UR short press" },
    { Sensor::ModeScenes, 0x02, 0x0300, 0x03, 0xFE,    S_BUTTON_2 + S_BUTTON_ACTION_HOLD,           "UR press long" },
    { Sensor::ModeScenes, 0x02, 0x0300, 0x01, 0x00,    S_BUTTON_2 + S_BUTTON_ACTION_LONG_RELEASED,  "UR press Release" },
    // Button lower / left 
    { Sensor::ModeScenes, 0x03, 0x0006, 0x01, 0x00,    S_BUTTON_3 + S_BUTTON_ACTION_SHORT_RELEASED, "LL short press" },
    { Sensor::ModeScenes, 0x03, 0x0008, 0x05, 0x00,    S_BUTTON_3 + S_BUTTON_ACTION_HOLD,           "LL press long" },
    { Sensor::ModeScenes, 0x03, 0x0008, 0x03, 0x00,    S_BUTTON_3 + S_BUTTON_ACTION_LONG_RELEASED,  "LL press release" },
    // Button lower / right 
    { Sensor::ModeScenes, 0x04, 0x0300, 0x4C, 0x03,    S_BUTTON_4 + S_BUTTON_ACTION_SHORT_RELEASED, "LR short press" },
    { Sensor::ModeScenes, 0x04, 0x0300, 0x03, 0xFE,    S_BUTTON_4 + S_BUTTON_ACTION_HOLD,           "LR press long" },
    { Sensor::ModeScenes, 0x04, 0x0300, 0x01, 0x00,    S_BUTTON_4 + S_BUTTON_ACTION_LONG_RELEASED,  "LR press release" },
    // end
    { Sensor::ModeNone,             0x00, 0x0000, 0x00, 0,    0,                                           nullptr }
};
```

But it don't use parameter, and deconz do, so we can have difference here.

pollierer, 2020-05-25:
Smanar tell me if a further test will help you. It´s also hard to understand what is happening by reading the log live, because in my case the console out has a certain lag (and much noise too ;) )

olicooper, 2020-05-25:
@pollierer I don't have the 4 button remote, so if you can test the new bindings in the comment above then that would be great!

Smanar, 2020-05-26:
@pollierer you can enable only the log you need, in our situation I think only --dbg-info=2 is usefull, to display own debug line.

Then if you have time, you can test the actual code but
- LL release will not work, I have same problem for the 3 button, I m on it atm.
- I think I m wrong on LL short press and LL press long and UR press Release.

For the second point, idk if you know a little the c++, but you can try the code in https://github.com/dresden-elektronik/deconz-rest-plugin/issues/374#issuecomment-633635928
Or just confirm me I m wrong, I will make the correction myself.

pollierer, 2020-05-26:
@Smanar I have no clue of c++, but I know Java at least. So was capable of changing the mapping object in the code to the version in your comment above. After that I removed all mappings from the object to get all "no event handler" messages to check the hex values.

some things are realy strange here:

1) Lower left Button press release event is set up right in the object, but the log still says:
long press start:
19:47:23:618 button 3001 LL press long
release event:
19:47:26:198 no button handler for: Switch 4x EU-LIGHTIFY ep: 0x03 cl: 0x0008 cmd: 0x03 pl[0]: 0x00

2) The Upper right and lower right press release happens 2 times if they are set up in the mapping. If I remove the mapping there is only one event.
19:47:42:313 button 2001 UR press long
19:47:43:299 button 2003 UR press Release
19:47:44:641 button 2003 UR press Release
19:47:54:618 button 4001 LR press long
19:47:55:602 button 4003 LR press release
19:47:57:246 button 4003 LR press release

All 4 Button short press events are working :)

I don´t know if there is a way to take into account that the switch is sending 2 events at long press start but what I can´t explain to myself is why there are 2 release events (see above).

Anyway, beside LL press release not working even if the hex values are in the mapping, it "somehow" works now^^

```

static const Sensor::ButtonMap osram4ButRemoteMap[] = {
//    mode                ep    cluster cmd   param button                                       name
    // Button uper / left
    { Sensor::ModeScenes, 0x01, 0x0006, 0x01, 0x00,    S_BUTTON_1 + S_BUTTON_ACTION_HOLD,           "UL press short" },
    { Sensor::ModeScenes, 0x01, 0x0008, 0x05, 0x00,    S_BUTTON_1 + S_BUTTON_ACTION_SHORT_RELEASED, "UL long press" },
    { Sensor::ModeScenes, 0x01, 0x0008, 0x03, 0x00,    S_BUTTON_1 + S_BUTTON_ACTION_LONG_RELEASED,  "UL press release" },
    // Button uper / right
    { Sensor::ModeScenes, 0x02, 0x0300, 0x4C, 0x01,    S_BUTTON_2 + S_BUTTON_ACTION_LONG_RELEASED,  "UR short press" },
    { Sensor::ModeScenes, 0x02, 0x0300, 0x03, 0xFE,    S_BUTTON_2 + S_BUTTON_ACTION_HOLD,           "UR press long" },
    { Sensor::ModeScenes, 0x02, 0x0300, 0x01, 0x00,    S_BUTTON_2 + S_BUTTON_ACTION_LONG_RELEASED,  "UR press Release" },
    // Button lower / left 
    { Sensor::ModeScenes, 0x03, 0x0006, 0x00, 0x00,    S_BUTTON_3 + S_BUTTON_ACTION_SHORT_RELEASED, "LL short press" },
    { Sensor::ModeScenes, 0x03, 0x0008, 0x01, 0x01,    S_BUTTON_3 + S_BUTTON_ACTION_HOLD,           "LL press long" },
    { Sensor::ModeScenes, 0x03, 0x0008, 0x03, 0x00,    S_BUTTON_3 + S_BUTTON_ACTION_LONG_RELEASED,  "LL press release" },
    // Button lower / right 
    { Sensor::ModeScenes, 0x04, 0x0300, 0x4C, 0x03,    S_BUTTON_4 + S_BUTTON_ACTION_SHORT_RELEASED, "LR short press" },
    { Sensor::ModeScenes, 0x04, 0x0300, 0x03, 0xFE,    S_BUTTON_4 + S_BUTTON_ACTION_HOLD,           "LR press long" },
    { Sensor::ModeScenes, 0x04, 0x0300, 0x01, 0x00,    S_BUTTON_4 + S_BUTTON_ACTION_LONG_RELEASED,  "LR press release" },
    // end
    { Sensor::ModeNone,             0x00, 0x0000, 0x00, 0,    0,                                           nullptr }};

```

Here is also a complete grep log with no button handler in osram4ButRemoteMap[] for your reference. Look at the timestamps. In cases where we have two "long press start" events, the second start event always appears somhow a second after the first event. Releasing the button triggers the 3rd event some seconds later, if you release the button:

```
UL long press button down:
20:16:59:611 no button handler for: Switch 4x EU-LIGHTIFY ep: 0x01 cl: 0x0008 cmd: 0x05 pl[0]: 0x00
UL long press button release:
20:17:04:086 no button handler for: Switch 4x EU-LIGHTIFY ep: 0x01 cl: 0x0008 cmd: 0x03 pl[0]: 0x00

LL long press button down:
19:33:28:516 no button handler for: Switch 4x EU-LIGHTIFY ep: 0x03 cl: 0x0008 cmd: 0x01 pl[0]: 0x01
LL long press button release:
19:33:32:519 no button handler for: Switch 4x EU-LIGHTIFY ep: 0x03 cl: 0x0008 cmd: 0x03 pl[0]: 0x00

UR long press button down:
19:35:13:182 no button handler for: Switch 4x EU-LIGHTIFY ep: 0x02 cl: 0x0300 cmd: 0x03 pl[0]: 0xFE
19:35:14:166 no button handler for: Switch 4x EU-LIGHTIFY ep: 0x02 cl: 0x0300 cmd: 0x01 pl[0]: 0x01
UR long press button release:
19:35:16:212 no button handler for: Switch 4x EU-LIGHTIFY ep: 0x02 cl: 0x0300 cmd: 0x01 pl[0]: 0x00

LR long press button down:
19:35:57:038 no button handler for: Switch 4x EU-LIGHTIFY ep: 0x04 cl: 0x0300 cmd: 0x03 pl[0]: 0xFE
19:35:58:023 no button handler for: Switch 4x EU-LIGHTIFY ep: 0x04 cl: 0x0300 cmd: 0x01 pl[0]: 0x03
LR long press button release:
19:36:00:377 no button handler for: Switch 4x EU-LIGHTIFY ep: 0x04 cl: 0x0300 cmd: 0x01 pl[0]: 0x00

all good is:

UL short press:
20:17:51:142 no button handler for: Switch 4x EU-LIGHTIFY ep: 0x01 cl: 0x0006 cmd: 0x01 pl[0]: 0x00
LL short press:
20:17:52:070 no button handler for: Switch 4x EU-LIGHTIFY ep: 0x03 cl: 0x0006 cmd: 0x00 pl[0]: 0x00
UR short press:
20:17:52:979 no button handler for: Switch 4x EU-LIGHTIFY ep: 0x02 cl: 0x0300 cmd: 0x4C pl[0]: 0x01
LR short press:
20:17:53:910 no button handler for: Switch 4x EU-LIGHTIFY ep: 0x04 cl: 0x0300 cmd: 0x4C pl[0]: 0x03

```
Anyway, althrough we are working here at the REST plugin, is there a chance to get the button show up in the pwa with some minor changes?
For me, my switch devices would get usable if all short press buttons are working and if I could use & configure the button in the web app. But I think this is be a different story...

olicooper, 2020-05-26:
@pollierer You describe having "two events" which might be similar to what I have just seen. Maybe this is why the switches haven't been supported yet! See the bottom part of https://github.com/dresden-elektronik/deconz-rest-plugin/issues/2763#issuecomment-634268212

Smanar, 2020-05-27:
For LL problem I think @olicooper have found the solution, just use 0x01 as parameter instead of 0x00.

Smanar, 2020-05-30:
No one have tried the last code yet ?

pollierer, 2020-05-30:
@Smanar 
> How are you removing it from maping? If you remove the line you can't have event at all ?
> 
By removing the mapping I refered to removing the Objects/Lines from the osram4ButRemoteMap Array to get the "no button handler" messages in the log.

I compiled your recend version and it looks good! :)

short press is all correct now, long press UL  und LL also.

UR long press does as expected:
13:08:25:871 button 2001 UR press long
13:08:26:858 no button handler for: Switch 4x EU-LIGHTIFY ep: 0x02 cl: 0x0300 cmd: 0x01 pl[0]: 0x01
13:08:29:811 button 2003 UR press Release

LR long press also:
13:10:28:398 button 4001 LR press long
13:10:29:382 no button handler for: Switch 4x EU-LIGHTIFY ep: 0x04 cl: 0x0300 cmd: 0x01 pl[0]: 0x03
13:10:33:036 button 4003 LR press release

Beside the "middle" Event at long press UR und LR everything seems working now. I don´t think it is neccessary to care about those additional "middle" events.

Thank you Smanar for the good work even without the hardware in place!

Anyway, please enlight me, am I right that the phoscon app (the pwa) is closed source? There is a beta app repository where you only have a link to the installed beta version of your pwa.

[78 more replies]

closed 2020-07-10 by manup