dresden-elektronik/deconz-rest-plugin #376: websocket trigger if light/device comes online again

holli73, 2018-01-20:
hello,

i have now apx. 13 osram bulbs in the garden which are powered only by motion detection - i use the websocket events for lights and groups when they come online to switch color,... - all works fine as long there is enough time between first on and off - but most of the time the second on is not fired - sometimes i get a light event for bri:254 f.ex.:

power on - device fires on websocket - 45 seconds later device off (power off) - few seconds later power on again -> NO event/trigger -> so i can not change something automatically because a missing trigger - sometimes i will get a bri or ct event but in 99% no event.

is there a way i can receive such events (if a device rejoins the network) to have always the possibility to change the light?
or is it possible that this lights store the last state before power off?

any idea/help as always highly appreciated 

thanks
holli

ebaauw, 2018-01-20:
> all works fine as long there is enough time between first on and off - but most of the time the second on is not fired

As you can see, it takes a considerable amount of time before deCONZ marks a light as `"reachable": false`, after you power it off.  If you power it on before that time, you won't get a `"reachable": true` notification on the web socket.

> sometimes i get a light event for bri:254

If the light powers on at different brightness or colour, deCONZ will issue a web socket notification for the power-on values.  However, this typically happens when deCONZ polls the light, so it might take some time after you power on the light (depending on the number of lights in your network).

> or is it possible that this lights store the last state before power off?

That's up to the light firmware, but typically: no.  I've only seen the GLEDOPTO do that (for colour and brightness - the light still turns on when powered on).

> any idea/help as always highly appreciated

You're not going to like this ;-)  Don't use traditional switches - these "smart" lights are really designed to be powered on all the time.

holli73, 2018-01-25:
> You're not going to like this ;-) Don't use traditional switches - these "smart" lights are really designed to be powered on all the time.

one more question - can i do something like a ping to a device and receive a valid up to date info?

so my solution is now to put 12 nodemcu's into my gardenlight's and react on the 230v motion detection and have the bulbs on all the time and just call my restendpoint once the motion sensors fires...

thanks
holli

closed 2018-01-25 by holli73

ebaauw, 2018-01-25:
> can i do something like a ping to a device and receive a valid up to date info?

Technically, yes; this is what the Hue bridge and the deCONZ REST API plugin do in the background.  This functionality is not exposed through the REST API, though.  You would have to write your own deCONZ plugin or use the native API to the RaspBee / ConBee.