dresden-elektronik/deconz-rest-plugin #380: File .elb and deConz a little problem...

VitoLeo92, 2018-01-26:
I purchased a RaspBee premium module and I want to interface it with an Xbee Pro S2C.

On the Xbee Pro S2C module I have the firmware (router): 23A7.elb.
Now I should load the firmware (coordinator) on RaspBee: 0x21A7.elb.

My question:
Can a firmware be converted from .elb to a .zigbee (or .bin.GCF)? I can do it?

manup, 2018-01-30:
>Can a firmware be converted from .elb to a .zigbee (or .bin.GCF)? I can do it?

No since the .elb firmware is made for XBee which uses a very different controller than the RaspBee/ConBee.

The XBee S2 can be configured to join the RaspBee network as a router when flashed with a ZigBee firmware, relevant parameters are:

[image]

gmezz23, 2018-02-07:
Hello everyone,

sorry for another issue (in real it is the same). This is my advanced settings panel of the router that I intend to set with my deConz

[image]

what I must set in order to match my raspbee?

[image]

Thank you so much. Sincerly,

Giovanni

manup, 2018-02-07:
From quick look, set scan channels SC and KY as in the screenshot I've posted above.

You also need to open the network in the WebApp (or search for lights in the Phoscon App).
The XBee should then join the network while search is active.

manup, 2018-02-07:
And your deCONZ settings is wrong, security must be `High - No Master but TC Link Key`.

manup, 2018-02-24:
Closing the issue for now since related parameters have been provided.

closed 2018-02-24 by manup