dresden-elektronik/deconz-rest-plugin #381: Lights keep turning on by themselves

Martin-Gardner, 2018-01-26:
I have an IKEA colour temp bulb and a Philips colour temp bulb in a bedroom group and a Hue dimmer group. The Philips colour bulb is also bound to an IKEA remote. I turn off the lights, then after some minutes they turn back on again! I would say that it is about three minutes before they are turned back on again. I have turned them off using the Phoscon app and the Home Assistant front end. This does not seem to happen so much when using the remotes. Any ideas what is going on here?

I have enclosed a picture on my setup within deCONZ, tell me if you need more information.

Cheers,

Martin

[image]

jonathanweinberg, 2018-01-31:
May I ask if you are dead certain there is no flickering of power where they are connected? No dimmer switch? I've had a devil of a time with just such a thing in a setup of my own once upon a time.

manup, 2018-01-31:
Further are there any rules/motion or sensors controls defined which also turn on the lights?

Martin-Gardner, 2018-02-01:
No physical dimmer and no rules or sensors.

Martin-Gardner, 2018-02-01:
The motion sensor is not configured and is in a draw anyway

onalzengin, 2018-02-17:
@Martin-Gardner If you are using fluxer component with Home Assistant, that is probably the reason.

Martin-Gardner, 2018-02-20:
reinstalled from todays backup - issue went away

closed 2018-02-20 by Martin-Gardner