dresden-elektronik/deconz-rest-plugin #384: IKEA bulbs not always responding

jurriaan, 2018-01-27:
I've just started using my Conbee with a network mostly containing IKEA bulbs. While it works okay most of the time I'm having issues where when turning all the lights off (using a scene in home assistant which turns them off individually using the REST API) some lights will stay on. I have to separately turn them on and off again in the home-assistant interface or the Phoscon app to turn them off (both methods will show the light as being off while it actually is on, so I have to turn them on again and then off for it to work).

Is there something I could do to fix this? Is home-assistant turning the lights off to quickly and overloading some queue?

I'm running deCONZ 2.04.99 with firmware 0x26190500.

joch, 2018-01-28:
I'm having the same issue, and temporarily had to go back to the Ikea hub for the lights to work properly. Will give it a try again with the next update. :)

Kane610, 2018-01-29:
The 2.05.00 is out in beta for RPI if you'd like to give it a spin

joch, 2018-01-29:
I know, but don't use a pi, so eagerly awaiting the other version. :)

jurriaan, 2018-01-29:
Same here, no release for Ubuntu yet..

jurriaan, 2018-02-01:
Just had some more time to look into this issue using the deCONZ GUI, the interesting part was that the bulb didn't respond to the off/toggle commands, but it did respond to the identify command when this happened, which is weird. Then I tried to send the level control 'Stop' command. After which the light started responding to all the other commands again! Not sure if this is always what happens. But it seems like the bulb was hanging in some kind of level control transition? I'm using fluxer in home assistant (Did the same thing with the Ikea hub) which sets the color temperature every 30 seconds with a transition time of 30 seconds. Maybe that's causing some issues here?

joch, 2018-02-01:
I'm seeing the same behavior, and I'm using fluxer in Home Assistant as well. Interesting find!

manup, 2018-02-01:
> I'm using fluxer (Flux Light Adjustment) in home assistant (Did the same thing with the Ikea hub) which sets the color temperature every 30 seconds with a transition time of 30 seconds. Maybe that's causing some issues here?

Maybe, the queue might get full over time or something like that. How many lights are involved? Do you have more details on what commands are issued by `fluxer` (on/off, brightness, ...)?

Kane610, 2018-02-01:
@manup are you using the new deconz component or the hue component?

manup, 2018-02-01:
Du limited time I haven't a home assistant installation running yet :( I was referring to plain deCONZ.

Kane610, 2018-02-01:
Too bad. I bet you sit on a lot of devices that would be nice to verify functionality of the component with

manup, 2018-02-01:
Absolutely time is really precious in the device integration world, but in the end only progress matters :)

jurriaan, 2018-02-01:
@manup I control six lights using the flux component. As far as I know flux sends commands to adjust the color temperature, not brightness (this is how I configured it). I've adjusted the configuration so it only updates the lights every two minutes with a transition time of 15 seconds. Hopefully this helps, will let you know! 

But the queue shouldn't get full when sending 6 lights a command every half a minute right? Also the light started working again after issueing a level control stop command using the deCONZ so I think it is something else. Maybe it has something to do with how the IKEA hub handles transitions. Maybe it stops them before issueing other commands?

onalzengin, 2018-02-01:
@jurriaan I had the exact same problem. I overcame it by setting transition time to 1 second and update interval to 10 mins, which works perfectly fine, I don't notice any color jumps.

epegzz, 2018-02-04:
I ran in this issue as well. Using latest deonz version (2.05.00) and latest homeassistant version. I have Philips Hue lights as well as Ikea Tradfri lights connected.

During transitions issued by Fluxer (which I have set to 30 seconds) the Ikea lights stop responding (doesn't matter whether controlling them via HA or directly from deconz).

The Hue lights however are staying responsive.

joch, 2018-02-07:
I changed transition to 0 and update time to 30 seconds for fluxer, and it has been running quite well for the past few days. I’m going to disable fluxer in my HA automations before changing the light, and then renabling afterwards. Also just increasing the update time to 10 minutes or so. That should take care of any remaining issue. Great being able to use conbee for the ikea lights as well!

jurriaan, 2018-02-07:
Same here, since tweaking the fluxer settings it works fine! Still wondering why this doesn't happen with the IKEA hub. What are they doing differently so the lights are still responding properly during transitions?

donnib, 2018-02-09:
@joch @jurriaan could one of you share your fluxer config from HASS, also have you done anything to update lights that are off i mean when you turn a light on that has been off that it get's updated from fluxer ? I did this without the time : https://community.home-assistant.io/t/flux-on-lights-that-are-off/27950/3

joch, 2018-02-09:
@donnib I currently have two flux components running:

https://github.com/joch/home-assistant-config/blob/0bc25b9bb1fb6bc32d4d983b3b635c808945d183/configuration.yaml#L109

One which updates the lights automatically, and one which updates the kitchen lights manually, since I want another color on those lights when they are dim. Automation:

https://github.com/joch/home-assistant-config/blob/0bc25b9bb1fb6bc32d4d983b3b635c808945d183/automation/kitchen/lights.yaml#L35

There are some quirks with this that I want to fix though, like disabling the automatic flux when light intensity is changed or on/off are triggered via automation and then enabling it again. Basically a workaround for the issue with Ikea lights together with conbee, since they work fine using the Ikea hub.

manup, 2018-02-24:
Since there seems a workaround via fluxer config can this issue be closed?

epegzz, 2018-03-10:
@manup IKEA bulbs are unresponsive during transitions when used with Conbee.

The workaround is to keep transitions short and don't use them often to minimize the chance of running into the issue.

This clearly is not a solution to the problem. So please, keep this issue open 😄

manup, 2018-03-10:
Interesting but when it seems to be a problem in the bulb firmware not deCONZ?

epegzz, 2018-03-10:
Unfortunately, I don't have an IKEA hub to reproduce the issue :( Do you have one by chance?

epegzz, 2018-03-10:
But yeah, I guess you're right and it's a bulb firmware issue

epegzz, 2018-03-10:
Okay, I just updated the bulb firmware and the issue is indeed gone! 🎉🎉

This issue can be closed then I guess 🙂

manup, 2018-03-10:
Cool good to know, in future version deCONZ should provide the IKEA OTA update files automatically when older firmware is detected.

closed 2018-03-10 by manup

epegzz, 2018-03-10:
Cool! 🙂