dresden-elektronik/deconz-rest-plugin #385: Xiaomi sensors gets new "names" in HASS on service restart

fiskfan, 2018-01-28:
I'm unsure if this issue belongs in this forum or not...

Currently im running "Home Assistant" which is integrated to deCONZ (https://home-assistant.io/components/deconz/). 

I have noticed that the XIAOMI temperature sensors which reports it's information in multiple ID:s in the REST interface is not behaving correctly in Home Assistant. In the deConz REST API, the sensors has three different ID:s. One for temperature, one for humidity and one for air preasure. In the REST AIP the ID for the meassurement type remains the same, but in Home Asstant the "ID" is changed on every restart of the service...

In home assistant, the sensor reports its friendly_name as how i configured it in the deconz web interface. And it appears that when home assistant loads this device, then it adds a _X in the friendly name and this number is not the same on every restart. 

This means that "sensor.env_bedroom" might be a temperature reading while "sensor.env_bedroom_2" is humidity until the service is restarted, then these two can be swapped with eachother. This makes things a bit difficult...

First example:
[image]

After restart, notice that the devices has been swapped:
[image]

Is there a way to configure the integration so that maybe the friendly_name gets the data type added efterwards in the name instead of a number? For example "sensor.env_bedroom_temperature" and "sensor.env_bedroom_humidity" since we have this information in the REST API? By appending the data type we also have a unique name and it also makes it easier to select the sensor from the list.

mathos77, 2018-01-28:
Hi, this is caused by 2 things.
First of all, deCONZ with the new Phoscon web UI, discovers the sensors properly. However, if yoou give the sensor a name (which are 3 sensors under the hood) they are all three named the same.
So the deCONZ component written by @Kane160 just adds them as they are, but in HASS you can not have multiple items with the same component_id. So a number is added behind it there.

Now....if you want to change the different sensors in the one box, individually, you need to do this via REST.

Kane610, 2018-01-29:
I've fixed the immediate issue for this, which is available in 0.62. Hass will always load deconz sensors in the same predictable order.

I've created an issue here and on Phoscons GitHub to get them to suffix sensors from multisensor devices with their type.

I'm also working on a simplified way of renaming sensors of multisensor devices through hass, discussions are ongoing about my solution though, so no timeframe.

mathos77, 2018-01-29:
Haha, I had your tag wrong. 
That would be awesome. Even though, the type could be used to suffix:
ZHAPresence, just cut off the ZHA and make it: name_StrToLower(Pressure)

But I do agree that Phoscon could resolve that better.

Kane610, 2018-01-29:
@mathos77 all sensors got a type so no need to do that :), but I don't want to hardcode suffixes in the middleware. The uncontrollable change of suffix on start have been fixed. So there shouldn't be any issue for users, they can just use customise to give it a better friendly name. They can also using the API rename each of their devices sensors to unique names.

fiskfan, 2018-01-29:
Thanks for your quick reply.

I just upgraded HASS to 0.62 but the issue remains where the device is imported in a random order. I have deleted the device thru phoscons and readded it without chaning the default namn that it receives but still the temperature/humidity/preassure is imported in a random order in HASS. - Am I missing something obvious?

A rename thru rest should be OK i think, but import in the same order in HASS or a default suffix in deCONZ would have been nice i think?

Kane610, 2018-01-29:
If you remove and reads the devices to Deconz then they will get different ids. The fix only solves restarting hass

Kane610, 2018-01-29:
Hass will soon add support for a entity registry this will also help improve this situation

fiskfan, 2018-01-29:
Yeah. I just removed and readded it to deconz to verify if i was missing something. 

HASS (0.62.0) still loads the temp/hum/press in random order on hass service restart even if i haven't made any adjustments in deconz.

What does it mean with "entity registry"?

Kane610, 2018-01-29:
That hass will couple a unique I'd with  a entity name

Kane610, 2018-01-29:
Maybe you could close this issue and open one in HASS instead and we can continue this discussion there. I'd like some debug information because my hypothesis doesn't seem to be correct if you still have issues

fiskfan, 2018-01-29:
Sure. i will go ahead and close this. thanks for your help in this one.

closed 2018-01-29 by fiskfan