dresden-elektronik/deconz-rest-plugin #387: State Slow to retrieve Light Settings in RESR

rmorriss, 2018-01-29:
Hello,

Using "modelid": "LIGHTIFY RT RGBW", the Hue and Sat respond OK when setting these values, but when querying the state of the light, these values seem to remain for quite some time.

For example, set this value:

[{"success":{"/lights/25/state/on":true}},{"success":{"/lights/25/state/bri":220}},{"success":{"/lights/25/state/hue":20000}},{"success":{"/lights/25/state/sat":220}}]

The checked the response 30 seconds later:

{"etag":"xxx","hascolor":true,"manufacturername":"OSRAM","modelid":"LIGHTIFY RT RGBW","name":"Office 04","state":{"alert":"none","bri":220,"colormode":"hs","effect":"none","hue":0,"on":true,"reachable":true,"sat":254,"xy":[0.287719,0.30123]},"swversion":"01020492","type":"Color dimmable light","uniqueid":"84:18:26:00:00:08:3b:b9-03"}

"bri" changes instantly.

Over one minute has passed and nothing has updated.  I found if I go back (20 minutes or so later) it might look OK.  Do I have something set wrong or is this a bug?  Thanks!

ebaauw, 2018-01-29:
Can you check what web socket notifications deCONZ issues?  I would expect deCONZ first to echo the values you just PUT, and then (after it’s queried the light) the weird hue and sat values.

I’ve never seen a `type` of `”Color dimmable light”` before.  Is this a US light with a ZHA endpoint or an EU light with ZLL endpoint?  Could you post a screenshot of the node in the deCONZ GUI?

I think this is a bug in the ZLL spec, which doesn’t specify how other colour attributes should change when setting colour through one mode, causing different manufacturers to take a different approach.  deCONZ follows the same approach as Philips, but OSRAM and IKEA don’t.  If you’re up to some reverse engineering, you might check the _Color control_ cluster (0x0300) in the deCONZ GUI, in particular what values the attributes show after changing the colour through _Hue_ and _Saturation_, through _Current X_ and _Curent Y_, through _Color temperature_, and through _Enhanced Hue_ and _Saturation_.  Note that there’s two values for hue, an 8-bit value and a 16-bit value.  I’ll bet only one of them is updated and deCONZ reads the other, which is still 0.  There’s also two _Color mode_ attributes which could be out of sync, confusing the hell out of deCONZ.  See #53.

wvuyk, 2018-01-29:
I would like to add to this issue that the Webhook response on the above is not returning hue or sat values, but x and y values. This is causing some confusion.

`19:28:50.2857090 -- (WebCommander.HandleWebQueue) Dequeue command: json {"e":"changed","id":"25","r":"lights","state":{"y":21476},"t":"event"} new queue size is 0 on XXXXXXXXXXX
19:28:50.2813120 -- (WebCommander.HandleWebQueue) Dequeue command: json {"e":"changed","id":"25","r":"lights","state":{"x":20413},"t":"event"} new queue size is 0 on XXXXXXXXXXX
`
Also, when sending a Hue command

`{
"on": true,
"bri": 180,
"hue": 43680,
"transitiontime": 10
}`

Without the sat command, which works for Philips bulbs, will return a "success" result for the Hue command, but will not execute, unless you also add the sat command.

rmorriss, 2018-01-30:
So without going farther into a rathole, I will confirm that YES they are ZHA and NOT ZLL.  It looks like OSRAM doesn't make their firmware available.  I've emailed them, I hope to have some positive response.  Meanwhile, with that confirmed, I can do anything you'd like me to do - check whatever - I thought it just made sense to first confirm these are in fact ZHA devices.  All 60+ of them!  I'm running 0x01020492 firmware on all devices.

rmorriss, 2018-01-31:
Totally forgot the attachments @ebaauw !
[image]
[image]

closed 2019-03-30 by stale[bot]