dresden-elektronik/deconz-rest-plugin #388: Development setup - doc ammendment

buzzdev, 2018-01-29:
Hello,
this is not a bug or something, just an improvement to your README.md.
Maybe it might be helpful for someone, to add folowing to the https://github.com/dresden-elektronik/deconz-rest-plugin#install-deconz-development-package-optional

In order to be able to compile deconz-rest-plugin following needs to be installed:
```
sudo apt install libqt5websockets5-dev libqt5serialport5-dev qt5-qmake qt5-default
```

Thanxs

manup, 2018-01-30:
The dependencies are listed in the package and step 3. should take care to install them:

```
 sudo apt update
 sudo apt install -f
```

buzzdev, 2018-01-30:
Hi Manu,
yes these are listed, but that doesn't really installed everything needed to compile and i had to manually install the ones i've listed above.
Then the compilation was working.
On Ubuntu 16.04 and also ElementaryOS 0.4.1

manup, 2018-01-30:
Hmm strange, ok I'll check it out and adjust dependencies in the package if something is missing.

buzzdev, 2018-01-30:
@manup I just installed a brand new Ubuntu 16.04 server (virtual machine) and strangely enough, all went through without need to install any additional packages.

Only thing i can think of is, that i used localised ubuntu repos on those other two setups, where i needed to install extra packages, and on this one i explicitly used the ubuntu's Main repo servers.

```
$ history
    1  wget https://www.dresden-elektronik.de/deconz/ubuntu/beta/deconz-2.04.99-qt5.deb
    2  wget https://www.dresden-elektronik.de/deconz/ubuntu/beta/deconz-dev-2.04.99.deb
    3  sudo dpkg -i deconz-2.04.99-qt5.deb
    4  sudo apt update
    5  sudo apt install -f
    6  sudo dpkg -i deconz-dev-2.04.99.deb
    7  sudo apt update
    8  sudo apt install -f
    9  mkdir Documents
   10  cd Documents/
   11  git clone https://github.com/dresden-elektronik/deconz-rest-plugin.git
   12  cd deconz-rest-plugin/
   13  git checkout -b mybranch V2_04_99
   14  qmake
   15  make -j2
```
Build successful.

manup, 2018-01-31:
We also just use a very vanilla Ubuntu to test our setup, so maybe in the localized repositories some of the meta packages are more slim than the main packages.

buzzdev, 2018-02-02:
Thank you for help.

closed 2018-02-02 by buzzdev