dresden-elektronik/deconz-rest-plugin #389: Zigbee Network loosing connection to Paul Neuhaus Q-lamps

Schmetterfaust, 2018-01-29:
Hi,
I´m not sure if this is an issue or I´m doing something wrong.
Situation:
Raspberry 2 with Raspbee running deconz 2.04.99 and firmware 26190500
The Neuhaus Q-Lamps can be controlled via included remote control.  The manual states to link the remote control 1st to a gateway before linking it to the lamp. This works so i can control the lamp via remote control and deconz Web.
But after some time the link between lamp and Raspbee is lost and control via Web is not possible anymore. Control via remote is still possible.
If I reset the remote and just add the lamp to the zigbee network and keep the remotecontrol unlinked it works stable without problems. As soon as I add the remote the problem starts again.
Any idea? Wrong sequence ? 
Regards
Schmetterfaust

manup, 2018-01-30:
>If I reset the remote and just add the lamp to the zigbee network and keep the remotecontrol unlinked it works stable without problems. As soon as I add the remote the problem starts again.

How do you add the remote?
I figure when you pair the remote with the light they create their own network to which the RaspBee has no access.

Schmetterfaust, 2018-01-30:
Ok. This would explain why raspbee can't control the device any more.
I have opened the network via Web and followed the manual. Press on/off button on control while it is near the raspbee. The control confirms this and I can see it in deconz as battery device.
But unfortunately the rc "wins"

sheineme, 2019-02-28:
If it's still of interest, I have the setup working to control a Q-Flag light from deconz and wth the Q remote (since a few weeks). I followed the instructions from a PDF document provided by the manufacturer - just ask your favourite search-engine for "Q FIBEL". At the time of writing the link was [https://www.paul-neuhaus.de/fileadmin/user_upload/paul-neuhaus/Q_FIBEL2.1.pdf](url). 

I cannot control the light - neither from deconz nor by the remote - when it is switched off with the in-wall light switch. But ok: no electricity, no ZigBee radiocommunication ;)

closed 2019-07-05 by stale[bot]