dresden-elektronik/deconz-rest-plugin #390: Request: Power Meter Switches

Tom-ahawk, 2018-01-29:
1) Please add Power Meter Switches to the deConz App/gateway.
http://www.climax.com.tw/psm29zb-zb.php
 
2) The Phoscon App, displays my Gateway as  RaspBee (with a firmware update 0x26190500 ). When I have ConBee, - on a RPI3.

ebaauw, 2018-01-29:
Do you have the R variant or the plain one?  The R should work already (exposed as light - no metering support).  For the plain one, we need to change deCONZ to deal with end-device lights, see https://github.com/dresden-elektronik/deconz-rest-plugin/issues/356#issuecomment-358759456.

Tom-ahawk, 2018-01-29:
Thanks for replying. 

I have the PSM-29ZBSR: ZigBee Power Switch with Meter and Router function, and it works like a light, but no power metering. -I hope this will be added.

ebaauw, 2018-02-18:
@Tom-ahawk could you please include some screenshots of what the switch looks like in the deCONZ GUI.  In particular: and overview of the endpoints and clusters and the cluster info panel for the Basic cluster and for the Electrical Management cluster.  I'm hoping/assuming they report power through the Active Power attribute of that cluster.  Is this in Watt or in 0.1 Watt?

Tom-ahawk, 2018-02-26:
@ebaauw  I am running the ConBee on a RPI3 and the Hassio Os. Unfortunately I am not able to connect Deconz installed on my PC, to the ConBee on my RPI3. So, have only the Phoscon App to look at, - with limited information.

Tom-ahawk, 2018-10-13:
@ebaauw I hope this can help, getting the Power Switch reporting power consumption. Ny the way, its registered in deCons as a light, not a switch.

[image]
[image]

manup, 2019-01-24:
The Climax power meter switches should work with recent versions, closing the issue for now.

closed 2019-01-24 by manup