dresden-elektronik/deconz-rest-plugin #391: New Sunricher remotes

mathos77, 2018-01-29:
Hi @manup, @ebaauw ,

I have received my Sunricher.com devices (remote) today.
They are mains-powered remotes, with in this case, 4 groups and such.
One is CT, whites, the other is color.

http://www.sunricher.com/4-groups-cct-zigbee-touch-remote-controller-sr-zg9002t4-cct-eu.html
and
http://www.sunricher.com/4-groups-rgbw-zigbee-touch-remote-controller-sr-zg9003t4-rgbw-eu.html

Now, I have my Conbee available to sniff anything using Bitcatcher.
Can you guide me through on how we would perhaps get these supported as sensors (with multi-touch eventually) ?

Initially I do not even see them popping up in deCONZ UI. I most probably set some debugging options ?

Kind regards, Tom.

mathos77, 2018-01-29:
Hmmm, seems I have it paired now:
[image]

ebaauw, 2018-01-30:
I’m not sure these can be supported with buttonevents by the deCONZ REST API, the description indicates it needs to be bound to individual lights, not groups.  deCONZ will have a hard time eavesdropping on the light commands, see https://github.com/dresden-elektronik/deconz-rest-plugin/issues/230#issuecomment-359568335.

I’d say first order of business it to bind the switch’s _On/Off_, _Level control_, and _Color control_ clusters of the first endpoint to a light and see that the light can be controlled as group 1.  You should be able to bind up to 12 lights per endpoint.

Next, try binding the second endpoint to a group instead of to an individual light, and see whether the lights in the group respond to the switch.  If that works, sniff what commands the switch sends on each button, I’d expect this to be fairly standard.  Also, we’d need a screenshot of the _Basic_ cluster.

mathos77, 2018-01-30:
Hi Erik,

The switch can be connected to individual or a group of lights directly, but also to a controller (Which is what I now have done.)

I will make the screenshots, and hook it up to a Hue bulp to see what it does.
Sniffer-wise that is.

Kind regards, Tom.

manup, 2018-01-30:
The website shows a error (might be a temporarily issue). 

>I’m not sure these can be supported with buttonevents by the deCONZ REST API, the description indicates it needs to be bound to individual lights, not groups. deCONZ will have a hard time eavesdropping on the light commands, see #230 (comment).

Maybe, but if the cluster can be bound to a light it also can be bound to RaspBee/ConBee as unicast binding, same as attribute reporting works.

mathos77, 2018-01-30:
Here is the Node:
[image]

And per item, the cluster Info: (For Color Dimmer Switch, I only have done the 0x01, for the Scene I have a separate 2 screens)

[image]

[image]

[image]

[image]

[image]

ebaauw, 2018-01-30:
Could you try and read the attributes in the _Basic_ cluster?  Double-click on each attribute to read them individually if needed.  We need at least the _Manufacturer Name_ and _Model Identifier_.

mathos77, 2018-01-30:
Hi Erik,
The information is the same for all (4xColorDimmer (0x01-0x04)&& 1xColorScene (0x0A))

[image]

[image]

mathos77, 2018-01-31:
@ebaauw Hi Erik,

Do you have tips for me, sniffering the thing ?
If I filter on 0x81F4 I am not seeing that much, so I might need to apply different filters.
I still have to hook it up to a bulb.
Just too much work atm, so little time :(

ebaauw, 2018-01-31:
Make sure to set the right filter: both source and destination NWK address, with an || (or) - BitCatcher defaults to && (and).  Don't filter on the MAC layer.
[image]

mathos77, 2018-02-02:
I see the regular traffic, Routing requests and stuff like that
But when I let's say do anything changing the CT or dimmer or power, I do not see anything in the sniffer (using the filter)

ebaauw, 2018-02-03:
Does BitCatcher decrypt the packages?  Did you enter your network key?

It migh help to place the ConBee, the RaspBee and the device being monitored physically close to each other.  I prefer to use my test Raspberry with RaspBee for sniffing, with only a few devices connected.

BitCatcher is kind of crude.  I mostly use it for capturing, then export the full log and analyse it in WireShark.

s1738berger, 2018-03-15:
Any news realting to the wall remotes from sunricher? ( the first remotes I found that look great and intuitive )
I would love to use them with deconz to control ikea and osram bulbs that are already working fine with deconz ( with a RaspBee )
Did you order the remotes as sample direct from sunricher ?

alferrari, 2018-04-13:
Same question here. Where di you get the devices ? We could share the effort to make them work.

closed 2019-03-30 by stale[bot]

alferrari, 2019-08-16:
It worked for me!

velzend, 2019-08-17:
Hi @alferrari,

Good to hear it worked for you. What steps did you perform in order to add the wall switches/ buttons to deconz. I like to get events from the web socket. 

I have the 8-fold battery powered wall button switch. It has 4 rows of ON/OFF (1/0).
In order to pair this device I logged-in d the PhosconGW, went directly to switches, clicked new, clicked other in the dialog and clicked next. According to the dialog I have 3 min. to pair the client device.
On the button switch I hold down the left and right button for approx 5 seconds. The blue signal led turns on for about 3 seconds and nothing happens. I tried to manually reset the switch. (by holding the same left and right button till the led turns on and quickly press the left button 5-times, now the led flashes 3 times). 

Any help is appreciated. 

Thanks

caplam, 2019-12-07:
any news on sunricher remotes ?
I have a ZG2819S-RGBW and i'd like to catch commands in a home automation system.
It exposes 4 home automation color dimmer switch. I guess one for each zone. Each HA has 13 clusters but only basic and power configurations clusters are populated.
As i'm knew to zigbee i''m struggling to learn basics.
I hope to have time to buy a sniffer (any recommendantions for a macos user?) and start gathering useful stuff.

rweise, 2020-08-06:
Now i own a iCasa remote that is definitely the same, but with other firmware and it works out OTB.
but the Sunricher FW will not work, only Pair 1 and the Master ON/OFF will show a button event :-(

Mimiix, 2020-08-06:
@rweise Can you open a new device request issue and add the details requested in the template?

rweise, 2020-08-06:
I will make this tomorrow. By the way, it is also the same Firmware only Manufacturer and Devicename are different and what a bad joke, icasa is not shown in Phoscon , sunricher is - but will not work in IOBroker over deconz.