dresden-elektronik/deconz-rest-plugin #393: Change state to a Osram plug

almadsen, 2018-02-02:
This is not an issue, but more of an ask for some help.
I'm trying to change the state to an Osram smart+ plug, on and/or off.
Have a ConBee installed on an raspberry pi b 3 and node-red with websocket.
No problem getting update status from websocket but how do I go about to change the status to the plug?
It's being read as a light by deConz, but that shouldn't be a problem.

Would be very thankful for some help, as I'm not getting it to work on my own.

markbeee, 2018-02-02:
To change the status of your Osram plug you can send a REST "command" (HTTP PUT).
In Node-Red you may use a function (or template) node like:
var msg1 = {};
msg1.payload = "{\"on\": true}";
return msg1;

and send this to a http request node:
METHOD: Put
URL: IPofYourConBee/api/yourAPIKey/lights/numberOfOsramPlug/state

Using a function node, you could change the payload according to the input (msg.payload) of the function like so:
var msg1 = {};
msg1.payload = "{\"on\":" + msg.payload + "}";
return msg1;

and then send this to the http request node.

almadsen, 2018-02-03:
That worked like a charm, thanx, buuut for curiosities sake, isn'r it possible with websocket or mqtt to do this?

ebaauw, 2018-02-03:
The web socket interface is relatively new and does not provide any authentication.  That's bad enough for only reporting status, but imho unacceptable for controlling your home.  Of course the authentication on the REST interface is a joke as well, as it runs over plain HTTP.

deCONZ doesn't support mqtt at all.  I think it's on @manup's long-term backlog, but there's plenty of other, higher priority issues to address.

markbeee, 2018-02-03:
MQTT over TLS (SSL) would be my favorite, you could also choose a more unsafe method like username/ password combo with MQTT and additionally you would have a QOS which is not implemented in REST. However I do not know how much effort is necessary to implement a MQTT functionality into deCONZ.
Currently I would NOT expose deCONZ to the outside world in any way, even with the (weak) API-Key functionality provided.

manup, 2018-02-18:
> deCONZ doesn't support mqtt at all. I think it's on @manup's long-term backlog, but there's plenty of other, higher priority issues to address.

Indeed MQTT and proper TLS encryption will come and they are highly wanted, but I'm not yet safe when to schedule this due all the other open items on the never ending todo list.

Closing this issue for now since the main question was answered.

closed 2018-02-18 by manup