dresden-elektronik/deconz-rest-plugin #394: [Question] deRFusb23E06 compatibility

asciidisco, 2018-02-02:
Hi there,

I must admit I haven't checked back with this project in a while & I was happy to see that it evolved so well 👍 

Now to my question:

I´ve bought a `deRFusb23E06` stick (the one with the green dot on it) a couply of years ago (and actually never did anything with it), but now I thought I´d give it a try. Installation of the deconz software worked flawlessly & the stick is recognised:

```bash
pi@raspberrypi:~ $ lsusb
Bus 001 Device 006: ID 1cf1:0025 Dresden Elektronik deRFusb23E06
```

But somehow it doesn't seem to work. I suspect it´s because it´s either running on a very old firmware or has none installed (research has shown that the one with the green dot should´ve at least something installed).

So, is there any guide on how I can proceed with this (downloading & updating the stick firmware f.e.). Maybe I´m even missing some more fundamental thing, as GCFFlasher returns this:

```bash
pi@raspberrypi:~ $ sudo GCFFlasher -l
GCFFlasher V2_11 (c) dresden elektronik ingenieurtechnik gmbh 2017/12/10
RaspBee: /dev/ttyS0 (default)

no FTDI devices found
```

So yeah, any info or guide would be helpful to me. Thanks in advance :)

ErnstEeldert, 2018-02-06:
Not sure about this particular device, but I also find the output of GCFFlasher -l somewhat confusing, as I think it is telling you it found a Raspbee at /dev/ttyS0 (I get the exact same output).

Here's what I use to flash my Raspbee device:
`sudo GCFFlasher -f deCONZ_Rpi_0x261a0500.bin.GCF`

And I pull the firmware like this:

`wget https://github.com/dresden-elektronik/deconz-rest-plugin/files/1565537/deCONZ_Rpi_0x261a0500.bin.GCF.zip`

I don't know if that firmware works for your device though, so checkout the dresden elektronik website for that.

manup, 2018-02-08:
GCFFlasher only supports RaspBee and ConBee devices, deRFusb23E06 is currently not maintained and only older firmware is available which must be flashed with the SAM-BA tool.

asciidisco, 2018-02-16:
@manup Thank you for the heads up.

If someone would be able to give me a quick start tutorial and/or share knowledge how they got it work with a 'blank' `deRFusb23Exx` then it would be much appreciated.

closed 2019-03-30 by stale[bot]