dresden-elektronik/deconz-rest-plugin #395: REST API doesn't return unreachable lights after restart?

ErnstEeldert, 2018-02-05:
deconz version: 2.0.5.1
raspbee firmware: 0x261d0500

I have noticed that after a restart of deconz (headless), unreachable lights are not visible in the REST api as before the restart, until they become reachable again. This is somewhat of a problem, as I have Home Assistant talking to the REST API, so if I restart Home Assistant in the mean time, the light is not visible there as well. Since Home Assistant is not able to dynamically add devices from deconz (afaik), this means I have to restart Home Assistant at a later point to have the reachable light visible up the stack (I also have homebridge configured on top of Home Assistant).

Is this expected behavior?

ebaauw, 2018-02-05:
See https://github.com/dresden-elektronik/deconz-rest-plugin/issues/97.

I build a workaround into homebridge-hue, to delay startup until all known lights are avaiable.  Unfortunately, this requires maintaining a resourcelink with all known lights.

ErnstEeldert, 2018-02-05:
Sjeez. Not sure how I missed that in my search. Thanks for pointing out!

closed 2018-02-05 by ErnstEeldert

ebaauw, 2018-02-05:
Not to worry, searching is a lot easier when you know it’s there, somewhere.  Moments like these make me fantasise about setting up a FAQ...