dresden-elektronik/deconz-rest-plugin #397: Schwaiger multi sensor ZHS05 support

Framspott, 2018-02-06:
I am currently doing my master thesis in the University of applied siences in Salzburg.
Therefore I have to implement a Java ZigBee interface for using it to gather data from sensors.

I am currently using the Raspbee module for the Raspberry Pi. With this module I already can control lights
over the deconz webapp and the REST API of the manufacturer dresden elektronik.

Now I want to use Java and therefore your API, but I a questions regarding your API.

Until now I only (successfully) used Zigbee Hue products from Phillips, but I want to use a multi sensor from the manufacturer Schwaiger (Schwaiger ZHS05 Funk-Multisensor). Is such a sensor supported by your library? Or is there a possibility to use custom sensors/actors?

Thanks in advance for your answers!

All the best

Framspott

manup, 2018-02-06:
The Schwaiger sensors are not supported yet if you just need some sensors there are few other options which already work.

The deCONZ REST documentation lacks behind but *Homebridge Hue* as listed &mdash; them marked with `(2)` :-)

https://www.npmjs.com/package/homebridge-hue

Framspott, 2018-02-07:
Hi manup,

thanks for the fast reply! Is there a way to support you integrating it in the REST API? Or is there any other way for communicating with the raspbee (maybe for outputting messages...)?

All the best

Framspott

ebaauw, 2018-02-07:
@Framspott, Not sure what your research assignment is precisely, but if you were to use the deCONZ REST API, technically you’re not interacting with ZigBee from Java.  If your thesis is about IoT or home automation, that’s probably OK.  If it’s about hard-core technology, ZigBee, real-time interfacing, or something like that, you’d probably need to use the native serial protocol to interact with the RaspBee, see https://github.com/dresden-elektronik/deconz-rest-plugin/issues/158.  In that case you wouldn’t need Swaiger support in the deCONZ REST API plugin.

Framspott, 2018-02-07:
Hi ebaauw,

Thanks for the hint! The idea was to communicate with the REST API, because real time interfacing is not such a big deal. The application should later be used in Home automation. Could you be so kind and send me the private documentation of your serial interface? Can I use the Pi4J's serial communication functions in this case?

Thanks in advance

Framspott

ebaauw, 2018-02-07:
> Could you be so kind and send me the private documentation of your serial interface?

You have to ask @manup. I’m just a deCONZ user, not associated with dresden elektronik.

Framspott, 2018-02-07:
Ok thanks!

Framspott, 2018-02-12:
Dear @manup you said some days ago that it would be possible to integrate the Schwaiger sensor to the REST Api. Could I help you in any way to do so? I also currently trying to use the RaspBee with another library. Could anyone provide me the following specifications for the RaspBee and a Raspberry Pi 3?
- serial Port
- Baud rate
- channel

I would be very grateful if anyone could help me.

All the best

Framspott

manup, 2018-02-15:
Hi @Framespott regarding the RaspBee specs:

Serial port /dev/ttyS0
Baud rate 38400
Channel configuration as described here:
    https://github.com/dresden-elektronik/deconz-rest-plugin#software-requirements

Please also have a look at the RaspBee user manual https://www.dresden-elektronik.de/fileadmin/Downloads/Dokumente/Produkte/ZLL/RaspBee-BHB-en.pdf

Framspott, 2018-02-15:
Thanks @manup ! Now it works! It was all about the wrong port! I used the /dev/ttyAMA0 port before, but it has to be port /dev/ttyS0!

@manup Altough I am trying to use another Java Lib for interferring with ZigBee devices, I would be grateful if I could help to implement a support for the Schwaiger multi sensor ZHS05! Just let me know if you need any information regarding this device!

All the best and big THANKS!

Framspott

closed 2019-03-30 by stale[bot]