dresden-elektronik/deconz-rest-plugin #398: IKEA Tradfri motion sensor - brightness-support?

bjoernhoefer, 2018-02-06:
Hi,

before buying one I would like to know two things regarding the IKEA tradfri - in sepcial the motion sensor:

1. Can it be "paired" to a Raspberry Pi with the raspbee-module - without the need of an extra gateway/central?

1. Can the motion sensor also measure/report the brightness  and if so, will the information be pushed to the central (see above ;-) ) or do I need to somehow "pull" this information from the sensor?

Thanks and sorry for maybe not using the proper jargon - I'm pretty new to ZigBee...

Cheers

ebaauw, 2018-02-06:
1. The deCONZ software on your Raspberry acts as gateway (using the RaspBee for ZigBee radio).

2. The Trådfri motion sensor only reports whether it's dark or not (no lightlevel like the Hue motion sensor), and only when the dial on the back it set to Night setting.  The sensor only sends a notification to deCONZ when motion is detected.  deCONZ will push the sensor state, including `dark`, over the websocket, when it receives a notification from the sensor.  deCONZ will reset the state after the number of seconds specified on `config.duration` (default: 60).  Using `dc_eventlog` to monitor the websocket:
```
Feb 06 22:41:18 pi2 dc_eventlog[8946]: /sensors/227/state: {"dark":false,"lastupdated":"2018-02-06T21:41:18","presence":true}
...
Feb 06 22:42:18 pi2 dc_eventlog[8946]: /sensors/227/state: {"dark":false,"lastupdated":"2018-02-06T21:42:18","presence":false}

```

bjoernhoefer, 2018-02-06:
1. So I can turn on and off the lights without an extra gateway - cool.

1. ok, so I'll stay at my Xiaomi motion sensors... they are cheaper and do the same :-)

1. Additional Question: Do you know any cheap alternatives for "monitoring" the brightness (cheaper than the Hue motion sensor) - motion not necessarily needed.

Thanks for the fast response.

ebaauw, 2018-02-06:
No, I don't know any standalone ambient light sensors.

jsponz, 2018-02-15:
Hi,

is there any place to find how to pair the IKEA motion sensor with the Raspeegw? 

Also for the remote control.  I can't find any guideline :-( 

Thanks!

manup, 2018-02-15:
For that please use the *Phoscon App > Hamburger Menu > Devices > Sensors > Add Sensor*
There you find a wizard to pair the IKEA motion sensor.

You can open the Phoscon App from deCONZ or by opening it in a Browser `http://<Raspberry-IP>/pwa`

jsponz, 2018-02-16:
Fast and direct. 

Thanks!

tpetri, 2018-02-16:
> Additional Question: Do you know any cheap alternatives for "monitoring" the brightness (cheaper than the Hue motion sensor) - motion not necessarily needed.

The xiaomi aqara motion sensors do report the brightness don't they?

bjoernhoefer, 2018-02-17:
Thought that too, but no - they can't. At least the two I own only report motion. maybe there are different Models but mine are the latest, as far as I knie.

tpetri, 2018-02-17:
There are two versions - the older Mi series without luxmeter and the newer Aqara (comes with a turnable mounting stand) - you can also see the small light sensor hole on top of the motion sensor cover, and they have now a push button for resetting instead of the annoying needle-hole. Both are basically equally expensive by now, so anyone buying such a sensor might want to prefer the Aqara version.

manup, 2018-02-17:
Also be aware the Aquara sensors just report brightness when motion is detected, unlike the Philips motion sensor which reports measured values periodically.