dresden-elektronik/deconz-rest-plugin #399: Sensors are lost after reboot

belmondo2001, 2018-02-07:
I have some Xiaomi Motion Sensors which work wonderfully.
But after reboot or software update the sensors are lost.

I checked the zll.db and the sensors are not in the sensor table
after i have them  paired.

with lights and switch all is ok.

i tried with deconz-2.05.01 and deconz-2.04.99.

What did i wrong or is this an bug ?

mfg.
Juergen

manup, 2018-02-07:
A reboot should not result in such data loss, if the sensors aren't in the zll.db anymore it seems that the file wasn't written to filesystem, or sd-card may have been corrupted and the file couldn't be restored properly.

> I checked the zll.db and the sensors are not in the sensor table
after i have them paired.

Usually the zll.db file is written shortly after a sensor is added.

It's hard to pinpoint the exact cause but you may check:

- that the sd-card is ok with fsck
- use a class 10 or better sd-card
- always shutdown and reboot the raspberry properly from UI or commandline (never just remove from power)

belmondo2001, 2018-02-07:
I do not think that the zll.db is corrupt,
because i copy the file to another pc and then check the tables 
with a database tool.
if the file where corrupt a think i am not able to copy or select the contents of the database without errors,

New lights are stored in the database.

Is there a logfile whre i can watch what happens after pairing ?

mfg.
Juergen

manup, 2018-02-07:
The file might not be partially corrupted but we have seen cases were the file was reverted to older version after reboot. Still a strange issue haven't seen it in a while with newer sd-cards.

There is no log file but you can enable debug output by starting deCONZ on the commandline:

`$ deCONZ --dbg-info=1 --dbg-error=1 --http-port=80`

This should also print when the database is written

belmondo2001, 2018-02-07:
thanks,
i will try that.
if i install deconz on a new sd.card,
can i restore my lights and my config from the database or
how could i get my lights without repairing ?.

manup, 2018-02-07:
Easiest way is to copy the`~/.local/share/data/dresden-elektronik/` folder to the new card. The basic ZigBee network parameters like channel and network id are also kept on the RaspBee.

belmondo2001, 2018-02-07:
Problem found.
the files under /home/pi/.local/... are owned by user root,
but the deCONZ process runs as 'pi'.
after changing this no problems anymore.

closed 2018-02-07 by belmondo2001

sam-75, 2020-08-06:
hi, I have no share within .local
Any tips where the data is based on the SD card

Mimiix, 2020-08-06:
@sam-75 You do, but it probably is hidden.

If you still can't find it, feel free to open a new issue instead of using this one. Makes life easier 😄

sam-75, 2020-08-06:
no such file or directory for 
/home/pi/.local/share/data/dresden-elektronik/
after .local it ends - it is empty

Mimiix, 2020-08-06:
@sam-75 If you are running the HA addon, you have to go in the docker container.

sam-75, 2020-08-06:
ok i activated this later
whre is this in ?

sam-75, 2020-08-06:
sorry I had HA on for iobroker not for deconz ... so there should no docker contrainer

Mimiix, 2020-08-06:
I have no clue then :/ Best to open a issue with them.

sam-75, 2020-08-06:
ok

Smanar, 2020-08-06:
On my side it s /.local/share/dresden-elektronik/deCONZ