dresden-elektronik/deconz-rest-plugin #400: Ubuntu packages are missing

jurriaan, 2018-02-07:
The ubuntu equivalents of the following packages are missing:

http://www.dresden-elektronik.de/rpi/deconz/beta/deconz-2.05.01-qt5.deb
http://www.dresden-elektronik.de/rpi/deconz-dev/deconz-dev-2.05.00.deb
http://www.dresden-elektronik.de/rpi/deconz-dev/deconz-dev-2.05.01.deb

Can these be updated?

I'd like to contribute to this repo, but without the correct -dev packages it is not really possible.

manup, 2018-02-07:
Sorry for delay, Ubuntu packages are up now:

https://www.dresden-elektronik.de/deconz/ubuntu/beta/

jurriaan, 2018-02-07:
@manup Sorry to bother you again, but https://www.dresden-elektronik.de/deconz/ubuntu/beta/deconz-2.05.01-qt5.deb is a armv7h build, not a x86_64 one.

```
$ file usr/bin/deCONZ
usr/bin/deCONZ: ELF 32-bit LSB executable, ARM, EABI5 version 1 (GNU/Linux), dynamically linked, interpreter /lib/ld-linux-armhf.so.3, for GNU/Linux 2.6.32, BuildID[sha1]=2200c633b705cc03391cf61d30e14d9b952337db, not stripped
```

manup, 2018-02-07:
Ha damn, I need more coffee :) copied from the wrong folder..

should be the correct one now.

jurriaan, 2018-02-08:
Thanks a lot!

closed 2018-02-08 by jurriaan