dresden-elektronik/deconz-rest-plugin #401: can I add other devices working with the  ZigBee Home Automation profile to deConz?

marcogaudino, 2018-02-08:
Hi guys, 
i'm working from few days with deConz. can i visualize in deConz other device instead of the only lights?

I would like to visualize in the software, devices like door lock, smart socket and so on. 

Can someone help me? Thank you very much. 

Marco

manup, 2018-02-24:
Yes there are some sensors and switches from larger brands supported as well, do you have any specific in mind?

marcogaudino, 2018-02-24:
Yes, Yale door lock

manup, 2018-02-24:
Currently there is no support for this device it might be added if related debug info can be provided.

florischabert, 2018-03-01:
Hey @manup, I'm also interested in using a Yale door lock with deconz. Looks like supporting the zigbee lock cluster requires some deeper changes, right? What debug info would you need?

davidlzs, 2018-08-12:
@manup, I got Yale lock, yale Zigbee module, RPi3B+, RaspBee. Would you let me know what kind of debug need to gather? And how?

ebaauw, 2018-11-07:
See 
https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Request-Device-Support

manup, 2019-01-17:
Closing this for now, as suggested please open a device specific issue with the needed details.

closed 2019-01-17 by manup