dresden-elektronik/deconz-rest-plugin #404: deCONZ enhancements for IAS devices

ebaauw, 2018-02-09:
I've got a whole bunch of Heiman sensors on loan from @Oggymator, to integrate into deCONZ and homebridge-hue (see https://github.com/dresden-elektronik/deconz-rest-plugin/issues/263#issuecomment-362527633).  They all connect to deCONZ without any issue, and I'm analysing them through the web GUI and ZigBee sniffing.  They're _IAS Zone_ devices, except for the temperature/humidity sensor, which is a _Temperature Sensor_ and the Siren, which is an _IAS Warning Device_).

I'm trying to figure out the best way to support these in deCONZ, see also earlier discussions in #152, #269, and #352), and would like your views, @manup, @rtenklooster, and, of course, any-one else.

1. The _IAS Zone_ devices obviously provide an _IAS Zone_ cluster.  They send _Zone Status Change Notification_ commands when an alarm occurs _and_ when it's cleared.  They set the _Restore reports_ bit in the _Zone Status_.  deCONZ could/should take this as a hint not to use `config.duration` to clear the alarm.
2. The sensors provide their type through  _IAS Zone Status_, but this is rather coarse grained.  Both the combustable gas and the carbon-monoxide (CO) sensor report _Gas sensor_.  This means, any REST API client (i.c. homebridge-hue) would still need to check the `manufacturername` and `modelid`, even if deCONZ would provide a `zonetype` attribute.  Would it still make sense to provide `zonetype`?
3. Next to _Alarm 1_, the sensors also set/clear the _Tamper_ and _Battery_ bits in the  _Zone Status_.  The sensors contain a tamper button which gets depressed when you remove them from their base plate, and I had some almost depleted AAA batteries lying around.  I think it makes sense to expose these through deCONZ as well.  deCONZ could introduce a sensor `type` of `ZHAStatus` which provides the _Zone Status_ bitmap as `state.status`.  `CLIPGenericStatus` can be used to simulate a `ZHAStatus` sensor.  Alternatively, deCONZ could introduce a sensor `type` of `ZHAAlarm`, with separate Boolean attributes `state.alarm`, `state.tamper`, `state.battery`.  We'd need a corresponding `CLIPAlarm` to simulate these.
4. It makes sense to map a _Contact switch_ to a `ZHAOpenClose` sensor resource, and a _Motion sensor_ to a `ZHAPresence`, but what about the additional _Zone Status_ bits?  If we go for `ZHAAlarm`, we could expose `state.tamper` and `state.battery` to `ZHAOpenClose` and `ZHAPresence` as well.  `zonetype` wouldn't be needed here, as it's implied by `type`.
5. The Smoke, CO, and Gas sensors, as well as the Siren have an audible alarm.  The Siren and Smoke sensors advertise an _IAS WD_ cluster (0x0502).  The CO sensors seems to have the same firmware as the Smoke sensor, so I suspect it to have this cluster as well.  It's not defined in `general.xml` - I'll try and add it and see what will happen.  The ZCL defines an unnamed map8 attribute, which combines multiple sub-attributes: _Warning Mode_, a 4-bit enum, _Strobe_, a boolean, and _Siren Level_, a 2-bit enum.  I don't suppose `general.xml` will be able to support that?
6. If I follow the principle that `/sensors` provides resources for sensing devices, and `/lights` provides resources for actuators, deCONZ would expose the sirens under `/lights`.  Not only would that be unexplainable, we'd also run into the issue that the Smoke and CO sensors are ZigBee end-devices (the Gas sensor and Siren are mains powered and ZigBee routers).  I think we'd better expose this cluster as a `/sensors` resource?
7. ZCL defines two commands for _IAS WD_, _Start Warning_ and _Sqawk_, and one attribute, _MaxDuration_.  The latter can be exposed through `config.duration` (would introduce `config` to `/lights`).  The commands could be exposed through a combination of `state.alert` (for warnings: `"stop"`, `"burglar"`, `"fire"`, `"emergency"`, `"police panic"`, `"fire panic"`, and `"emergency panic"`) and `state.effect` (for strobe effects: `"strobe"`).  We could use `state.transitiontime` for _Warning duration_ or provide a new `state.duration`.  For _Siren level_ and _Stobe level_ we would need new `state` attributes (`"low"`, `"medium"`, `"high"`, `"very high"`), as for _Strobe Duty Cycle_ (int8).  At the moment this is academic - I still need to figure our what the devices support.  They make a hell of a noise, so this might take some time, as I need to recover my hearing after each test...
8. The temperature/humidity sensor exposes two _Temperature sensor_ endpoints, one with the _Temperature sensing_ cluster, and one with the _Humidity sensing_ cluster.  Both endpoints contain a basic cluster, but with a different value for _Model_: `TH-T_V15` vs `TH-H_V15`.  No big deal, I simply whitelisted both values, but this is the first time I've seen this.
9. The (ZigBee module for the) Siren and the Plug seem to be made by a different manufacturer, using MAC prefix of 0x000d6f (ember) and a _Manufacturer Code_ of 0x120b and 0x1002.  Where can I lookup these codes?
10. The plug provides _Simple Metering_ (0x0702) and _Electrical Measurement_ (0x0B04) clusters.  My OSRAM plugs only provide _Electrical Measurement_ - I think it best to expose only this cluster.  That would mean a new `/sensors` resource `type` ZHAPower` with a `state.power` attribute?  Or would we simply expose `state.power` as additional attribute to the `/lights` resource?  The OSRAM plug reports value 28000 when no power is drawn instead of 0, so we might need access to the `state.on` attribute to map 28000 it to 0 when `state.on` is false.  The plug is rated for over 3000W, so I suspect 28000 could be a valid value as well.
11. The value of the _Active Power_ attribute (0x050B) seems to be in deciWatt, rather than Watt, as specified by the ZCL.  Maybe `general.xml` is just missing the definitions for attributes ACPowerDevisor (0x0605) and ACPowerMultiplier (0x0604) - I need to check.
12. In `de_web_plugin.cpp` IAS_ZONE_CLUSTER_ID and OCCUPANCY_SENSING_CLUSTER_ID are combined in the same `case` block, followed by an `if` to split them: https://github.com/dresden-elektronik/deconz-rest-plugin/blob/475c614a33335e5fb3e8b3b383b42876dd7c2ccc/de_web_plugin.cpp#L2651-L2659
  I want to split these into two `case` blocks, but where should I place the code for the Centralite Motion Sensor-A?  Does it expose an _IAS Zone_ or an _Occupancy Sensing_ cluster?

manup, 2018-02-09:
Wow cool, that's a very helpful detailed list. For some points I'm not sure yet how to handle these cases, but here are some thoughts:

>1. The IAS Zone devices obviously provide an IAS Zone cluster. They send Zone Status Change Notification commands when an alarm occurs and when it's cleared. They set the Restore reports bit in the Zone Status. deCONZ could/should take this as a hint not to use config.duration to clear the alarm.

The double usage of `config.duration` becomes more confusing to me. Maybe we should use something more explicit for the purpose of clearing after inactivity like `config.autoclear` which accepts integer or 0 (off)?

> 2. The sensors provide their type through IAS Zone Status, but this is rather coarse grained. Both the combustable gas and the carbon-monoxide (CO) sensor report Gas sensor. This means, any REST API [...]

I think this also falls into the `/devices` endpoint consideration (see https://github.com/dresden-elektronik/deconz-rest-plugin/issues/269 to which I will write some thoughts later soon). 

Where possible I would not expose too much ZigBee specifics to the REST-API, devices too often provide 'bad' attributes which are not useful for the API user. Like in this case if we know for specific manufacturername and modelid the details we can provide a meaningful attribute like `gas` or more generic `measuredvalue` and provide a `valuetype` like 'combustable gas' or 'carbon-monoxide gas'. 

> 3. Next to Alarm 1, the sensors also set/clear the Tamper and Battery bits in the Zone Status. The sensors contain a tamper button which gets depressed when you remove them from their base plate, [...]

>Alternatively, deCONZ could introduce a sensor type of ZHAAlarm, with separate Boolean attributes state.alarm, state.tamper, state.battery. We'd need a corresponding CLIPAlarm to simulate these.

Agree, the Climax sensors also provide tamper and battery alarm flags which are currently not exposed. I would prefer the state.alarm, state.tamper, state.battery approach since it's on a higher level and doesn't expose the ZigBee internals.

> 4. It makes sense to map a Contact switch to a ZHAOpenClose sensor resource, and a Motion sensor to a ZHAPresence, but what about the additional Zone Status bits? If we go for ZHAAlarm, we could [...]

ACK :)

> 5. The Smoke, CO, and Gas sensors, as well as the Siren have an audible alarm. The Siren and Smoke sensors advertise an IAS WD cluster (0x0502). The CO sensors seems to have the same firmware as the [...]

> The ZCL defines an unnamed map8 attribute, which combines multiple sub-attributes: Warning Mode, a 4-bit enum, Strobe, a boolean, and Siren Level, a 2-bit enum. I don't suppose general.xml will be able to support that?

Yes, general.xml is currently not suitable for that, handles only classic bits and enums.

> 6. If I follow the principle that /sensors provides resources for sensing devices, and /lights provides resources for actuators, deCONZ would expose the sirens under /lights. Not only would that be [...]

Related to https://github.com/dresden-elektronik/deconz-rest-plugin/issues/269

>9. The (ZigBee module for the) Siren and the Plug seem to be made by a different manufacturer, using MAC prefix of 0x000d6f (ember) and a Manufacturer Code of 0x120b and 0x1002. Where can I lookup these codes?

To be honest I haven't found a official list for those. I did just google the codes in the past.

> 10. The plug provides Simple Metering (0x0702) and Electrical Measurement (0x0B04) clusters. My OSRAM plugs only provide Electrical Measurement - I think it best to expose only this cluster. That [...]

I like the exposestate.poweras approach to keep it simple.

>11. In de_web_plugin.cpp IAS_ZONE_CLUSTER_ID and OCCUPANCY_SENSING_CLUSTER_ID are combined in the same case block, followed by an if to split them:

>I want to split these into two case blocks, but where should I place the code for the Centralite Motion Sensor-A? Does it expose an IAS Zone or an Occupancy Sensing cluster?

When I remember correctly it provides both clusters, I need to check it.

ebaauw, 2018-02-09:
7. Hacked something into general.xml, see attached [general.xml.gz](https://github.com/dresden-elektronik/deconz-rest-plugin/files/1711111/general.xml.gz).  I can now control the siren from the deCONZ GUI.  
[image]
  I can set _Max duration_ and issue a _Start warning_.  It doesn't seem to act differently for different values of _Siren_, _Strobe_ and _Squawk level_.  It only seems to distinguish between _Warning Mode_ 0b0000 (stop) and other values (as in silent vs scaring the shit out of me).  It seems to support _Strobe_, _Warning duration_, and _Strobe duty cycle_ in full.  Also the _Sqawk_ command incl. _Strobe_ and _Sound for disarmed_ seem to work.
  The _IAS WD_ on the Smoke sensor seems inoperable - I cannot get it to do anything, so even setting _Max duration_, while poking the reset button to wake the sensor.
  @rtenklooster, could you see if this works for you as well?

10. While at it, added the multiplier and divisor (as well as the Smart plug device type) to `general.xml`.  Indeed, deciWatts.
[image]
  Interestingly, it doesn't advertise that it supports anything in attribute 0x0000, but 0x050b changes when power is drawn.  It doesn't support Reporting on this attribute, though, so it needs to be polled (another reason to keep treating it as a light).  Same for the OSRAM plugs and my ubisys dimmer.  Haven't yet installed the new general.xml to my production environment, so don't yet know what these report for 0x0605.

manup, 2018-02-11:
Nice, I wonder how the sound of this siren is :)

Shall I merge your gerneral.xml into the next release?

Note to myself: move general.xml from core into the rest-plugin git repo.

ebaauw, 2018-02-11:
> Nice, I wonder how the sound of this siren is :)

It's really loud, you can probably hear it in Dresden when I'm sounding it in Amsterdam.

> Shall I merge your gerneral.xml into the next release?

Did some more changes (all the _Electrical Measurement_ attribute that the ubisys dimmer exposes).
[general.xml.gz](https://github.com/dresden-elektronik/deconz-rest-plugin/files/1714539/general.xml.gz)

> Note to myself: move general.xml from core into the rest-plugin git repo.

Cool!  Is there a reason it uses \r\n as newlines?  Or is it just the editor you're using?  Maybe consider changing it to a json file for easier maintenance?

rtenklooster, 2018-02-11:
Okay. Just upgraded to the latest version. Used your general.xml.
The Xiami Smart power plug:
[image]
[image]

So the power consumption is working great. It's changing instantly.

ebaauw, 2018-02-11:
Huh?  It's reporting power consumption though the Analog Input cluster?  Which endpoint?  In Watts?

rtenklooster, 2018-02-11:
Yes it is.
02 -> 000C
And yes, it's in watts. I have double checked the readings from within the mi home app.
Exactly the same values.

The 03-000C reports zero. This could be the counter for daily consumption, or the grand total. As my mi home app reports those two fields as zero.

AntiHeld889, 2018-02-15:
is it possible to add the power consumption in homebridge-hue-plugin? to show the consumption in the eve App?

ebaauw, 2018-02-16:
Yes, but we need to support it in deCONZ first.

ebaauw, 2018-02-18:
Everything should be supported in v2.05.04, see https://github.com/dresden-elektronik/deconz-rest-plugin/pull/420, except for the Siren and for setting the siren duration in the Fire sensor.

Some updates:
7. The Fire sensor allows setting _Max Duration_ through the _IAS WD_ cluster.  In theory, this could be exposed as `config.maxduration`, and set through `config.pending`, but I see little value and a whole lot of work;
8. The Heiman _does_ support attribute reporting for _Active Power_.  The OSRAMs don't.  Also, they don't support the _Mulitplier_ nor _Divisor_ attributes.  They report 28000 when no power is drawn.

For further discussions on `state.power`, please see #423.

ebaauw, 2018-02-18:
> is it possible to add the power consumption in homebridge-hue-plugin? to show the consumption in the eve App?

homebridge-hue@0.6.5 in combination with deCONZ 2.05.04.  Note that the OSRAM plugs report nothing useful.

wvuyk, 2018-02-21:
It seems the xiaomi power plug is not yet forwarded to the REST API? Will this be possible?

ebaauw, 2018-02-21:
I think it will be, but, being Xiaomi, they report power consumption in a non-traditional way, see https://github.com/dresden-elektronik/deconz-rest-plugin/issues/153.

wvuyk, 2018-02-21:
It is not yet visible in the rest api, maybe I was unclear? the power consumption reporting would be a nice to have with it, not major priority on that yet.

to0b, 2018-02-22:
I’ve added successfully a heiman smoke sensor to deCONZ. When does the tempered indicator occur? I’ve released the sensor from the base plate but the tempered flag doesn’t change.

ebaauw, 2018-02-22:
Same here (and same for the CO sensor).  I thought the small button being pressed when the sensor is attached to the base plate would be a tamper detection switch (like for the door and for the water sensor), but it doesn't seem to do anything.  It's also held down by the battery, but if you remove the battery the sensor goes dead.

I'm not sure if the lowbattery alarm works for the smoke and for the CO sensor either - I don't have an almost depleted battery of this type to test.

to0b, 2018-02-22:
OK thanks for the information. When there are a few smoke sensors connected and one sensor detects smoke, do the others make also noise?

ebaauw, 2018-02-22:
I don't know for sure, I only have one on loan.  I don't think they can do that, though - I haven't been able to find a way to activate the siren (or flash) remotely.

Oggymator, 2018-02-23:
talked today with Heiman, not possible to activate the siren by remote.

ebaauw, 2018-02-27:
@manup I'm thinking about maybe exposing the Siren as a light, using some new values for `state.alert` to activate the siren, strobe, or both.  You can set these, but the Siren doesn't report the actual status back, so this would seems like the most appropriate match.  However, would it be wise (or even possible) to create a light without `state.on`?  I could imagine many applications not liking that.  Also, there's a max duration attribute, which would better be exposed through `config`, which we don't have for lights.  Maybe a ZHAAlert with settable `state.alert` would be more appropriate?  The `state` attribute of CLIP sensors can also be updated through the REST API, so it's not completely unheard of.  It wouldn't expose `state.lastupdated` though.

manup, 2018-02-27:
>However, would it be wise (or even possible) to create a light without state.on? I could imagine many applications not liking that.

Yes I imagine some applications might even crash assuming the state.on attribute to be there.

I'm still very lean towards to this */devices* endpoint thing, heres an example which might reflect the siren:

     GET /devices

```json
    {
        "00:21:2e:ff:ff:00:11:80-07-0500" : {

            "name": "Bob",
            "type": "Warning device",
            "modelid": "SIR-3000",
            "manufacturername": "Lego",
            "state": {
                "alert": "strobe",
                "reachable": true,
                "lastupdated": "2018-02-27T19:42:19"
            },
            "config": {
                "maxduration": 300
            }
        }
    }
```

To sum it up in C++ it would be:

* RActor (?)
* RAttrUniqueId, RAttrName, RAttrType, RModelId, ...
* RStateAlert, RStateLastUpdated
* RConfigMaxDuration

ebaauw, 2018-02-27:
> The double usage of `config.duration` becomes more confusing to me. Maybe we should use something more explicit for the purpose of clearing after inactivity like config.autoclear which accepts integer or 0 (off)?

Yes it's confusing, mea culpa, at least partly.  There's different ZigBee mappings for `config.duration`, see also #317.  In historical order:
1. No ZigBee mapping at all (e.g. motion sensors based on IAS Zone, and CLIPPresence sensors);
2. The _PIR Occupied to Unoccupied Delay_ attribute of the _Occupancy sensing_ cluster (e.g. Hue motion sensor);
3. The _On Time_ parameter of the _On with timed off_ command of the _OnOff_ cluster (e.g. Trådfri motion sensor);

The other confusing part is that there's two different types of motion sensors:
- A. Those that only report motion detected (e.g. the Trådfri motion sensor, motion sensors based on IAS Zone, without the _Restore reports_ set in the _Zone status_);
- B. Those that report both motion detected and motion no longer detected (e.g. the Hue motion sensor, and motion sensors based on IAS Zone, with _Restore reports_ set in the _Zone status_ like Heiman);

Combining these two aspects, we get the following cases:
- A1 (motion sensors from before my time (I assume), CLIPPresence where `state.presence` isn't reset explicitly, Trådfri motion sensor, when controlling lights through rules): deCONZ should reset `state.presence` to false automatically, after `config.duration` seconds;
- A3 (Trådfri motion sensor, when controlling lights directly using `config.group`): The lights will turn themselves off after _On Time_ seconds (configured through the dial on the sensor).  Arguably, deCONZ should turn off `state.presence` at the same time to match the light states.  To enable this, I (ab)used `config.duration` to reflect _On Time_, making it read-only;
- B1: (Heiman motion sensor, CLIPPresence where `state.presence` is reset explicitly): `config.duration` only exists in the REST API.  Arguably, deCONZ should delay setting `state.presence` to false, `config.duration` seconds after it receives the report that motion is _no longer_ detected.  Alternatively, don't expose `config.duration` for these type of sensors, set `state.presence` to false on receiving the report that motion is no longer detected, and tell people to use a rule with a `ddx` condition instead;
- B2: Hue motion sensor (B2): `config.duration` is mapped to _PIR Occupied to Unoccupied Delay_ causing the sensor to delay the notification of _Occupancy_ false, after it no longer detects motion.  deCONZ should set `state.presence` to false when it receives the report that _Occupancy_ is 0, irrespective of the value of `config.duration`.

I think it makes a lot of sense to:
- Use a different `config` attribute, let's call it `delay`, for B2 (read-write) and A3 (read-only);
- Reserve `config.duration` for automatically resetting `state.presence` to false after receiving the motion detected report;

This would lead to the following:
- Hue motion sensor: only `config.delay`, no automatic resetting of `state.presence` to false;
- Trådfri motion sensor: both `config.delay` and `config.duration`.  If `config.duration` > 0, reset `state.presence` to false after `config.duration` seconds, otherwise reset `state.presence` to false after `config.delay` seconds;
- Others (incl. CLIPPresence): only `config.duration`. If `config.duration` > 0, reset `state.presence` to false after `config.duration` seconds, and ignore any motion no longer detected reports; otherwise honour the motion no longer detected report and no automatic resetting of `state.presence` to false.

manup, 2018-02-27:
> * Use a different config attribute, let's call it delay, for B2 (read-write) and A3 (read-only);

B2 includes two more devices working similar as the hue motion sensor by using *PIR Occupied to Unoccupied Delay* &mdash; the FLS-NB1 and FLS-NB2 these are special combined (mains-powered) sensor-lights which also ZCL report the presence state.

If configured via binding to a group they further send *On with timed off* commands like the IKEA sensor, but here the *OnTime* parameter is dynamic and contains the remaining time, since the command is send every x seconds, it should be ignored and only `state.presence` will be set to true, if not already set to true by a prior report. The FLS-NBx devices are already fully implemented it's very important that they keep working when we modify the implementation. :)

The extra `config.delay` attribute is a good idea we should do that. The recent implementation does reset `state.presence` only if the ZCL reports aren't received for some time (based on max report interval) so it shouldn't interfere in normal operation mode.

> * Trådfri motion sensor: both config.delay and config.duration. If config.duration > 0, reset state.presence to false after config.duration seconds, otherwise reset state.presence to false after config.delay seconds;

Agree, that should work for all uses cases we had seen in the past for the sensor.

> * Others (incl. CLIPPresence): only config.duration. If config.duration > 0, reset state.presence to false after config.duration seconds, and ignore any motion no longer detected reports; otherwise honour the motion no longer detected report and no automatic resetting of state.presence to false.

Agree

jsponz, 2018-03-03:
Hi,

I have just bought the Siren module. Could you help me with the pairing?

Thanks a lot!
Jorge

ebaauw, 2018-03-03:
You need to reset the Siren by (I think) pressing and holding the reset hole for 5+ seconds.  The SIren should react to indicate it's been reset.  Then open the network from the web app or Phoscon, and briefly press the reset hole on the Siren.  It should appear in the GUI.

For now, you can only control the Siren from the GUI.  The REST API creates a ZHAAlarm and a ZHAWarning resource, but neither of these does anything yet.

jsponz, 2018-03-03:
Thanks! Quick answer. 
Where should it appear? Should I click on "Add new..."

oleg77s, 2018-03-03:
@ebaauw , what can you say about Heiman HS1RC Smart Remote Control
[image]

and Heiman HS1EB Smart Emergency Button ?
[image]

Oggymator, 2018-03-03:
I don’t like these remotes, battery life is bad.

Normally it is used for the Heiman Secrurity Gateway to arm and disarm without smartphone.

Don’t think is very useful for HomeKit

ebaauw, 2018-03-03:
> what can you say about Heiman HS1RC Smart Remote Control and Heiman HS1EB Smart Emergency Button?

I'd expect these to be IAS Zone devices, just like most of the other Heiman sensors.  If you can provide me screenshots, adding support shouldn't be too difficult.  Probably best to expose them as ZHASwitch and issue appropriate buttonevents for press/release of each/the button.

> Where should it [the Siren] appear? Should I click on "Add new..."

Yes, Add new... in the Phoscon app.  It will not appear there, though.  You need to check the deCONZ GUI and/or the RESP API for a new ZHAAlarm and ZHAWarning resource.

wvuyk, 2018-03-08:
I have added the Xiaomi smoke sensor today here. Am I right assuming there is no battery information yet on this Xiaomi device?

ebaauw, 2018-03-08:
Looking at the code, deCONZ (v2.05.10 or later) should create `config.battery` for the Xiaomi smoke sensor as well.  I think/hope @manup cracked the Xiaomi magic enough to get the battery value for all Xiaomi devices that report it.

wvuyk, 2018-03-08:
So maybe it is not reporting it yet here... At least the battery value is not changed from 0, I would doubt it is  still 0. Installed it this morning, normally the value gets populated after a while?

`{
"ep": 1,
"etag": "8f7ee3071feaa4288a5f1bd82b2e538b",
"state": {
"lastupdated": "2018-03-08T08:47:17",
"fire": false,
"lowbattery": false,
"tampered": false
},
"config": {
"on": true,
"battery": 0,
"reachable": false
},
"name": "Brand alarm",
"type": "ZHAFire",
"modelid": "lumi.sensor_smoke",
"manufacturername": "LUMI",
"uniqueid": "00:15:8d:00:01:d7:05:20-01-0500"
}`

ebaauw, 2018-03-08:
Maybe the smoke sensor works differently.  Does `config.battery` get populated when you press the reset button?  Do you see messages in the deCONZ log (I think these are --dbg-info=1):
```
Mar 08 20:29:40 pi2 deCONZ[22580]: 20:29:38:988 0x00158D0001xxxxxx skip Xiaomi attribute 0x0005
Mar 08 20:29:40 pi2 deCONZ[22580]: 20:29:38:988 0x00158D0001xxxxxx extract Xiaomi special
```

ebaauw, 2018-03-08:
`journalctl -au deconz` or `journalctl -au deconz-gui`, depending on which service you're running.

wvuyk, 2018-03-08:
There is not much helpfull in there Eric... 

`Mar 08 19:46:16 raspberrypi systemd[1]: [/etc/systemd/system/deconz-gui.service:12] Unknown lvalue 'AmbientCapabilities' in
Mar 08 19:46:16 raspberrypi systemd[1]: Starting deCONZ: ZigBee gateway -- GUI/REST API...
Mar 08 19:46:16 raspberrypi systemd[1]: Started deCONZ: ZigBee gateway -- GUI/REST API.
Mar 08 19:46:18 raspberrypi deCONZ[1323]: libEGL warning: DRI2: failed to authenticate
`

ebaauw, 2018-03-10:
Looks like you’re running deCONZ without any of the debuging enabled.  You need to add `--dbg-info=1` to the service definition file, `/etc/systemd/system/deconz-gui.service`.  That would involve some serious Linux interaction...

ebaauw, 2018-03-11:
Added support for the Heiman siren.  It's exposed as a `/lights` resource (for the _IAS_WD_ cluster 0x0502), and as a `/sensors` resource (for the _IAS_Zone_ cluster, 0x0500).

The light is for controlling the siren, using the following values for `state.alert`:
- `"none"`: turn off the alarm:
- `"select"`: turn on the siren and strobe light for 1 second;
- `"lselect"`: turn on the siren and strobe light for 5 minutes;
- `"blink"`: turn on the strobe light for 5 minutes.

Note that you can limit the maximum alarm time by writing the _Max Duration_ attribute in the _IAS WD_ cluster through the deCONZ GUI.  This attribute isn't exposed by the REST API.

The sensor is mainly for reporting the level of the (backup) battery and checking that the siren is reachable.  It does send supervision reports after manually enrolling it in the deCONZ GUI, but I haven't seen any alarm (not even a low battery alarm when the siren was disconnected from mains power and backup battery was down to 4%).

For reference, here's the light resource:
```json
{
  "etag": "830aad9771beb8394cad8cd4c63f8624",
  "hascolor": false,
  "manufacturername": "Heiman",
  "modelid": "WarningDevice",
  "name": "Siren",
  "state": {
    "alert": "none",
    "reachable": true
  },
  "swversion": null,
  "type": "Warning device",
  "uniqueid": "00:0d:6f:00:0b:xx:xx:xx-01"
}
```
And here's the sensor resource:
```json
{
  "config": {
    "battery": 100,
    "on": true,
    "reachable": true
  },
  "ep": 1,
  "etag": "36b57b448301a606bdd5b80422232b94",
  "manufacturername": "Heiman",
  "modelid": "WarningDevice",
  "name": "Siren",
  "state": {
    "alarm": false,
    "lastupdated": "2018-03-11T14:47:33",
    "lowbattery": false,
    "tampered": false
  },
  "swversion": "2017.9.27",
  "type": "ZHAAlarm",
  "uniqueid": "00:0d:6f:00:0b:xx:xx:xx-01-0500"
}
```

Some other remarks:
- The `swversion` for the sensor is actually the _Date Code_; for the light it's empty (`null` instead of `""`, I think that's part of the Daylight changes).  @manup, I think the logic to use the _Date Code_ when _SW Build ID_ isn't supported is still missing from lights?
- The siren reports value 0x81 for the _Power Source_ (0x0007) attribute of the _Basic_ cluster.  This value is missing from general.xml.  @manup, can you please add the following line:
```xml
  <value name="Mains (single phase) with battery backup" value="0x81"></value>
```
- I removed `state.on` from the light resource, as it's not doing anything.  I'm not sure if all REST API clients will handle a light without `state.on` correctly.

manup, 2018-03-12:
> The swversion for the sensor is actually the Date Code; for the light it's empty (null instead of "", I think that's part of the Daylight changes). @manup, I think the logic to use the Date Code when SW Build ID isn't supported is still missing from lights?

Yes needs to be added, when I recall correctly ubisys also doesn't support that or returns a empty string.

> <value name="Mains (single phase) with battery backup" value="0x81"></value>

It's in 2.05.12

> I removed state.on from the light resource, as it's not doing anything. I'm not sure if all REST API clients will handle a light without state.on correctly.

Shouldn't be a problem I guess it wasn't used much or at all.

ebaauw, 2018-03-12:
> when I recall correctly ubisys also doesn't support that

Correct.

> It's in 2.05.12

Yeah, saw that.  Thanks!

> Shouldn't be a problem I guess it wasn't used much or at all.

`state.on` for a _light_, not `config.on` for a sensor.  The latter still needs to be implemented, by the way (blocking any update to state attributes).  I don't use it myself, but I do expose it to HomeKit.  It's being used by quite a few of my users, e.g. to disable a Hue-bridge connected Hue motion sensor sensor at night.  I use a CLIPGenericStatus for that, in combination with some gateway rules, so the motion still registers, but the lights don't turn on.

manup, 2018-03-12:
Ahh I see thought it was `config.on` yes this might break some apps since they almost always expect the on attribute. I need to check Phoscon App :) If it causes issues in other apps a dummy `state.on` should be fine.

oleg77s, 2018-04-05:
@ebaauw 
Heiman HS1RC Smart Remote Control
[image]
[image]
[image]
[image]
[image]

ebaauw, 2018-04-25:
Could you please also post screenshots of the node and its clusters and of the _Node Info_ panel?

ZacGrose, 2018-04-25:
Sure — attached below.
[image]
[image]

ebaauw, 2018-04-25:
Thanks.  Sensors need to be whitelisted.  As the siren is battery powered, it needs to be whitelisted for the light resource as well.  I’ll have a look this weekend.

closed 2018-06-12 by ebaauw

ithinkcode, 2019-03-14:
@ebaauw I recently got my hands on a few conbees and i am in the process of moving my zigbee2mqtt deployment over. Any thoughts on the Heiman Motion sensor performance ? I have been using Xiaomi Mijia sensors and they have a blind spot of about 2 mins during which new motion is undetected ? Do you know if Heiman suffers from this issue as well ? Thanks in advance!

ebaauw, 2019-03-14:
Sorry, I don’t know.

nebukadezza, 2019-06-12:
Dear ebaauw, all, had any of you paired a Heiman HS1CA-N monoxide sensor with deCONZ succesfully? Meanwhile sensor search is active in Phoscon I enabled network search on the monoxide sensor but it's not visible on the UI nor the sensor indicates any found host. Same results if the sensor is restarted by battery removal, not doing anything with it during the search. Thank you for comments in advance, Cheers

xhemp, 2019-11-18:
@ZacGrose Are you still using your SwannOne siren? Is it working as intended? I am having some issues with mine, just wondering if we can chat about it. Thanks!