dresden-elektronik/deconz-rest-plugin #405: Announcement: ZSHARK - Wireshark Sniffer for ConBee (Beta)

manup, 2018-02-09:
## ZSHARK

A helper tool to transfer sniffer data from ConBee USB dongle to a Wireshark instance.

**Features**
* Local and remote capture
* For Ubuntu, Windows and Raspbian
* Automatic sniffer firmware installation (via GCFFlasher)

Remote capture allows forwarding sniffer data to another computer, for example sniff on a Raspberry Pi and watch in Wireshark on a MacBook.

[image]

The beta version for all platforms can be downloaded at https://www.dresden-elektronik.de/zshark

**Note** deCONZ firmware can not be active at the same time as sniffer firmware on the ConBee. However both firmwares can be used; the deCONZ settings on the ConBee NVRAM will be preserved.

ebaauw, 2018-02-09:
macOS?  Please?

snozzlebert, 2018-02-09:
I think @manup means that you can use Wireshark on macOS while the ConBee with ZShark is running on another machine with Raspbian/Ubuntu/Windows.

manup, 2018-02-09:
>I think @manup means that you can use Wireshark on macOS while the ConBee with ZShark is running on another machine with Raspbian/Ubuntu/Windows.

Also possible ZShark is running in a Ubuntu VM on a Mac and forwards data to Wireshark which runs natively on the same Mac.

Not the perfect solution, a native macOS version is challenging due the firmware flashing part which needs super user rights. I can't provide a ETA but we may provide a ZShark version for macOS there firmware must be installed separately via GCFFlasher in a terminal (same goes for deCONZ).

ebaauw, 2018-02-09:
I want to ditch my Ubuntu VM, if possible.  I only use it for BitCatcher.  Happy to flash the ConBee manually, if that what it takes.

I don't suppose I can use a single Raspberry with a RaspBee _and_ a ConBee installed, running deCONZ on the RaspBee and ZShark on the ConBee in parallel.

manup, 2018-02-09:
>I don't suppose I can use a single Raspberry with a RaspBee and a ConBee installed, running deCONZ on the RaspBee and ZShark on the ConBee in parallel.

Just tried that, yes works too :)

[image]

ebaauw, 2018-02-09:
Indeed, that works!

I get more junk (ICMP messages and ACKs) then ZigBee messages when filtering on port 17754.  If I apply a display filter for zbee_nwk or zbee_aps, WireShark only shows the (ethernet frames with encapsulated) ZigBee frames.

I do seem to miss quite a few packages when sniffing the local deCONZ network.

The Network Settings dialog in the deCONZ GUI shows zeroes for all fields.  Not sure if this is related to ZShark, or RaspBee firmware 0x261D0500 (I use a self compiled version of the REST API plugin, so it updated to this version).  deCONZ seems to work fine nevertheless, even restarting does work.  ZShark was happy to update the ConBee firmware (this is different from the BitCatcher firmware?) while deCONZ was running.

ebaauw, 2018-02-24:
> I do seem to miss quite a few packages when sniffing the local deCONZ network.

I suspect it's due to radio interference between the RaspBee and the ConBee (both connected to the same Raspberry).  I dug up an old USB Female-A to A 5m extension cable (from a previous life), and connected the ConBee through that.  It works (even through 5m is pushing from a USB standard perspective).  I can now place the ConBee halfway in between the RaspBerry and the device being sniffed, and it looks like I now capture the full traffic.

Can I only enter an IP address in the Remote Capture field?  I tried a hostname instead, but that doesn't seem to work?

manup, 2018-02-24:
> It works (even through 5m is pushing from a USB standard perspective). I can now place the ConBee halfway in between the RaspBerry and the device being sniffed, and it looks like I now capture the full traffic.

Good catch I'll try this too, radio interference can be a beast :)

We also are investigating in the missed packages in some scenarios like OTA traffic, for one the RX circular buffer was quite small (8) we raised that to 32. Also sniffer is running at 38400 baud which we can raise since ConBee has a FTDI in a future firmware.

Meanwhile here is the version with larger circular buffer, you may give it a try, it might help but this isn't verified yet:

[sniffer-1.00-32packets.zip](https://github.com/dresden-elektronik/deconz-rest-plugin/files/1754298/sniffer-1.00-32packets.zip)

Flashed as usual:

    $ sudo GCFFlasher_internal -d 0 -f sniffer-1.00-32packets.bin

Framspott, 2018-02-27:
Hi,

ist there a possibility to sniff the communication between a Zigbee device and a Raspbee?
I want to know the communication over the serial port /dev/ttyS0 between my Hue bulb and my Raspberry.

All the best

Framspott

manup, 2018-02-27:
To see whole ZigBee packets you need a ConBee which acts as sniffer device to monitor the ZigBee traffic. The RaspBee can't act as sniffer and coordinator at the same time.

manup, 2018-07-30:
ZSHARK is now officially released.

If you are running an older ZSHARK firmware it's strongly advised to update the firmware contained in the package, we fixed some nasty bugs which should improve sniffer sensitivity a lot.

https://www.dresden-elektronik.de/funktechnik/products/software/zshark/?L=1

ebaauw, 2018-07-30:
Dare I ask about macOS support?

manup, 2018-07-30:
We currently experimenting with the Qt Installer Framework (ZSHARK Windows) and automated builds (deCONZ Raspbian/Ubuntu). As soon as it's stable I hope to also derive automated macOS builds for ZSHARK and deCONZ.

manup, 2018-11-27:
Yes maybe also add a reference in the Wiki, otherwise the Website always has the latest release.

https://www.dresden-elektronik.de/funktechnik/products/software/zshark/?L=1

ebaauw, 2019-03-27:
Pin

manup, 2019-03-27:
Cool didn't know there is a pin option :)

ebaauw, 2019-03-27:
It's still a bit of hack, I'm afaid ;-)

timbru31, 2019-04-10:
I assume the CC2531 USB Stick is not supported? Have a spare one lying around. (so far I've used https://github.com/andrewdodd/ccsniffpiper)

-
A quick question, too: @manup - It seems that a NWK key with 32bit MIC (Message Integrity Code) is used (security level 5) - why no 128bit key for the MIC (security level 7)? Any reason?

antoineQue, 2019-05-24:
Wow, thank you ! 
This is pretty cool: auto flash, fast and reliable. I was able to capture packet that other radio (like ApiMote) did'nt retrieve well. After flashing to sniffer firmware, is there a way to auto reflash for deconz use ?

lynaghk, 2019-06-04:
Does ZShark support the ConBee II?

On Windows 10 deCONZ lists the ConBee II as a device option, but ZShark's device list is empty:

[image]

I've also tried Ubuntu 18.04 LTS in a VirtualBox VM, but ZShark doesn't pick up the device there either (though `lsusb` shows that it's visible within the VM: `Bus 002 Device 002: ID 1cf1:0030 Dresden Elektronik`)

manup, 2019-06-04:
Sorry this is an older PDF, ZSHARK will support ConBee II but it isn't ported yet. There is no ETA, but I hope it will be done within the next two months.

lynaghk, 2019-06-05:
Alright, I'll find another solution. Thanks for the quick reply!

MarcoBodega, 2019-09-18:
> Sorry this is an older PDF, ZSHARK will support ConBee II but it isn't ported yet. There is no ETA, but I hope it will be done within the next two months.

are there any advancement on ZSHARK for ConBee II ?

mascygp, 2019-10-27:
> 
> 
> Sorry this is an older PDF, ZSHARK will support ConBee II but it isn't ported yet. There is no ETA, but I hope it will be done within the next two months.

there are some news about ZSHARK for ConBee II ?

Otnow, 2019-11-04:
> 
> 
> > Sorry this is an older PDF, ZSHARK will support ConBee II but it isn't ported yet. There is no ETA, but I hope it will be done within the next two months.
> 
> there are some news about ZSHARK for ConBee II ?

+1

manup, 2019-11-04:
Sorry no news yet. The work on it will continue as soon as the new bootloader and deCONZ firmware is finished.

mfilippo, 2019-11-19:
Hi,
where can we get ZSHARK?

tannyl, 2019-11-22:
The link to the Zshark software (https://www.dresden-elektronik.de/funktechnik/products/software/zshark/?L=1) does not work anymore, nor can I find it on the phoscon.de as otherwise state on dresden-elektronik.de? (there seems to be a lot of 404 going on your site?)

FlyingPersian, 2019-11-23:
Am I right to assume that Zshark cannot be installed newly atm cuz there isn'r any download available?

ebaauw, 2019-11-23:
I saved the Raspbian package when I installed it: [zshark-1.00.02.deb.gz](https://github.com/dresden-elektronik/deconz-rest-plugin/files/3882857/zshark-1.00.02.deb.gz).

manup, 2019-11-24:
Sorry for the delay, turns out the relaunch of dresden-elektronik.de broke many links. We are working to fix these asap (auto redirect to new location).

For ZSHARK downloads please refer to:

https://phoscon.de/downloads/zshark/

FlyingPersian, 2019-11-25:
Can I use this tool with a CC2531 USB Dongle on Ubuntu or a RPI? I don't have a CC Debugger, but the tutorials I find describe using a CC Debugger. Is it needed though?

kentwinther, 2019-12-07:
I just got my ConBee II, and that's great, but I am not able to select it in the Zshark windows app?
Is ConBee II still not supported by Zshark?

kentwinther, 2019-12-10:
Hi again...
Is ConBee II supported by Zshark or not?
I bought the ConBee II in order to do ZigBig sniffing!

FlyingPersian, 2019-12-10:
According to here yes: https://www.dresden-elektronik.de/funk/software/zshark.html

kentwinther, 2019-12-11:
@FlyingPersian Yes, that's right. ConBee II should be supported by Zshark according to the website. But I´m not able to select my ConBee II in the Zshark app. It does not showup in the dropdown list...
Do you have a ConBee II and does it work in Zshark?

manup, 2019-12-11:
Unfortunately the ConBee II is not yet supported in ZSHARK, the support for it is scheduled for 2020.

kentwinther, 2019-12-11:
@manup That's too bad, as I bought this for sniffing...
Do you know when this will be supported in 2020 (month)?

dandrzejewski, 2019-12-31:
After installing this firmware and playing with it, then going back to the normal firmware, none of my zigbee devices can connect and are not responsive.  They're all EcoSmart bulbs from Home Depot.

dandrzejewski, 2019-12-31:
> 
> 
> After installing this firmware and playing with it, then going back to the normal firmware, none of my zigbee devices can connect and are not responsive. They're all EcoSmart bulbs from Home Depot.

Interestingly, I power-cycled ONE of the bulbs and the entire mesh came back online.

I'm not 100% sure of this, but it looks like the _direct_ links between the raspbee and the other devices have not come back, but it all seems to be functioning using the bulb I power-cycled as a repeater.

ebaauw, 2019-12-31:
> After installing this firmware and playing with it, then going back to the normal firmware, none of my zigbee devices can connect and are not responsive.

Reflashing the RaspBee or ConBee with the ZShark firmware and then back with the deCONZ firmware will likely lose the data in the device’s non-volatile memory.  You might want to double-check that the network parameters (PAN ID, network key, channel) are still valid.

> Interestingly, I power-cycled ONE of the bulbs and the entire mesh came back online.

That’s not how ZigBee works.  Your bulbs have been “online” all the time; they don’t need the coordinator to communicate with each other.  It’s just the RaspBee or ConBee not having discovered the network.

The graph in the GUI is just a graphical representation of the info deCONZ has received from your devices.  It does _not_ show any “active” connections, just the info from the routing tables from the time the device was last queried.  When you power-cycle a device, it sends a _Device Announcement_ message, which deCONZ picks up to query the device with prejudice.

manup, 2019-12-31:
>Reflashing the RaspBee or ConBee with the ZShark firmware and then back with the deCONZ firmware will likely lose the data in the device’s non-volatile memory. You might want to double-check that the network parameters (PAN ID, network key, channel) are still valid.

The values should be stable and preserved since ZSHARK firmware doesn't touch the NVRAM of ConBee I and RaspBee I, which is stored in the EEPROM.

dandrzejewski, 2020-01-03:
> Reflashing the RaspBee or ConBee with the ZShark firmware and then back with the deCONZ firmware will likely lose the data in the device’s non-volatile memory. You might want to double-check that [...]

The OP in this issue states that this should not happen.  But it did end up changing the PAN ID, which I changed back.

> 
> > Interestingly, I power-cycled ONE of the bulbs and the entire mesh came back online.
> 
> That’s not how ZigBee works. Your bulbs have been “online” all the time; they don’t need the coordinator to communicate with each other. It’s just the RaspBee or ConBee not having discovered the network.

Got it.  The UI doesn't make that clear (to me, at least).

> The graph in the GUI is just a graphical representation of the info deCONZ has received from your devices. It does _not_ show any “active” connections, just the info from the routing tables from the [...]

Right, well my point is that I want to rely on this device to be an interface between HomeSeer and ZigBee. After going through this, the fact that it lost track of all the devices when the original post said it should not is concerning.  And the fact that things didn't automatically come back up is, too.

This same issue even happens if I simply power cycle the pi (shutting it down cleanly of course) and then bring it back up.  Even after several hours the device is unable to talk to any other devices.  I would expect things should be able to heal within minutes at worst.

Nevertheless, power cycling one bulb seems to get it "un-stuck," probably for the reason you stated - it sends a device announcement.  

On closer inspection, it appears that the raspbee is only able to contact the other devices "through" the bulb I power cycled (that bulb is a repeater). That's according to the diagram in deCONZ, and evidenced by the fact that everything else stops working again if I power off that bulb.

Is there some parameter I'm missing, or is this a bug, or is there something I just don't understand about how ZigBee is supposed to work?

ebaauw, 2020-01-03:
> On closer inspection, it appears that the raspbee is only able to contact the other devices "through" the bulb I power cycled (that bulb is a repeater). That's according to the diagram in deCONZ, and evidenced by the fact that everything else stops working again if I power off that bulb.

In theory, that could be valid, if this one bulb is the only device in direct range of the RaspBee or ConBee.  What happens after you power-cycle another bulb?

Cyrille63, 2020-04-11:
Hi
ConBee II is not supported by Zshark, have you got a support date?
Thk

closed 2020-06-15 by stale[bot]

tkl-git, 2020-09-17:
> Unfortunately the ConBee II is not yet supported in ZSHARK, the support for it is scheduled for 2020.

Hi, do you have updated information? I think Zshark support for ConBee 2 would be a very useful feature and would really help people to debug Zigbee network problems.

abmantis, 2020-10-08:
Any way to run zshark on a headless machine (without VNC)?

thedayofcondor, 2020-12-24:
Any progress on the zshark support for the conbee ii?

deennoo, 2021-01-03:
how to revert from sniffer on classic Conbee Usb Dongle Please ?

On each update 2 solutions : 
- if zshark is on and connected to conbee usb : gccf try to rebot again and again and again the device
- if zshark is on and not connected to conbee usb : Fw update fail because of snifferv1.0 fw

```
STARTING APP
Sniffer V1.0 [ras (426 ms)
STARTING APP
Sniffer V1.0 [raspbee]
 (442 ms)
could not sync with bootloader
retry, failed
```

Can some one help ?

Smanar, 2021-01-03:
The classic command line to update firmware not working ?

manup, 2021-01-03:
> The classic command line to update firmware not working ? https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Update-deCONZ-manually

Yes that's the correct way to install the deCONZ firmware again (note all configuration parameters are preserved).

A note on ConBee II support, unfortunately this didn't get finished in 2020. There is some active work in progress to get it out in Q1 2021.

abmantis, 2021-02-16:
> Sniffer V1.0 [raspbee]

I am also unable to rever a raspbee from sniffer fw to deconz fw: 
```
GCFFlasher V3_17 (c) dresden elektronik ingenieurtechnik gmbh
Reboot device /dev/ttyAMA0 (RaspBee)
action: reset device RaspBee
wiringPi 2.52 initialized
RaspBee Bootloader premium
Vers. 1.02
build 2013/08/01
STARTING (286 ms)
STARTING APP
 (288 ms)
STARTING APP
Sniffer  (318 ms)
STARTING APP
Sniffer V1.0 [ra (320 ms)
STARTING APP
Sniffer V1.0 [raspbee]
 (323 ms)
could not sync with bootloader
```

azurit, 2022-02-11:
Where can i download the latest firmware? Cannot find it anywhere, thank you.

Smanar, 2022-02-11:
Good question.
The firmware is not provided on the package ?

azurit, 2022-02-11:
It is but is it a newest one?

Anyway, if it's used, zshark is showing error (that's why i'm searching for other versions): Firmware install error

Smanar, 2022-02-12:
It seem it s the good version, others users are using the same link https://forum.phoscon.de/t/announcement-zshark-is-now-available-for-conbee-ii/207

I have never tried on my side, I think there is a different firmware for conbee 1 and conbee 2 ?

azurit, 2022-02-12:
I'm running it directly on Ubuntu linux. It's a Conbee 2. I was also trying to install firmware manually but no luck:
```
$ sudo GCFFlasher_internal -d 0 -f /usr/share/zshark/fw/sniffer_v1_0.bin -x 3
GCFFlasher V3_17 (c) dresden elektronik ingenieurtechnik gmbh
11:38:20:469 using firmware file: /usr/share/zshark/fw/sniffer_v1_0.bin
11:38:20:499 ls dev: /dev/ttyACM0 (0x1CF1/0x0030) sn: DE2480257
11:38:20:499 ls dev: /dev/ttyS4 (0x8086/0x8C3D) sn:
11:38:20:499 ls dev: /dev/ttyS0 (0x0000/0x0000) sn:
11:38:20:499 search FTDI device: no.0
11:38:20:499 device 0 not found
```
deCONZ and Phoscon App are both working fine.

Smanar, 2022-02-12:
Have you tried just

`GCFFlasher_internal -l`

azurit, 2022-02-12:
```
$ GCFFlasher_internal -l
GCFFlasher V3_17 (c) dresden elektronik ingenieurtechnik gmbh
Path             | Vendor | Product | Serial     | Type
-----------------+--------+---------+------------+-------
/dev/ttyACM0     | 0x1CF1 | 0x0030  | DE2480257  | ConBee II
```

Smanar, 2022-02-13:
Ha ?
All look fine, sorry no idea on my side, will ask to others devs if someone have already do it.

[3 more replies]