dresden-elektronik/deconz-rest-plugin #406: Raspbee as router in Hue network

FrodoKenny, 2018-02-09:
I've been following the instructions in #21 to get my shiny new Raspbee to join my Hue network for monitoring purposes. Like mentioned in that thread, the  TC link key keeps resetting to the master key. After a bit of trial and error it does save if I don't set the TC address to 0xffffffffffffffff.

I've tried with 0x26050500, which is suggested to work, and with 0x260b0500, and the current 0x26190500.

Should this work? Anything else I can try?

Many thanks

ebaauw, 2018-02-09:
I haven't been able to set this up anymore for a long time...

FrodoKenny, 2018-02-09:
Right, maybe it's really a Hue update that broke it. Potentially the just released Hue bridge update to zigbee 3.0 helps, I haven't updated yet...

FrodoKenny, 2018-02-12:
Ok, I've tried a few more firmware versions, going back all the way to 0x19140500 and they all behave the same; the TC link key can be set as long as the TC address is the MAC address of the raspbee, if you change the TC address the TC link key resets to 5A...39. So there must be something else going on besides firmware changes.

Can this be looked into please? I specifically like to  use the Raspbee to analyse some issues in my Hue network. I'm happy to help in any way possible. Is this the appropriate place for this or do I need to raise a support request?

Also, it seems that somewhere along the way my Raspbee node got the name XBee and I can't seem to rename it. Is there some way to reset the NVRAM of the Raspbee?

closed 2019-03-30 by stale[bot]