dresden-elektronik/deconz-rest-plugin #407: Battery-powered vs Mains-powered devices

ebaauw, 2018-02-09:
The current deCONZ code seems to assume that end devices are battery powered and routers are mains powered.  This is not always the case.  Increasingly, mains-powered lights and switches don't have router functionality in their ZigBee stack.  The Heiman siren is a router, is mains powered and has a battery backup (value 81 for _Power Source_ attribute 0x0007 in the _Basic_ cluster), which I would want to expose.

I think we need to use something like `hasBattery()` and `isMainsPowered()` for device discovery and configuration instead of `isEndDevice()`.  Not sure where best to implement these; `isEndDevice()` is a deCONZ core function, defined in `/usr/include/deconz/node.h`.  I'm not sure how early in the discovery this would be known.  Would we base it on the _Power Source_ attribute in the _Basic_ cluster?  On the _Power Configuration_ cluster?  On the Power Descriptor?  Neither seem to read early enough.

manup, 2018-02-09:
Agree, isEndDevice() doesn't fit all devices anymore.

Usually for the `rxOnWhenIdle` flag from the NodeDescriptor can be used for that purpose. We should check in the NodeInfo panel that all the related (end) devices have this flag set to true. The REST-API already has access to it via `deCONZ::Node::nodeDescriptor().receiverOnWhenIdle()` method.

NodeDescriptor is also available very early in discovery.

ebaauw, 2018-02-09:
Thanks, @manup.  This could be the key to exposing end-device lights.

I'm afraid this won't help me for the Heiman Siren, though.  I'm wondering why you only put the _Power Configuration_ in the fingerprint for sensors from these manufacturers:
https://github.com/dresden-elektronik/deconz-rest-plugin/blob/475c614a33335e5fb3e8b3b383b42876dd7c2ccc/de_web_plugin.cpp#L2601-L2613

I would want to put in it regardless, also for the other sensor types and then expose `config.battery` if the cluster is in the fingerprint, instead of when it's an end device.
https://github.com/dresden-elektronik/deconz-rest-plugin/blob/475c614a33335e5fb3e8b3b383b42876dd7c2ccc/de_web_plugin.cpp#L2960-L2963

Or do you know of mains-powered sensors than expose _Power Configuration_, but don't have a (backup) battery?

manup, 2018-02-10:
>I would want to put in it regardless, also for the other sensor types and then expose config.battery if the cluster is in the fingerprint, instead of when it's an end device.

The cluster is added here mainly to create the binding for attribute reporting later on, in my past tests not all devices supported this, that's the reason I whitelisted the 'proper' ones.

>Or do you know of mains-powered sensors than expose Power Configuration, but don't have a (backup) battery?

Wrong assumption here was that every end-device is battery powered. The NodeDescriptor.rxOnWhenIdle flag should be used here instead.

ebaauw, 2018-02-10:
> The cluster is added here mainly to create the binding for attribute reporting later on

They're whitelisted (again) in bindings.cpp, so I suppose we needn't repeat that here.
https://github.com/dresden-elektronik/deconz-rest-plugin/blob/475c614a33335e5fb3e8b3b383b42876dd7c2ccc/bindings.cpp#L1131-L1140

> Wrong assumption here was that every end-device is battery powered. The NodeDescriptor.rxOnWhenIdle flag should be used here instead.

Won't help me with the Heiman Siren, which is mains-powered with a battery backup.  It provides a _Power Configuration_ cluster for the backup battery.  I want to expose this through `config.battery`.  It doesn't support attribute reporting on _Battery Percentage_, though.

I did a check against all my devices: it seems that `nodeDescriptor().receiverOnWhenIdle()` consistently reports the value we want/need: false for battery-operated devices; true for mains-operated devices.  The Siren is still receiving while on backup battery.  I don't have any end-device lights or plugs myself, so cannot test that part.

manup, 2018-02-18:
It seems that the mains powered end-devices are working now, shall we close the issue?

closed 2018-02-18 by ebaauw