dresden-elektronik/deconz-rest-plugin #409: Firmware update not showing on raspbee deCONZ

wvuyk, 2018-02-10:
I have installed the 2.05.01 update on the RaspBee here 3 days ago. But until now the update is not showing in the webapp or the Phosconapp. I saw somewhere that it could take some time before it woud show up, but is this not a bit long now?

Wim

manup, 2018-02-10:
Hmm strange, which firmware version is shown in *deCONZ > Help > About deCONZ*?

wvuyk, 2018-02-10:
Version 0x26190500 here
[image]

manup, 2018-02-11:
It might be related to the firmware version state advertised by the online update service. I'll check this out, 2.05.02 with firmware version 0x261e0500 will be available soon.

wvuyk, 2018-02-12:
2.05.02 showed the firmware update correctly again. Thanks!

closed 2018-02-12 by wvuyk