dresden-elektronik/deconz-rest-plugin #410: Wrong state on first sensor update

MrPsi, 2018-02-11:
I have a Xiaomi Aqara door/window sensor paired with my Conbee. The Conbee is connected to a Raspberry PI 3. If I restart the Raspberry, while the door is closed, and wait for deConz to start. Phoscon will show the sensor as unavailable. Now if I open the door the sensor will update and be available in Phoscon, but it will show it as closed, even though the door is open. I don't think this is expected behavior? Shouldn't Phoscon show the door as open?

Running version 2.05.01 of deConz.

manup, 2018-02-11:
>I don't think this is expected behavior? 

Yes it should work normally, I have one installed for the roof window which shows correct state.
Does yours update the state if you open/close the door or does it always show closed?

MrPsi, 2018-02-11:
It is just when I open the door after it is unavailable it will display it as closed when it should be open.
If I close the door and open it again it will show it as open.

I have tried to reset the gateway and re-add the sensor, with the same result.

manup, 2018-02-11:
Ok I understand, I think this can be fixed during the next releases.

MrPsi, 2018-02-11:
Awesome, thanks :)

MrPsi, 2018-02-16:
I install the new version today (2.05.02), to check if my issue was fixed. After I installed the new version I was told that there were a new firmware version to install (0x261e0500). I clicked to update the firmware. But after I restarted it told me to update the firmware again. So I tried again, but the same thing keeps happening. It does not seem to install the new firmware, even though each time I try to update it tells me that it succeeded.
Is there a way to check which firmware that is actually installed? I'm running without a GUI, and I can't find any way to check in the REST API, only to update.
Is the firmware available for download, so I can update it manually from the command line?

MrPsi, 2018-02-16:
It seems like all my lights have disappeared and all my sensors have stopped working too.

manup, 2018-02-16:
Hi to manually update the firmware of your ConBee do the following:

    $ sudo systemctl stop deconz
    $ sudo GCFFlasher_internal -d 0 -f /usr/share/deCONZ/firmware/deCONZ_Rpi_0x261e0500.bin.GCF
    $ sudo systemctl start deconz

There is no easy way to see the firmware version without the deCONZ GUI yet. A bit hack way is to open the Phoscon App and in the Browsers developer console type `app.config.get('fwversion')`.

MrPsi, 2018-02-16:
The console says that I have version 0x261e0500, so I'm not sure why Phoscon tells me that I can update to that version.

I loaded a backup of my lights and sensors too, so that seems to work again.

manup, 2018-02-16:
Ok, maybe it took a while to rediscover the network, after a firmware update neighbour tables and routing informations is rebuild from scratch.

MrPsi, 2018-02-16:
Yes, maybe I was too quick to load the backup. Will try and wait a bit longer next time if it happens again. Thanks for your help.
I will do some testing with the door sensor and get back to you regarding the issue.

manup, 2018-02-17:
The issue is not fixed yet but should be soon.

MrPsi, 2018-02-18:
I updated to version 2.05.03 today and got the same issue with the firmware update button. The button to update the firmware showed even though I had that firmware installed already. When I tried to check which firmware that was installed by typing `app.config.get('fwversion')` in the console I got `0x00000000` as a response. After some restarts i finally got `0x261e0500` as a response and the update button disappeared.

manup, 2018-02-18:
That is strange, it seems that the connection to the device can't be established reliably.
Do you have further USB devices connected to the Raspberry?

MrPsi, 2018-02-18:
No I only have the Conbee connected to it.
I can't say if it was the restarts that fixed the problem or if it just started working after some time though.
But if I remember correctly I could still see the state of the sensors even though I got `0x00000000` as a response for the firmware.

MrPsi, 2018-03-03:
The issue with the door sensor seems to be fixed now. Thanks for a quick response and fix.

closed 2018-03-03 by MrPsi