dresden-elektronik/deconz-rest-plugin #411: Problem beim Installieren (Raspian Desktop x86)

Nastras, 2018-02-11:
Hallo,

da ihr ja ein Deutsches Unternehmen seit versuche ich es mal auf Deutsch und bitte um Hilfe.

Ich habe bisher einen Raspee an meinem RaspberryPi im Betrieb gehabt dieser hat auch sehr gut funktioniert. Da ich etwas mehr Rechenleistung brauchte habe ich mir auf einem PC Raspian Desktop Stretch installiert und heute bei euch im Shop einen Conbee Stick bestellt.

Voller Vorfreude auf den Stick habe ich gerade probiert deconz zu installieren auf dem oben genannten Rechner. Leider bekomme ich folgende Fehlermeldung:

[image]

Ich hoffe ihr könnt mir weiterhelfen?

lg

wrmulf, 2018-02-11:
Wrong architecture. You need the [i386-binaries](https://www.dresden-elektronik.de/deconz/ubuntu/beta/deconz-2.05.02-qt5.deb) for your computer.

Nastras, 2018-02-11:
Great, that works !

Where can I find an overview of the downloads for the different systems?

Thank you!!!

closed 2018-02-11 by Nastras